marcone/teslausb #81: Tesla wakeup API doesn't seem to work

patrikrunald, 2019-06-15:
I have configured teslausb to keep the car awake until it's done copying files to a fileserver on the LAN. tesla_api.json is correctly populated and I can see in archiveloop.log that it's sending the command but it doesn't seem to work as the file transfers stop after a few minutes when the car goes to sleep.

> Fri 14 Jun 16:30:16 MDT 2019: Moving '2019-06-14_08-52-36/2019-06-14_08-50-10-front.mp4'
Fri 14 Jun 16:30:25 MDT 2019: Moved '2019-06-14_08-52-36/2019-06-14_08-50-10-front.mp4'
Fri 14 Jun 16:30:29 MDT 2019: Sending wakeup API command...
Fri 14 Jun 16:30:30 MDT 2019: Moving '2019-06-14_08-52-36/2019-06-14_08-50-10-left_repeater.mp4'
Fri 14 Jun 16:30:40 MDT 2019: Moved '2019-06-14_08-52-36/2019-06-14_08-50-10-left_repeater.mp4'
Fri 14 Jun 16:30:40 MDT 2019: Moving '2019-06-14_08-52-36/2019-06-14_08-50-10-right_repeater.mp4'
Fri 14 Jun 16:30:50 MDT 2019: Moved '2019-06-14_08-52-36/2019-06-14_08-50-10-right_repeater.mp4'
Fri 14 Jun 16:30:50 MDT 2019: Moving '2019-06-14_08-52-36/2019-06-14_08-51-11-front.mp4'
Fri 14 Jun 16:31:00 MDT 2019: Moved '2019-06-14_08-52-36/2019-06-14_08-51-11-front.mp4'
Fri 14 Jun 16:31:00 MDT 2019: Moving '2019-06-14_08-52-36/2019-06-14_08-51-11-left_repeater.mp4'

That's the last entry until I make sure it has power again and the log says:

> Thu  3 Nov 11:17:09 MDT 2016: Starting archiveloop...
Thu  3 Nov 11:17:10 MDT 2016: Archiving...
Thu  3 Nov 11:17:10 MDT 2016: Disconnecting usb from host...
Thu  3 Nov 11:17:10 MDT 2016: Disconnected usb from host.
Thu  3 Nov 11:17:10 MDT 2016: Ensuring cam archive is mounted...
Thu  3 Nov 11:17:10 MDT 2016: Mounting /mnt/archive...
mount error(16): Device or resource busy
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
Thu  3 Nov 11:17:10 MDT 2016: Failed to mount /mnt/archive.
Thu  3 Nov 11:17:11 MDT 2016: Sleeping before retry...
Thu  3 Nov 11:17:12 MDT 2016: Retrying...
Thu  3 Nov 11:17:12 MDT 2016: /mnt/archive is already mounted.
Thu  3 Nov 11:17:12 MDT 2016: Ensured cam archive is mounted.
Thu  3 Nov 11:17:12 MDT 2016: Starting recording archiving...
Thu  3 Nov 11:17:12 MDT 2016: Ensuring cam file is mounted...
Thu  3 Nov 11:17:12 MDT 2016: Mounting /mnt/cam...
Thu  3 Nov 11:17:12 MDT 2016: Mounted /mnt/cam.
Thu  3 Nov 11:17:12 MDT 2016: Ensured cam file is mounted.
Thu  3 Nov 11:17:12 MDT 2016: Running fsck on /mnt/cam...
fsck from util-linux 2.29.2
fsck.fat 4.1 (2017-01-24)
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
 Automatically removing dirty bit.
Performing changes.
/dev/loop0: 285 files, 161179/3785199 clusters
Thu  3 Nov 11:17:16 MDT 2016: Finished fsck on /mnt/cam.
Thu  3 Nov 11:17:16 MDT 2016: Moving clips to archive...
Thu  3 Nov 11:17:16 MDT 2016: Creating output directory ''
Thu  3 Nov 11:17:16 MDT 2016: Creating output directory '2019-06-14_08-39-11'
Thu  3 Nov 11:17:16 MDT 2016: Creating output directory '2019-06-14_08-46-07'
Thu  3 Nov 11:17:16 MDT 2016: Creating output directory '2019-06-14_08-52-36'
Thu  3 Nov 11:17:16 MDT 2016: Moving '2019-06-14_08-52-36/2019-06-14_08-51-11-left_repeater.mp4'
Fri 14 Jun 20:03:10 MDT 2019: Moved '2019-06-14_08-52-36/2019-06-14_08-51-11-left_repeater.mp4'
Fri 14 Jun 20:03:14 MDT 2019: Sending wakeup API command...
Fri 14 Jun 20:03:16 MDT 2019: Moving '2019-06-14_08-52-36/2019-06-14_08-51-11-right_repeater.mp4'

I am using the main-dev so perhaps this feature isn't completed yet?

marcone, 2019-06-15:
Can you try running the command manually and see what happens?
I.e, ssh in to the pi and run
```
sudo -i
/root/bin/tesla_api.py wake_up_vehicle
```

norse76, 2019-06-16:
The output is:

```
# /root/bin/tesla_api.py wake_up_vehicle
Sat 15 Jun 20:07:52 MDT 2019: Sending wakeup API command...
```
I added the --debug param and here's the output. Looks good to me, not sure why the transfer process was interrupted.

```
# /root/bin/tesla_api.py --debug wake_up_vehicle
Sat 15 Jun 20:10:14 MDT 2019: Loading mutable data from disk...
Sat 15 Jun 20:10:14 MDT 2019: Calling wake_up_vehicle()...
Sat 15 Jun 20:10:14 MDT 2019: Sending wakeup API command...
Sat 15 Jun 20:10:17 MDT 2019: "online"
```

marcone, 2019-06-18:
I didn't write the wakeup code, but that looks to me like it's sending the request and getting a response back. I was going to suggest asking about this over on [the fork of the person that wrote the code](https://github.com/coredumperror/teslausb), but I see that @coredumperror has turned off issue reporting for that repo.

coredumperror, 2019-06-18:
Huh, I didn't do that on purpose. Maybe it's off by default on my account or something? I'm not really sure how to fix that.

Strange that the wakeup mechanism isn't working. I remember using that specific command to keep my car awake in the past... I'll have to do some more testing, when I get the chance.

coredumperror, 2019-06-18:
@norse76 Do you have a Model 3? That's the only kind of Tesla I have, so maybe other Models behave differently?

norse76, 2019-06-18:
@coredumperror I have a Model X. It happened yesterday again. I'll try to do some more testing tonight and hit the Tesla API to check state of the car after I run the wakeup command.

appark, 2019-06-20:
Has anyone noticed that USB ports in the Telsa 3 do not power off?  https://www.reddit.com/r/teslamotors/comments/c2iqn6/teslausb/

raushel, 2019-06-20:
They do power off eventually, but sentry mode (always on) or while charging (on the 3 anyway) keep them powered for longer.

appark, 2019-06-20:
I don't have sentry mode enable.  I am not charging.  USB ports have been on for 10 hours

norse76, 2019-06-20:
I played around with the Tesla API last night and woke up the car, enabled HVAC etc but nothing will make the USB ports power on apart from opening the door. The USB ports don't even stay on when charging. Must be the Model X behaving very differently from the Model 3.

coredumperror, 2019-06-20:
> I played around with the Tesla API last night and woke up the car, enabled HVAC etc but nothing will make the USB ports power on apart from opening the door. The USB ports don't even stay on when charging. Must be the Model X behaving very differently from the Model 3.

Thank you for that update. 

Curious... Now some people's Model 3s are never powering off their USB ports, and yet Model X is rarely powering them *on*. This is going to be confusing to properly document.

thijsdejong, 2019-06-30:
I'm getting the following error messages. All the variables are 100% correct:

```
Sun 30 Jun 15:28:18 BST 2019: Unable to create access token:
Sun 30 Jun 15:28:18 BST 2019: {'response': 'authorization_required_for_txid_``'}
```

I set it up like this:
```
export tesla_email="email@example.com"
export tesla_password="password"
# Please also provide your vehicle's VIN, so TeslaUSB can keep the correct
# vehicle awake.
export tesla_vin="5YJ3E7blablabla"
```

coredumperror, 2019-07-01:
I believe the problem is those quote marks. Those aren't necessary, and the parser for the settings file will pick them up as being part of the credentials. Take those out and try again, and it should work.

Phontana, 2019-07-12:
> I believe the problem is those quote marks. Those aren't necessary, and the parser for the settings file will pick them up as being part of the credentials. Take those out and try again, and it should work.

I did this wrong the first time as well, causing my account to get blocked eventually because of too many incorrect login attempts. The documentation isn't clear about the quotation marks, and it seems obvious they are needed here as well since the other passwords also need quotation marks.

coredumperror, 2019-07-15:
Alright, I'll tweak the docs to make it more clear that quotes are not to be used.

marcone, 2019-07-15:
Wouldn't it be better to parse the variables in bash, and pass them to the python script as command line parameters or environment variables?
(setup could write them to teslausb.conf, which gets sources by archiveloop, and then the python script can just read them from os.environ I think)

coredumperror, 2019-07-15:
It would be better! Unfortunately, I have no idea how to do that. My shell coding skills are super amateurish.

marcone, 2019-07-16:
This (untested) change copies the email/password/vin values defined in the setup config file to teslausb.conf, which is sourced by archiveloop and so should ensure they are available in the environment for the python script to read: https://github.com/marcone/teslausb/commit/2ae9e44e276f92c5c4ffbba0e9cff6e621b4c373

(names to look for are TESLAAPI_EMAIL, TESLAAPI_PASSWORD and TESLAAPI_VIN)

hokiegir1, 2019-08-02:
Question...does the API code have a parameter to only work when the charger is connected? I'm noticing if I plug the car in, my transfers complete well, but if not, it shuts down after some undetermined amount of time without completing.

marcone, 2019-08-03:
@hokiegir1 that sounds like TeslaApi is not configured correctly (wrong username, password or VIN), and that plugging in the car is what keeps it awake.

gonzonia, 2019-08-10:
I'm having the same problem. The archives aren't finishing unless I connect it to a different power source (i.e. my computer) or leave the door open so the Model X doesn't go to sleep.  I've tested the API calls using
 root/bin/tesla_api.py --debug get_odometer
Sat 10 Aug 02:52:06 BST 2019: Loading mutable data from disk...
Sat 10 Aug 02:52:06 BST 2019: Calling get_odometer()...
Sat 10 Aug 02:52:09 BST 2019: 5376

and /root/bin/tesla_api.py --debug wake_up_vehicle
Sat 10 Aug 02:54:51 BST 2019: Loading mutable data from disk...
Sat 10 Aug 02:54:51 BST 2019: Calling wake_up_vehicle()...
Sat 10 Aug 02:54:51 BST 2019: Sending wakeup API command...
Sat 10 Aug 02:54:54 BST 2019: "online"

So the error doesn't appear to be with the authentication. It just seems to be going to sleep before it can finish copying. Either it's taking too long to copy the files, or the wake up isn't waking up the USB.

hokiegir1, 2019-08-10:
I just got the same results (well, my own odometer reading, not 5376) using the above, so it's not a credential issue. It could be a slow connection for me, but it usually only does a few files before disconnecting unless the car is plugged in. Haven't tested an external battery on the pi yet. It's on the to-do list, but I've had other priorities first (which is why even this took so long to test).

coredumperror, 2019-08-12:
Maybe you guys have a really slow WiFi connection in your garage? If I recall correctly, TeslaUSB is designed to send the wakeup signal once every 5 minutes, but only between file transfers. If a transfer is taking a very long time, it might not get the chance to send the wakeup signal before the car falls asleep.

Though looking at the commit log, it seems that the wakeup signal was actually moved to a separate script about a month ago. The wakeup is now executed every 2 minutes, and it doesn't see to be tied directly to the transfer process any more. Have you been using the latest code?

gonzonia, 2019-08-12:
I looked at the script and it was going every 5. I changed it to 100 seconds to see if that helps.

marcone, 2019-08-12:
@gonzonia if it was 5 minutes then you're using the old code that would trigger the wakeup between archiving files and is susceptible to the problem mentioned above . The current code does the wakeup in a separate background task every 2 minutes. You should update:
```
sudo -i
/root/bin/remountfs_rw
/root/bin/setup-teslausb selfupdate
/root/bin/setup-teslausb
```

closed 2019-08-12 by marcone