marcone/teslausb #83: Pushover changes to system not being saved

cyb3rl4d, 2019-06-18:
Hi, great project thanks so much for all your hard word.

I'm trying to save pushover settings to the pi after setup. I'm running the **remountfs_rw** command first and its working as I can see the "**export pushover...**" commands in my bash history. Before reboot I've checked they are correct with a export -p but after reboot they are no longer displayed. I'm thinking they dont stay as a variable but when running "**setup-teslausb diagnose**" I can see it print the following. Can you let me know if I'm doing anything wrong here?

```
configure: Pushover not configured.
configure: Pushover not configured.
```

Thanks

marcone, 2019-06-18:
> Hi, great project thanks so much for all your hard work.
> 
> I'm trying to save pushover settings to the pi after setup. I'm running the **remountfs_rw** command first and its working as I can see the "**export pushover...**" commands in my bash history. Before reboot I've checked they are correct with an **export -p** but after reboot they are no longer displayed. 

That's because exporting a variable simply defines/exports a variable in the current shell. After a reboot you're in a different/new shell.

> I'm thinking they dont stay as a variable but when running "**setup-teslausb diagnose**" I can see it print the following. Can you let me know if I'm doing anything wrong here?
> 
> ```
> configure: Pushover not configured.
> configure: Pushover not configured.

If you want to configure pushover (or any aspect of teslausb, really), you need to define the appropriate settings in the teslausb_headless_setup.conf file, and then run setup-teslausb

closed 2019-06-18 by marcone

cyb3rl4d, 2019-06-18:
great thanks will do