marcone/teslausb #85: [Question] Do we need to manually delete archived footage?

skaforey, 2019-06-20:
Seems like my SD card is filled up pretty quickly with sentry events.  It's not completely full yet, but I want to avoid any problems.  Do I need to manually delete them, or will Tesla automatically overwrite old footage? I've heard of some weird issues with Sentry mode preventing autopilot from working when the USB drive is full.....(wtf??)

jhtown, 2019-06-20:
I have the same issue that the tesla sometimes cannot overwrite the old files, or maybe due to the saved files have occupied all the space. **Is that a way to manually delete files through ssh**, so at least I don't have to fetch the Pi zero from the car. 
Thank you very much,

XmemphistoX, 2019-06-20:
What you're describing has nothing to do with this project.   However there are two types of recordings.

Sentry and 1 hour looping recording.  The way that they are stored and deleted are different.

Sentry (event recording) does not auto delete.  Of course it wouldn't, you wouldn't want a video of someone breaking into your car to be auto deleted once it got full.  This is what this project aims to automate.  It will move all sentry videos off the flash storage to a share on your network.

1 hour video looping recording will auto delete after 1 hour to be replaced by new footage.  If you're driving at 10am drive for two hours until 12. You will only have footage from 11am-12.  It does this so it can continuously record without filling up the storage.

Sorry I meant three types. Manual saves.  When you hit the camera button on the screen it will save the previous 10 minutes of footage.  This footage will never delete manually.  TeslaUSB will also move these saves to your network storage.

Manually saved files - TeslaUSB will move them to your cifs
Sentry Videos - TeslaUSB will move them to your cifs
1 hour looping recordings - TeslaUSB will not move them (this is already automatically managed)

skaforey, 2019-06-20:
For sentry clips, I'm not seeing them "Moved".  They get copied to my SMB share, but I still see them on the memory card as well, in the archive folder. 

# du -m | sort -n | tail -1
49239   ./archive

marcone, 2019-06-20:
Not sure which "archive folder" you mean, but saved recordings (the ones in the TeslaCam/SavedClips folder, created by Sentry mode or manually pressing the save button), are the recordings that get archived by teslausb, and should be deleted by teslausb after archiving.
If they're not, check archiveloop.log to see if it says anything about why they weren't deleted.

skaforey, 2019-06-20:
I'm talking about /mnt/archive.

The files are definitely getting copied to my SMB share, but it doesn't seem like they're being cleared out of the /mnt/archive folder.  I'm not sure if this is expected behavior or not.

raushel, 2019-06-20:
/mnt/archive is mapped to your smb share.

skaforey, 2019-06-20:
Well now I feel completely stupid!  Thanks for clearing that up.  I'm embarrassed to call myself a software engineer at the moment

closed 2019-06-20 by marcone

marcone, 2021-09-02:
It now keeps the most recent recordings on the drive (so that they can still be viewed in the car even after archiving), while removing older recordings to ensure enough space is available for new recordings.