marcone/teslausb #86: [feature request] Accespoint stand alone.

dmachadol, 2019-06-20:
Can you create the function of access point to share samba videos when you are not registered to a wifi infrastructure?
When we are traveling or away from home, we can play videos directly with the phone.

dhtguru, 2019-07-22:
I was wandering the same thing.  If traveling it would be practical if there was a way to get to the videos when you are not going to be back home for a few days, if you need to.

RSickenberg, 2019-08-15:
Why did you closed this? Was a good idea

dmachadol, 2019-08-15:
I’m not close this feature request. 🤷🏻‍♂️

roisterous, 2019-08-17:
So we basically want to add this functionality right? https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md 

But we'd also need a way of it identifying when we get home so it can switch from WAP host mode to Wireless connection (slave?) mode. I'm assuming it can't do both without a second Wireless network card/usb. So I'm not sure what could be used to trigger the swapping between modes - you could build a http front end or something to make it switch over but then you'd need to remember to do that all the time. 

Getting it to switch over to WAP mode seems easy enough though, just detect if your home wi-fi network is in range and if it is no longer present for 10 minutes or something, switch.

Perhaps just checking if anything is connected to the WAP mode every 30 seconds and if not swap the modes to look for the home network and if not found swap back to WAP mode? I guess this would work if the switching process was quick but I think it'd also have problems if your phone maintains connection to the WAP.

I dono. Ideas?

marcone, 2019-08-17:
Apparently it's possible to do both at the same time: https://github.com/lukicdarkoo/rpi-wifi

roisterous, 2019-08-17:
Awesome, have you tried it?

skyred, 2019-08-29:
Using this docker set up might be easier, plus, it also provides a new solution for managing WIFI connections. https://imti.co/iot-wifi/

marcone, 2019-09-03:
Wifi AP is now implemented in the main branch. Combined with SMB access it's now possible to watch recordings while on the go.

closed 2019-09-03 by marcone

skyred, 2019-09-03:
WOW, you are fast! Please make a release?

marcone, 2019-09-03:
There is no need for a new release, (re)running setup will use the latest setup scripts.
See https://github.com/marcone/teslausb/blob/main-dev/doc/teslausb_setup_variables.conf.sample for the variables to define.

skyred, 2019-09-05:
I re-flushed my Pi, then tried this feature. The setup process completed, but I didn't see the WIFI AP.

below is my teslausb-headless-setup.log

````
Sat 15 Jun 22:27:38 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sat 15 Jun 22:27:38 BST 2019 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Sat 15 Jun 22:27:38 BST 2019 : Rebooting...
Sat 15 Jun 22:28:03 BST 2019 : Detecting whether to update wpa_supplicant.conf
Sat 15 Jun 22:28:03 BST 2019 : Grabbing main setup file.
Sat 15 Jun 22:28:05 BST 2019 : Starting setup.
Sat 15 Jun 22:28:05 BST 2019 : reading config from /root/teslausb_setup_variables.conf
Sat 15 Jun 22:28:06 BST 2019 : curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev
Sat 15 Jun 22:28:06 BST 2019 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Sat 15 Jun 22:28:10 BST 2019 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Sat 15 Jun 22:28:13 BST 2019 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Sat 15 Jun 22:28:17 BST 2019 : 'curl -s https://api.github.com/repos/marcone/teslausb/git/refs/heads/main-dev' failed, retrying
Wed  4 Sep 17:48:26 BST 2019 : curl -o /tmp/setup-teslausb https://raw.githubusercontent.com/marcone/teslausb/86022c0645904d778e1f56de69005754345a6f03/setup/pi/setup-teslausb
Wed  4 Sep 17:48:28 BST 2019 : Downloaded /tmp/setup-teslausb ...
Wed  4 Sep 17:48:28 BST 2019 : /root/bin/setup-teslausb is up to date
Wed  4 Sep 17:48:28 BST 2019 : Updating package index files...
Wed  4 Sep 17:56:56 BST 2019 : curl -o /root/bin/tzupdate.py https://raw.githubusercontent.com/marcone/tzupdate/develop/tzupdate.py
Thu  5 Sep 00:58:04 CST 2019 : Set system timezone to Asia/Shanghai.
Thu  5 Sep 00:58:12 CST 2019 : Verifying that the requested configuration is valid...
Thu  5 Sep 00:58:12 CST 2019 : curl -o /tmp/verify-configuration.sh https://raw.githubusercontent.com/marcone/teslausb/86022c0645904d778e1f56de69005754345a6f03/setup/pi/verify-configuration.sh
Thu  5 Sep 00:58:13 CST 2019 : Downloaded /tmp/verify-configuration.sh ...
Thu  5 Sep 00:58:13 CST 2019 : Verifying that there is sufficient space available on the MicroSD card...
Thu  5 Sep 00:58:13 CST 2019 : There is sufficient space available.
Thu  5 Sep 00:58:22 CST 2019 : Downloading additional setup scripts.
Thu  5 Sep 00:58:22 CST 2019 : curl -o /root/bin/remountfs_rw https://raw.githubusercontent.com/marcone/teslausb/86022c0645904d778e1f56de69005754345a6f03/run/remountfs_rw
Thu  5 Sep 00:58:23 CST 2019 : Downloaded /root/bin/remountfs_rw ...
Thu  5 Sep 00:58:23 CST 2019 : curl -o /root/bin/make_snapshot.sh https://raw.githubusercontent.com/marcone/teslausb/86022c0645904d778e1f56de69005754345a6f03/run/make_snapshot.sh
Thu  5 Sep 00:58:24 CST 2019 : Downloaded /root/bin/make_snapshot.sh ...
Thu  5 Sep 00:58:24 CST 2019 : curl -o /root/bin/mount_snapshot.sh https://raw.githubusercontent.com/marcone/teslausb/86022c0645904d778e1f56de69005754345a6f03/run/mount_snapshot.sh
Thu  5 Sep 00:58:25 CST 2019 : Downloaded /root/bin/mount_snapshot.sh ...
Thu  5 Sep 00:58:25 CST 2019 : curl -o /root/bin/mount_image.sh https://raw.githubusercontent.com/marcone/teslausb/86022c0645904d778e1f56de69005754345a6f03/run/mount_image.sh
Thu  5 Sep 00:58:29 CST 2019 : Downloaded /root/bin/mount_image.sh ...
Thu  5 Sep 00:58:29 CST 2019 : curl -o /root/bin/release_snapshot.sh https://raw.githubusercontent.com/marcone/teslausb/86022c0645904d778e1f56de69005754345a6f03/run/release_snapshot.sh
Thu  5 Sep 00:58:31 CST 2019 : Downloaded /root/bin/release_snapshot.sh ...
Thu  5 Sep 00:58:31 CST 2019 : curl -o /tmp/create-backingfiles-partition.sh https://raw.githubusercontent.com/marcone/teslausb/86022c0645904d778e1f56de69005754345a6f03/setup/pi/create-backingfiles-partition.sh
Thu  5 Sep 00:58:34 CST 2019 : Downloaded /tmp/create-backingfiles-partition.sh ...
Thu  5 Sep 00:58:34 CST 2019 : curl -o /tmp/create-backingfiles.sh https://raw.githubusercontent.com/marcone/teslausb/86022c0645904d778e1f56de69005754345a6f03/setup/pi/create-backingfiles.sh
Thu  5 Sep 00:58:36 CST 2019 : Downloaded /tmp/create-backingfiles.sh ...
Thu  5 Sep 00:58:36 CST 2019 : curl -o /tmp/make-root-fs-readonly.sh https://raw.githubusercontent.com/marcone/teslausb/86022c0645904d778e1f56de69005754345a6f03/setup/pi/make-root-fs-readonly.sh
Thu  5 Sep 00:58:37 CST 2019 : Downloaded /tmp/make-root-fs-readonly.sh ...
Thu  5 Sep 00:58:37 CST 2019 : curl -o /tmp/configure.sh https://raw.githubusercontent.com/marcone/teslausb/86022c0645904d778e1f56de69005754345a6f03/setup/pi/configure.sh
Thu  5 Sep 00:58:38 CST 2019 : Downloaded /tmp/configure.sh ...
Thu  5 Sep 00:58:38 CST 2019 : Fixing the modules-load parameter in /boot/cmdline.txt...
Thu  5 Sep 00:58:38 CST 2019 : Fixed cmdline.txt.
Thu  5 Sep 00:58:47 CST 2019 : Checking existing partitions...
Thu  5 Sep 00:58:48 CST 2019 : Modifying partition table for backing files partition...
Thu  5 Sep 00:58:48 CST 2019 : Modifying partition table for mutable (writable) partition for script usage...
Thu  5 Sep 00:58:49 CST 2019 : Writing updated partitions to fstab and /boot/cmdline.txt
Thu  5 Sep 00:58:49 CST 2019 : Formatting new partitions...
Thu  5 Sep 00:58:56 CST 2019 : Mounting the partition for the backing files...
Thu  5 Sep 00:58:56 CST 2019 : Mounted the partition for the backing files.
[... cut ...]
# Please also provide your vehicle's VIN, so TeslaUSB can keep the correct
# vehicle awake.
# export tesla_vin=5YJ3E1EA4JF000001

# Uncomment and change if you want setup scripts to be pulled
# from a different repo than github.com/marcone/teslausb
# export REPO=marcone

# Uncomment and change if you want a different branch than main-dev
# export BRANCH=main-dev

````
I got the above log via wifi connection (not AP). I also tried to manually enter in the AP ssid and password as if it is hidden, but still no luck. Any suggestion on where to debug next?

marcone, 2019-09-05:
Does `ifconfig` show the "ap0" interface?
Does `ps aux|grep hostapd` show hostapd is running?
Does `service --status-all` show that the dnsmasq service is running?

skyred, 2019-09-05:
> Does `ifconfig` show the "ap0" interface?

Yes, see below:
```
pi@teslausb:/boot $ ifconfig
ap0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 192.168.66.1  netmask 255.255.255.0  broadcast 192.168.66.255
        ether b8:27:eb:3c:4e:ee  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 244  bytes 15739 (15.3 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 244  bytes 15739 (15.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.2.232  netmask 255.255.255.0  broadcast 192.168.2.255
        inet6 fe80::ba27:ebff:fe3c:4eee  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:3c:4e:ee  txqueuelen 1000  (Ethernet)
        RX packets 2712  bytes 215993 (210.9 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2425  bytes 412739 (403.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
````

> Does `ps aux|grep hostapd` show hostapd is running?
Looks yes
````
pi@teslausb:/boot $ ps aux|grep hostapd
pi        2651  0.0  0.3   4332  1508 pts/0    S+   09:55   0:00 grep --color=auto hostapd
````

> Does `service --status-all` show that the dnsmasq service is running?
Yes, see below
````
pi@teslausb:/boot $ sudo service --status-all
 [ - ]  alsa-utils
 [ + ]  avahi-daemon
 [ + ]  bluetooth
 [ - ]  busybox-klogd
 [ + ]  busybox-syslogd
 [ - ]  console-setup.sh
 [ + ]  cron
 [ + ]  dbus
 [ - ]  dhcpcd
 [ + ]  dnsmasq
 [ + ]  fake-hwclock
 [ + ]  hostapd
 [ - ]  hwclock.sh
 [ - ]  keyboard-setup.sh
 [ + ]  kmod
 [ + ]  networking
 [ - ]  nfs-common
 [ + ]  nmbd
 [ + ]  ntp
 [ - ]  paxctld
 [ - ]  plymouth
 [ - ]  plymouth-log
 [ + ]  procps
 [ + ]  raspi-config
 [ ? ]  rng-tools
 [ - ]  rpcbind
 [ - ]  rsync
 [ + ]  samba
 [ - ]  samba-ad-dc
 [ + ]  smbd
 [ + ]  ssh
 [ - ]  sudo
 [ + ]  udev
````

I tried to look for the AP on my Essential phone, Nexus 6P, and Dell XPS 13. None of them could see it.

marcone, 2019-09-05:
> > Does `ps aux|grep hostapd` show hostapd is running?
> Looks yes
> 
> ```
> pi@teslausb:/boot $ ps aux|grep hostapd
> pi        2651  0.0  0.3   4332  1508 pts/0    S+   09:55   0:00 grep --color=auto hostapd
> ```

Actually unless you forgot to paste a line, it's not running, and what it shows there is just the grep itself.

You could try running it manually:
 `hostapd -dd -P /run/hostapd.ap0.pid /etc/hostapd/hostapd.conf`
and see if that makes the AP appear, or prints any useful error messages.

skyred, 2019-09-05:
Below is the error message:

````
pi@teslausb:~ $ sudo hostapd -dd -P /run/hostapd.ap0.pid /etc/hostapd/hostapd.conf
random: Trying to read entropy from /dev/random
Configuration file: /etc/hostapd/hostapd.conf
ctrl_interface_group=0
rfkill: initial event: idx=0 type=1 op=0 soft=0 hard=0
rfkill: initial event: idx=1 type=2 op=0 soft=0 hard=0
nl80211: Using driver-based roaming
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Using driver-based off-channel TX
nl80211: Supported vendor command: vendor_id=0x1018 subcmd=1
nl80211: Use separate P2P group interface (driver advertised support)
nl80211: Enable multi-channel concurrent (driver advertised support)
nl80211: use P2P_DEVICE support
nl80211: Disable use_monitor with device_ap_sme since no monitor mode support detected
nl80211: interface ap0 in phy phy0
nl80211: Set mode ifindex 3 iftype 3 (AP)
nl80211: Setup AP(ap0) - device_ap_sme=1 use_monitor=0
nl80211: Subscribe to mgmt frames with AP handle 0x1c5df88 (device SME)
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x1c5df88 match=
nl80211: Register frame command failed (type=208): ret=-22 (Invalid argument)
nl80211: Register frame match - hexdump(len=0): [NULL]
nl80211: Could not configure driver mode
nl80211: deinit ifname=ap0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
nl80211: Remove beacon (ifindex=3)
netlink: Operstate: ifindex=3 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211 driver initialization failed.
hostapd_interface_deinit_free(0x1c5acf0)
hostapd_interface_deinit_free: num_bss=1 conf->num_bss=1
hostapd_interface_deinit(0x1c5acf0)
ap0: interface state UNINITIALIZED->DISABLED
hostapd_bss_deinit: deinit bss ap0
ap0: AP-DISABLED 
hostapd_cleanup(hapd=0x1c5b9e0 (ap0))
hostapd_free_hapd_data: Interface ap0 wasn't started
hostapd_interface_deinit_free: driver=(nil) drv_priv=(nil) -> hapd_deinit
hostapd_interface_free(0x1c5acf0)
hostapd_interface_free: free hapd 0x1c5b9e0
hostapd_cleanup_iface(0x1c5acf0)
hostapd_cleanup_iface_partial(0x1c5acf0)
hostapd_cleanup_iface: free iface=0x1c5acf0

````

I am not sure what to do next

marcone, 2019-09-05:
I'm not sure either. One difference between your system and mine is that on mine, hostapd is not running as a service:
```
root@teslausb:~# service --status-all | grep hostapd
 [ - ]  hostapd
```
but it does run as a process:
```
root@teslausb:~# ps aux|grep hostapd
root       330  0.0  0.7   7036  3284 ?        Ss   14:30   0:00 /usr/sbin/hostapd -B -P /run/hostapd.ap0.pid /etc/hostapd/hostapd.conf
root     14899  0.0  0.4   4360  1864 pts/0    S+   19:20   0:00 grep hostapd
```

So maybe you need to stop the hostapd service before running it manually.

skyred, 2019-09-05:
I tried what you suggested, but it seems AP is still disabled:

````
pi@teslausb:~ $ sudo systemctl stop hostapd
pi@teslausb:~ $ service --status-all | grep hostapd
 [ - ]  hostapd
 [ ? ]  rng-tools
pi@teslausb:~ $ sudo hostapd -dd -P /run/hostapd.ap0.pid /etc/hostapd/hostapd.conf
random: Trying to read entropy from /dev/random
Configuration file: /etc/hostapd/hostapd.conf
ctrl_interface_group=0
rfkill: initial event: idx=0 type=1 op=0 soft=0 hard=0
rfkill: initial event: idx=1 type=2 op=0 soft=0 hard=0
nl80211: Using driver-based roaming
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Using driver-based off-channel TX
nl80211: Supported vendor command: vendor_id=0x1018 subcmd=1
nl80211: Use separate P2P group interface (driver advertised support)
nl80211: Enable multi-channel concurrent (driver advertised support)
nl80211: use P2P_DEVICE support
nl80211: Disable use_monitor with device_ap_sme since no monitor mode support detected
nl80211: interface ap0 in phy phy0
nl80211: Set mode ifindex 3 iftype 3 (AP)
nl80211: Setup AP(ap0) - device_ap_sme=1 use_monitor=0
nl80211: Subscribe to mgmt frames with AP handle 0x13c6f88 (device SME)
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x13c6f88 match=
nl80211: Register frame command failed (type=208): ret=-22 (Invalid argument)
nl80211: Register frame match - hexdump(len=0): [NULL]
nl80211: Could not configure driver mode
nl80211: deinit ifname=ap0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
nl80211: Remove beacon (ifindex=3)
netlink: Operstate: ifindex=3 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211 driver initialization failed.
hostapd_interface_deinit_free(0x13c3cf0)
hostapd_interface_deinit_free: num_bss=1 conf->num_bss=1
hostapd_interface_deinit(0x13c3cf0)
ap0: interface state UNINITIALIZED->DISABLED
hostapd_bss_deinit: deinit bss ap0
ap0: AP-DISABLED 
hostapd_cleanup(hapd=0x13c49e0 (ap0))
hostapd_free_hapd_data: Interface ap0 wasn't started
hostapd_interface_deinit_free: driver=(nil) drv_priv=(nil) -> hapd_deinit
hostapd_interface_free(0x13c3cf0)
hostapd_interface_free: free hapd 0x13c49e0
hostapd_cleanup_iface(0x13c3cf0)
hostapd_cleanup_iface_partial(0x13c3cf0)
hostapd_cleanup_iface: free iface=0x13c3cf0

````

marcone, 2019-09-05:
Sorry, I don't know then.

skyred, 2019-09-05:
This points to a kernel bug: https://raspberrypi.stackexchange.com/a/92643

@marcone what's your kernal?

Mine is: 
````
pi@teslausb:~ $ uname -r
4.19.50+
````

skyred, 2019-09-05:
OK, this is a confirmed kernel issue: https://github.com/raspberrypi/firmware/issues/1117

marcone, 2019-09-05:
I have 4.19.66+ (both after upgrading an existing install, and after flashing it from scratch)

skyred, 2019-09-05:
Without updating kernel, I installed hostapd 2.6 (upgraded from 2.4) provided at https://github.com/raspberrypi/firmware/issues/1117#issuecomment-468352313, after the installation, the AP comes up, but the Pi doesn't connect to managed wifi anymore.

Then, I rebooted the Pi, and ran:
````
pi@teslausb:~ $ sudo ifconfig
ap0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 192.168.66.1  netmask 255.255.255.0  broadcast 192.168.66.255
        inet6 fe80::ba27:ebff:fe3c:4eee  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:3c:4e:ee  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 20  bytes 3569 (3.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 104  bytes 7604 (7.4 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 104  bytes 7604 (7.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.2.232  netmask 255.255.255.0  broadcast 192.168.2.255
        inet6 fe80::ba27:ebff:fe3c:4eee  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:3c:4e:ee  txqueuelen 1000  (Ethernet)
        RX packets 392  bytes 30768 (30.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 396  bytes 61567 (60.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
````
````
pi@teslausb:~ $ sudo ps aux|grep hostapd
root       354  0.0  0.4   7036  2076 ?        Ss   23:23   0:00 /usr/sbin/hostapd -B -P /run/hostapd.ap0.pid /etc/hostapd/hostapd.conf
pi        1692  0.0  0.3   4332  1732 pts/0    S+   23:30   0:00 grep --color=auto hostapd
````
````
pi@teslausb:~ $ service --status-all | grep hostapd
 [ - ]  hostapd
 [ ? ]  rng-tools
````

The 3 command outputs seem to be same as yours above now. Are you also running hostapd 2.6?

**However**, after this reboot, I still cannot see the AP on my phones or desktop (The Pi reconnects with managed wifi, so I could get the output above.)

marcone, 2019-09-05:
I have hostapd 2.6.
I'm curious why you had an older kernel and hostapd than me. Did you not start from the prebuilt image? Did the setup process for some reason not update everything to the latest version?

sTevefor3, 2019-09-05:
I did a fresh install on a blank 128 Gb SD card last night, using the same teslausb_setup_variables.conf as my working copy but modded with the new entries for AP.

I see the same results as @skyred ; glad I kept my working copy safe on another SD card.  Yikes!

skyred, 2019-09-05:
> I'm curious why you had an older kernel and hostapd than me.

I was wondering the same. 

> Did you not start from the prebuilt image?

Yes. But, I am living in Shanghai, China. The default Raspberry Pi URL is too slow to finish an update. Therefore, I used the published mirrors from the official site https://github.com/marcone/teslausb/issues/226. I don't think this could cause the problem, could it?

The log didn't show any failed updates.  @marcone Are you sure https://github.com/marcone/teslausb/releases/tag/v1.3 is the latest?

marcone, 2019-09-05:
> Yes. But, I am living in Shanghai, China. The default Raspberry Pi URL is too slow to finish an update. Therefore, I used the published mirrors from the official site #226. I don't think this could cause the problem, could it?

Maybe the Chinese mirrors are out-of-date?

> The log didn't show any failed updates. @marcone Are you sure https://github.com/marcone/teslausb/releases/tag/v1.3 is the latest?

Yes, that's the latest prebuilt. As part of the setup process, it downloads the latest (HEAD revision) scripts from my repo, and runs `apt-get --assume-yes upgrade` as one of the final steps.

skyred, 2019-09-06:
> Maybe the Chinese mirrors are out-of-date? 

Yes, you are right. I just re-enabled the default end points. Then, I can see the difference:

````
pi@teslausb:/etc/apt/sources.list.d $ apt-cache policy hostapd
hostapd:
  Installed: 2:2.6-21~bpo9~rpt1
  Candidate: 2:2.6-21~bpo9~rpt1
  Version table:
 *** 2:2.6-21~bpo9~rpt1 500
        500 http://archive.raspberrypi.org/debian stretch/main armhf Packages
        100 /var/lib/dpkg/status
     2:2.4-1+deb9u4 500
        500 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages
        500 http://mirrors.zju.edu.cn/raspbian/raspbian stretch/main armhf Packages (THIS IS MY MIRROR)
````
But, even this mirror is out of date, that doesn't explain the kernel out of date, or does it?
Anyway, I will find a better mirror and reflush the image.

skyred, 2019-09-06:
I re-flushed my Pi, this time, I added a mirror like this:

`deb http://mirrordirector.raspbian.org/raspbian/ stretch main contrib non-free rpi` in `/etc/apt/sources.list.d/raspi.list`

Then, I got the same (updated) kernel, and hostapd 2.6, and everything works as it advertised.

Thanks so much for the support!!