marcone/teslausb #87: Headless Setup Complete.  Wont Mount Cam/Music in Windows

cadeucsb, 2019-06-20:
Ran through the headless setup, all seemed to work fine.  TESLA_SETUP_FINISHED and WIFI_ENABLED are there.  I am able to SSH without any issue.  The attached log doesnt seem to show any glaring issues.

Plugging in the Pi (via USB labeled port, not PWR) to my PC, it boots and I can SSH in but the cam/music drives dont mount in windows correctly.  4 partitions do mount plugging in the Pi to the PC, but windows immediately asks to format (which I dont).  

Looking in Windows 10 Disk Management, I see the Disk (a 256gb card) showing: a 44MB boot partition, a 1.65GB partition, a 236.7GB partition and a 101MB partition.  

[teslausb-headless-setup.log](https://github.com/marcone/teslausb/files/3311297/teslausb-headless-setup.log)

marcone, 2019-06-20:
The sd card should have 4 partitions following setup, and the sizes you list match what I would expect for a 256GB card, but none of those partitions are ever directly exposed to the attached PC or car through the Pi. They should only be visible when you plug the sd card directly into the car/PC.

cadeucsb, 2019-06-20:
just to make sure I am clear...the headless instructions state:
"Boot it in your Pi, give it a bit, watching for a series of flashes (2, 3, 4, 5) and then a reboot and/or the CAM/music drives to become available on your PC/Mac."

I am not seeing Cam/Music drives after a reboot, just the drives/partitions noted above,  I should see Cam/Music correct?  If so, the log doesnt appear to contain any setup failures and not sure why I dont.

marcone, 2019-06-20:
Yes, you should see the music and cam drives. What's more interesting to me is that you're apparently seeing the actual partitions on the sd card, which should not be possible.
Please run
```
sudo -i
/root/bin/setup-teslausb diagnose
```
to get full diagnostics.

cadeucsb, 2019-06-20:
Randomly, I just popped the card back in the Pi (after using an sd card reader to look at the log posted above) and I got CAM and MUSIC mounted in windows (it even copied all my music properly).  I didnt do anything differently, so no idea why.  

Next ill try in the car and see what happens.

Out of curiosity, how would I export the 'diagnosis' you mentioned to a file?  With all the music copying, there is more than I can assess in the shell.  Not all of my music directories copied and there are a ton of 'couldnt make directory' errors in the diagnose output.  There is plenty of space on the MUSIC volume.

marcone, 2019-06-20:
> Randomly, I just popped the card back in the Pi (after using an sd card reader to look at the log posted above) and I got CAM and MUSIC mounted in windows (it even copied all my music properly). I didnt do anything differently, so no idea why.

That sounds like you didn't actually plug the Pi in to the PC as you said in your first comment, but that instead you took the sd card out of the Pi, and then plugged the sd card into a reader attached to the PC. That certainly explains what you were seeing then: you were seeing the actual partitions of the sd card, not the emulated USB drive exposed by the running teslausb software.

> Next ill try in the car and see what happens.
>
> Out of curiosity, how would I export the 'diagnosis' you mentioned to a file? With all the music copying, there is more than I can assess in the shell. [...]

Just save them to a file. If you're using CIFS backup, and in wi-fi range, the easiest is probably to redirect the output to the CIFS share, e.g. ssh into the Pi, then run:
```
sudo -i
/root/bin/setup-teslausb diagnose > /mnt/archive/diagnostics.txt
```
then go find `diagnostics.txt` on your file server.

cadeucsb, 2019-06-20:
> That sounds like you didn't actually plug the Pi in to the PC as you said in your first comment, but that instead you took the sd card out of the Pi, and then plugged the sd card into a reader [...]

You are totally correct on that, my fault.  The partitions are what I was seeing when plugged in via the reader (their sizes just didnt make a ton of sense and is why i questioned it).  When just the Pi plugged in, nothing was mounting at all.  I even walked through the One Step two full times from scratch and nothing.  Appears to be mounted now.

Now I just need to figure out why it stopped copying my 100GB of music over when there is plenty of space.  It looks like the copy functions just moves through the folder structure alphabetically and randomly cut off as one folder is incomplete and all folders after it are non-existent.

cadeucsb, 2019-06-20:
Diagnostics attached.  Around line 1156, it just starts erroring on the music copy.  It gets to about the M's of my Rap directory then cant finish that directory or any directory after it.  All directories before it copied without issue.

If nothing is apparent, ill just see it resolves the copying after a couple reboots and testing in the car.

[diagnostics.txt](https://github.com/marcone/teslausb/files/3312183/diagnostics.txt)

marcone, 2019-06-20:
Looks like it encountered a filesystem error and remounted the music disk read-only:
```
8755 [   95.854413] FAT-fs (loop0): error, corrupted directory (invalid entries)
8756 [   95.854428] FAT-fs (loop0): Filesystem has been set read-only
```
You can try rebooting the Pi, which will end up doing a filesystem check/repair before resuming the copy, and see if it continues where it left off.

cadeucsb, 2019-06-20:
After a sudo reboot, i can access via SSH and export diagnostics (attached), but again, no MUSIC or CAM drives mounted in windows.
[diagnostics2.txt](https://github.com/marcone/teslausb/files/3312233/diagnostics2.txt)

marcone, 2019-06-20:
At the time of the diagnostics, it was still busy copying music to your music drive.
Since the music drive is mounted on the Pi at that point, it can't also be mounted on the PC during that time.

cadeucsb, 2019-06-20:
That's probably the issue and the drives finally mounted last time after the filesystem errored and the copy ceased.  

How are you able to tell the copy was still in progress?  on line 8491, I see:

"Thu 20 Jun 13:56:41 MDT 2019: Copied 0 music file(s), skipped 10439 previously-copied files, encountered 7337 errors."

So I assumed it was done.

marcone, 2019-06-20:
Read further. That particular section of the diagnostics ends at line 8567, with
 `Thu 20 Jun 15:45:27 MDT 2019: Copying music from  archive...`

cadeucsb, 2019-06-20:
appears that is still the status now.  I will let it go and see what happens.  thanks for the assistance.

marcone, 2019-06-20:
BTW you can kinda tell what it's doing by looking at the LED. Flashing rapidly means it's archiving or copying music. When it's flashing slowly, the drives should be connected to the car/PC, with a double-blink for "connected to wifi" and a single blink for "not connected to wifi".

closed 2019-06-20 by marcone

miles267, 2019-10-05:
> /root/bin/setup-teslausb diagnose > /mnt/archive/diagnostics.txt

@marcone when I run the command above, it returns:
-bash: /mnt/archive/diagnostics.txt: Read-only file system

However if I run mount /mnt/archive, it works without any error or issues and I can then write to the /mnt/archive (my CIFS file share).