marcone/teslausb #89: STOP: The archive couldn't be mounted with CIFS version 3.0

Subroc919, 2019-06-21:
This is showing up in my teslausb-headles-setup.log file while running the first time set up. Suggested fix is to change CIFS version but I've tried 2.1, 2.0, 1.0 with no luck.

I can mount the drive on a Mac using the address and credentials I configured so I'm not sure what's going on.

wwwb0n3zcom, 2019-06-21:
Please post how you enabled CIFS exactly (copy and paste what's in your teslausb_setup_variables.conf excluding the personal info). I had this same problem with 3.0 but mine works perfectly fine with 2.0. I made the mistake the first time which is why I'm asking you to copy paste.

wwwb0n3zcom, 2019-06-21:
You can also try the command line via SSH to test:

sudo -i
/root/bin/remountfs_rw
mount -t cifs //server_ip/tesla_share /tmp/archivetestmount -o vers=2.0,credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777

Subroc919, 2019-06-21:
My config was as follows, changing the version from 3.0 to 2.1 etc:
```
# Default variables for CIFS (Windows/Mac file sharing) setup
export ARCHIVE_SYSTEM=cifs
export archiveserver=C2300-Share
export sharename=Volume_A1
export shareuser=admin
export sharepassword=admin
# export cifs_version="3.0"
export camsize=100%
```

marcone, 2019-06-21:
Assuming that your password is not actually "admin", check that it's properly escaped, if needed.
If that's not it, I'd try mounting manually like @wwwb0n3zcom showed above, so it's easier to iterate over the various parameters and maybe get more helpful error messages.

Subroc919, 2019-06-21:
I get an error saying "unable to find suitable address." When I try that...

marcone, 2019-06-21:
You did replace "server_ip/tesla_share" in that command line with the correct names, right?
I.e. based the config you listed earlier:
```
mount -t cifs //C2300-Share/Volume_A1 /tmp/archivetestmount -o vers=2.0,credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777
```
if so, then maybe the hostname you specified doesn't actually resolve to an IP address?

Subroc919, 2019-06-21:
Yeah, I even changed the name of the server through my router to see if that was causing it.

Is there any way to verify that I have the right names for the server and share?

marcone, 2019-06-21:
Since you said it worked from a Mac using the same names and credentials, it seems like already have them. If there's a problem with the name not resolving, you can always try using the IP address instead of the name.

Subroc919, 2019-06-22:
Yeah, I'm about ready to throw in the towel. Seems I don't really have the technical chops to get this working.

Thanks for your help anyway!

closed 2019-06-22 by marcone

Jonkheer73, 2019-07-29:
Hi,
First the installation script would not complete and was stuck at step 2.
I found out that it was trying to connect to my Synology Nas with DSM 6.2 but failed with cifs 3.0.
As long that it could not connect it will not proceed.
Synology DSM 6.2 does not support cifs 3.0 of 2.1.
When I added export cifs_version="2.0" at the beginning of the install script I succesfully finished the installation.

jhaltd, 2019-11-25:
I couldn't get it to work either, until I added the following line to my /etc/fstab:

//dsmserver/TeslaCam  /mnt/archive  cifs   credentials=/root/.teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,vers=2.0,noperm 0 0

Added a line to my /etc/hosts (I tried using the IP in the FSTAB and it wouldn't work, so I used the machine name and forced the name resolution by putting the entry in the hosts file):

192.168.100.45 dsmserver

And, of course, created the /root/.teslaCamArchiveCredentials file:

username=my_user
password=my_password

Now it's syncing away with my Synology Rackstation NAS (DSM 6.2.2)

Dunc