marcone/teslausb #93: NTP sync?

Jonnehs, 2019-06-22:
Works great, excellent job! At least in my tests, M3 still not here yet (UK)

Is there a default NTP setting I can turn on, or should that be part of the script? I just noticed that although it sets the timezone correctly my flashed image thinks its november 2016.

I can supply NTP at its default gateway when its on the home wifi, so I guess it just needs a variable in the setup conf to run the command? (if there is one!)

kdvlr, 2019-06-22:
I added this line to my script for the same issue. Can you add this to /etc/rc.local before the last line (before "exit 0") , reboot, and let me know if it is working? I can then submit an upstream fix

```

# Update time using ntp
if nc -zw1 google.com 443
then
  ntpdate -u pool.ntp.org
fi
```

Jonnehs, 2019-06-22:
not sure ntpdate works because of the lack of network at boot, just shooting in the dark here but no it didnt work

marcone, 2019-06-22:
ntp is installed and enabled by default:
```
root@teslausb:~# service --status-all
 [ - ]  alsa-utils
 [ + ]  avahi-daemon
 [ + ]  bluetooth
 [ - ]  busybox-klogd
 [ + ]  busybox-syslogd
 [ - ]  console-setup.sh
 [ + ]  cron
 [ + ]  dbus
 [ + ]  dhcpcd
 [ + ]  fake-hwclock
 [ - ]  hwclock.sh
 [ - ]  keyboard-setup.sh
 [ + ]  kmod
 [ + ]  networking
 [ - ]  nfs-common
 [ + ]  nmbd
 [ + ]  ntp
 [ - ]  paxctld
 [ - ]  plymouth
 [ - ]  plymouth-log
 [ + ]  procps
 [ + ]  raspi-config
 [ ? ]  rng-tools
 [ - ]  rpcbind
 [ - ]  rsync
 [ + ]  samba
 [ - ]  samba-ad-dc
 [ + ]  smbd
 [ + ]  ssh
 [ - ]  sudo
 [ + ]  udev
```
The only problem is that since the pi has no battery backed hardware clock, if you power it from the car's USB, and drive out of wifi range before it fully boots, then ntp won't be able to set the time/date.

Jonnehs, 2019-06-22:
it didnt work for me until i did 2 things

sudo timedatectl set-ntp True

and then

nano /etc/systemd/timesyncd.conf

added my ntp server to NTP= (which was blank)

Jonnehs, 2019-06-22:
root@teslausb:~# ntpq -pn
     remote           refid      st t when poll reach   delay   offset  jitter
==============================================================================
 0.debian.pool.n .POOL.          16 p    -   64    0    0.000    0.000   0.004
 1.debian.pool.n .POOL.          16 p    -   64    0    0.000    0.000   0.004
 2.debian.pool.n .POOL.          16 p    -   64    0    0.000    0.000   0.004
 3.debian.pool.n .POOL.          16 p    -   64    0    0.000    0.000   0.004

Dunno why but the default pools didnt work until i did that

pi@teslausb:~ $ ntpq -pn
     remote           refid      st t when poll reach   delay   offset  jitter
==============================================================================
*10.10.50.1      208.91.112.50    3 u  105  128  377    5.127   -0.239   0.212
 209.115.181.102 .STEP.          16 u    - 1024    0    0.000    0.000   0.000
pi@teslausb:~ $

Internal ntp worked fine first time

marcone, 2019-06-22:
Did you flash the prebuilt teslausb image, or plain stretch lite?
With the prebuilt teslausb image, I've never had to do any of things you list above, it just works. Or at least, it's getting the correct time from *somewhere*, despite the `NTP=` line being blank (well, commented out) too in my /etc/systemd/timesyncd.conf.

Jonnehs, 2019-06-22:
yeah I flashed the pre-build. No idea, strange one. Maybe the default NTP pools dont work for me for some reason. I wanted to change it to my internal anyway so google-fu has worked out how to do that now.

kdvlr, 2019-06-22:
For me, both the stretch lite and the prebuilt image did not work. In addition I also had to download ntpdate which allowed me to use a non-standard port. This is  because for me AT&T in the US blocks the default ntp port.

marcone, 2019-06-22:
Great... I'm getting AT&T internet next week...

marcone, 2019-06-22:
So as far as NTP being enabled by default. On my pi, `/var/log/boot.log` shows this:
```
Starting LSB: Start NTP daemon...
```
and `service --status-all` shows the ntp service is active. Is that not the case for you? Or is the ntp service active, but not working because of port blockage or other problems?

marcone, 2019-06-23:
Also, `logread | grep ntp` shows ntpd's logs.

Jonnehs, 2019-06-23:
I'll reflash when i get my VIN to reconfigure it and let you know what the out of the box behavior is.

Its not a network issue though for me, my PC can query the debian pools just fine;

C:\Users\jonne>w32tm /stripchart /computer:0.debian.pool.ntp.org /dataonly /samples:5
Tracking 0.debian.pool.ntp.org [149.56.121.16:123].
Collecting 5 samples.
The current time is 23/06/2019 10:18:35.
10:18:35, +15.2661167s
10:18:37, +15.2641327s
10:18:39, +15.2658896s
10:18:41, +15.2649678s
10:18:43, +15.2663188s

hard to say what the problem was originally for me now because it just works with my local NTP

kdvlr, 2019-06-24:
This is my ntp update log on AT&T
```

Jun 23 18:54:04 teslausb daemon.info ntpd[595]: Soliciting pool server 2001:418:3ff::1:53
Jun 23 18:54:24 teslausb daemon.info ntpd[595]: Soliciting pool server 198.46.223.227
Jun 23 18:54:33 teslausb daemon.info ntpd[595]: Soliciting pool server 98.152.165.38
Jun 23 18:54:35 teslausb daemon.info ntpd[595]: Soliciting pool server 23.252.63.82
Jun 23 18:55:10 teslausb daemon.info ntpd[595]: Soliciting pool server 163.237.218.18
Jun 23 18:55:31 teslausb daemon.info ntpd[595]: Soliciting pool server 12.167.151.1
Jun 23 18:55:37 teslausb daemon.info ntpd[595]: Soliciting pool server 72.5.72.15
Jun 23 18:55:42 teslausb daemon.info ntpd[595]: Soliciting pool server 208.67.72.50
Jun 23 18:56:16 teslausb daemon.info ntpd[595]: Soliciting pool server 64.79.100.197
Jun 23 18:56:35 teslausb daemon.info ntpd[595]: Soliciting pool server 195.21.137.209
Jun 23 18:56:42 teslausb daemon.info ntpd[595]: Soliciting pool server 66.228.59.187
Jun 23 18:56:43 teslausb daemon.info ntpd[595]: Soliciting pool server 45.76.244.202
Jun 23 18:56:44 teslausb daemon.info ntpd[595]: Soliciting pool server 138.197.114.122
Jun 23 18:56:45 teslausb daemon.info ntpd[595]: Soliciting pool server 173.230.144.109
Jun 23 18:56:46 teslausb daemon.info ntpd[595]: Soliciting pool server 159.203.82.102
Jun 23 18:56:46 teslausb daemon.info ntpd[595]: Soliciting pool server 204.11.201.12
Jun 23 18:56:47 teslausb daemon.info ntpd[595]: Soliciting pool server 69.164.213.136
Jun 23 18:56:48 teslausb daemon.info ntpd[595]: Soliciting pool server 66.96.98.9
```

[Click here for](https://about.att.com/sites/broadband/network) the link to AT&T's broadband practices page where they explain why they block NTP. Search for NTP

bahree, 2019-06-26:
> I added this line to my script for the same issue. Can you add this to /etc/rc.local before the last line (before "exit 0") , reboot, and let me know if it is working? I can then submit an upstream fix
> 
> ```
> 
> # Update time using ntp
> if nc -zw1 google.com 443
> then
>   ntpdate -u pool.ntp.org
> fi
> ```

I was having the same issue, and after the first time it automatically ran setup-teslausb selfupdate, the date changed back to Nov 2016 and no matter what I tried did not fix it. But adding this to rc.local did fix it for me and is now the right date! 

And I used the latest official image (re-downloaded and re-flashed it again just to make sure). Here is what I saw if that helps narrow/debug this: The first time I ssh'd in the date was correct. When I ran setup-teslausb selfupdate, I got an error saying already running (this was already running - so something must run this automatically). And after rebooting then the date when from correct to Nov something in 2016. NTP is installed and doesn't seem to make any difference in my case.

marcone, 2019-06-26:
> When I ran setup-teslausb selfupdate, I got an error saying already running (this was already running - so something must run this automatically).

If setup-teslausb told you it was already running, then you probably ran it while the initial setup was still in progress. This is exactly why I added that message, to avoid two instances of the script running at the same time and stepping on each other's toes.

bahree, 2019-06-26:
OK that is good to know. But the date thing is still weird and somewhere between the first time booting up, and the selfupdate running, something is changing that. Not sure what. I will keep an eye on this for the next few days and report back here. Right now this is at home where I have mesh network covering the whole house, garage, etc. so network connectivity isn't an issue. Would be different when driving around and at work, etc.

bahree, 2019-06-29:
OK that didn't seem to fix it - it goes back to 2016 - see below. :(

`pi@teslausb-Model3:~ $ date
Thu  3 Nov 12:16:42 PDT 2016
pi@teslausb-Model3:~ $`

However when I reboot it and then check, then the date time is right. So something is causing the time to shift, but I don't know enough to figure out what or where. Here is the output post reboot (not too useful).
`pi@teslausb-Model3:~ $ date
Fri 28 Jun 20:42:18 PDT 2019
pi@teslausb-Model3:~ $`

marcone, 2019-06-29:
So at what point did you observe the 2016 date?
The Pi doesn't have a "real" clock, so the clock will reset back to 2016 on every boot until NTP corrects it, but in my experience it will have been updated by the time I'm able to ssh in.

ctwomey, 2019-06-29:
I am using Google drive for backup and it is failing because the date is too far off - the google cert is being rejected for "not yet" being valid - and because the pi can often start up when offline and only connect when I get home, putting the call to ntpdate in rc.local isn't going to help me. In the end I got it working by putting @bahree's snippet above into a function in /root/bin/archiveloop, to set the time as soon as the archive is reachable.

marcone, 2019-07-03:
@kdvlr for what it's worth, I'm on AT&T fiber now and NTP still works for me. From what I read though, AT&T only blocks NTP when the source port is port 123, which I expect won't normally be the case due to NAT.

@ctwomey that's a good point. I've noticed in my own logs that the time doesn't get set until it already started archiving, so forcing the time beforehand seems like a good idea.

kdvlr, 2019-07-03:
I agree with @ctwomey’s idea. 

On At&T, it started after they replaced my gateway to 5268AC. But none of my devices Windows and Linux are able to sync with NTP. Mac seems to work fine. 

Ntpdate -u uses non standard ports so that’s what I use. Looks like you also try to set the time zone automatically using tzupdate. That’s pretty neat, but still doesn’t work for me :-( .

marcone, 2019-07-03:
tzupdate is only used when you set the timezone to "auto", and only at install time, so honestly it's not a super useful feature. These days I just set the timezone to "America/Los_Angeles" in my .conf file.

I have a local change that sets the time at the appropriate places in archiveloop, but it uses ntpd (since ntpdate isn't installed by default, and the manpage for ntpd implies it's going away), and ntpd doesn't appear to have an option to use a nonprivileged port. 
(are you using the 5268AC as the main router? I have a DGW210, but I use it in passthrough mode with my real router)

kdvlr, 2019-07-04:
I do have a router behind the 5268 and that is part of the problem. If I use 5268 as the main router, then NTP works fine. My router does not forward to a different port which is part of the problem. Also, this seems to be a problem with 5268AC and not the other RGs. An old thread here - 

https://forums.att.com/t5/AT-T-Fiber-Equipment/Pace-5268AC-in-DMZplus-blocks-UDP-123-NTP/td-p/5014592/page/4 

I do think I am in the minority of users, and ntpd will work for the vast majority of users.

Jonnehs, 2019-07-11:
Got my VIN showing up in tesla.com source code today (yay!) so I'll reflash tomorrow and go through this again. Seems to have kicked off some good discussion anyway.

Jonnehs, 2019-07-12:
without changes i get this in the ntp logs;

pi@DraigoUSB:~ $ date
Thu  3 Nov 17:17:20 GMT 2016
pi@DraigoUSB:~ $ logread | grep ntp
Nov  3 17:17:08 DraigoUSB daemon.info ntpd[547]: proto: precision = 2.000 usec (-19)
Nov  3 17:17:08 DraigoUSB daemon.info ntpd[547]: Listen and drop on 0 v6wildcard [::]:123
Nov  3 17:17:08 DraigoUSB daemon.info ntpd[547]: Listen and drop on 1 v4wildcard 0.0.0.0:123
Nov  3 17:17:08 DraigoUSB daemon.info ntpd[547]: Listen normally on 2 lo 127.0.0.1:123
Nov  3 17:17:08 DraigoUSB daemon.info ntpd[547]: Listen normally on 3 wlan0 10.10.50.200:123
Nov  3 17:17:08 DraigoUSB daemon.info ntpd[547]: Listen normally on 4 lo [::1]:123
Nov  3 17:17:08 DraigoUSB daemon.info ntpd[547]: Listen normally on 5 wlan0 [fe80::5d42:c2b4:a1d0:7143%2]:123
Nov  3 17:17:08 DraigoUSB daemon.info ntpd[547]: Listening on routing socket on fd #22 for interface updates
Nov  3 17:17:10 DraigoUSB daemon.info ntpd[547]: Soliciting pool server 35.183.57.169
Nov  3 17:17:10 DraigoUSB daemon.info ntpd[547]: Soliciting pool server 138.197.135.225
Nov  3 17:17:11 DraigoUSB daemon.info ntpd[547]: Soliciting pool server 192.99.2.8
Nov  3 17:17:12 DraigoUSB daemon.info ntpd[547]: Soliciting pool server 129.128.12.20
Nov  3 17:17:35 DraigoUSB daemon.notice ntpd[641]: ntpd 4.2.8p10@1.3728-o Sat Mar 10 18:03:33 UTC 2018 (1): Starting
Nov  3 17:17:35 DraigoUSB daemon.info ntpd[641]: Command line: ntpd -q -x -g
Nov  3 17:17:35 DraigoUSB daemon.info ntpd[641]: proto: precision = 3.000 usec (-18)
Nov  3 17:17:35 DraigoUSB daemon.err ntpd[641]: unable to bind to wildcard address :: - another process may be running - EXITING

Jonnehs, 2019-07-12:
I just remove the debian pools from /etc/ntp.conf and change it to one that works. 

No idea why the debian ones don't work for my pi, they work from my PC

withinfocus, 2019-07-15:
For what it's worth, I attempted to set up this project using the latest Raspbian Buster and encountered a number of issues with NTP. I ended up just pulling the last Stretch and using that -- no issues with time sync. Not sure of the root cause.

sebnaf, 2020-04-06:
This might be related to this issue as well:
https://raspberrypi.stackexchange.com/questions/8231/how-to-force-ntpd-to-update-date-time-after-boot

`ntpd emits IP/UDP packets with the ToS field set to 0xc0. So that is quite similar to the other issue I (and many others) have with Raspberry Pi3 when operated over a Wi-Fi internal interface.

As a workaround, add the command

/sbin/iptables -t mangle -I POSTROUTING 1 -o wlan0 -p udp --dport 123 -j TOS --set-tos 0x00
to the file

/etc/rc.local
before the exit 0 line. Next, reboot and check.`

1activegeek, 2020-04-26:
I know this issue appears closed, but it's the only one I can find that is picking up on the NTP issues I'm facing. Will look to use some of hte fixes here, but my Pi Zero W is having issues with time consistently. It appears as though servers can't be resolved, but I can resolve them fine from my desktop. Not sure if this is a regression, or something else. Relevant logs below. The time drift seems to get longer and longer for everytime I disconnect it. Looking at roughly 20 minutes now. 
```
r 26 15:45:23 teslausb-model3 daemon.notice ntpd[403]: ntpd 4.2.8p12@1.3728-o (1): Starting
Apr 26 15:45:23 teslausb-model3 daemon.info ntpd[403]: Command line: /usr/sbin/ntpd -p /var/run/ntpd.pid -g -u 109:116
Apr 26 15:45:23 teslausb-model3 daemon.info ntpd[413]: proto: precision = 2.000 usec (-19)
Apr 26 15:45:23 teslausb-model3 daemon.notice ntpd[413]: leapsecond file ('/usr/share/zoneinfo/leap-seconds.list'): good hash signature
Apr 26 15:45:23 teslausb-model3 daemon.notice ntpd[413]: leapsecond file ('/usr/share/zoneinfo/leap-seconds.list'): loaded, expire=2020-06-28T00:00:00Z last=2017-01-01T00:00:00Z ofs=37
Apr 26 15:45:23 teslausb-model3 daemon.info ntpd[413]: Listen and drop on 0 v6wildcard [::]:123
Apr 26 15:45:23 teslausb-model3 daemon.info ntpd[413]: Listen and drop on 1 v4wildcard 0.0.0.0:123
Apr 26 15:45:23 teslausb-model3 daemon.info ntpd[413]: Listen normally on 2 lo 127.0.0.1:123
Apr 26 15:45:23 teslausb-model3 daemon.info ntpd[413]: Listen normally on 3 wlan0 10.0.14.55:123
Apr 26 15:45:23 teslausb-model3 daemon.info ntpd[413]: Listen normally on 4 lo [::1]:123
Apr 26 15:45:23 teslausb-model3 daemon.info ntpd[413]: Listen normally on 5 wlan0 [fe80::redact]:123
Apr 26 15:45:23 teslausb-model3 daemon.info ntpd[413]: Listening on routing socket on fd #22 for interface updates
Apr 26 15:45:23 teslausb-model3 daemon.info ntpd[413]: kernel reports TIME_ERROR: 0x41: Clock Unsynchronized
Apr 26 15:45:23 teslausb-model3 daemon.info ntpd[413]: kernel reports TIME_ERROR: 0x41: Clock Unsynchronized
Apr 26 15:45:24 teslausb-model3 daemon.err ntpd[413]: error resolving pool 0.debian.pool.ntp.org: Temporary failure in name resolution (-3)
Apr 26 15:45:25 teslausb-model3 daemon.err ntpd[413]: error resolving pool 1.debian.pool.ntp.org: Temporary failure in name resolution (-3)
Apr 26 15:45:26 teslausb-model3 daemon.err ntpd[413]: error resolving pool 2.debian.pool.ntp.org: Temporary failure in name resolution (-3)
Apr 26 15:45:27 teslausb-model3 daemon.err ntpd[413]: error resolving pool 3.debian.pool.ntp.org: Temporary failure in name resolution (-3)
```

marcone, 2020-04-26:
@1activegeek looks like some kind of DNS issue. What does `resolvconf -l` say?

1activegeek, 2020-04-26:
There are no DNS resolution issues I can find. Attempts at pinging the pool address comes back with an IP. And the resolvconf is responding correctly:
```

pi@teslausb-model3:~ $ resolvconf -l
# resolv.conf from wlan0.dhcp
# Generated by dhcpcd from wlan0.dhcp
domain my.internal.dom
search my.internal.dom
nameserver 10.0.10.1
nameserver 1.1.1.1
nameserver 8.8.8.8
```

sebnaf, 2020-04-26:
There are some known issues with WLAN and NTP on the Zero W.

/sbin/iptables -t mangle -I POSTROUTING 1 -o wlan0 -p udp --dport 123 -j TOS --set-tos 0x00

Fixed it for me.

1activegeek, 2020-04-26:
While I was hopeful, that did not resolve this either. I was able to use some of the commands on that link you posted above though to get it to at least run an update manually. Upon disconnecting though and plugging back into power, it again fails to keep the time and shows the time error on boot on the `logread` output.

marcone, 2020-04-26:
For what it's worth, I see the same "error resolving pool 0.debian.pool.ntp.org: Temporary failure in name resolution (-3)" messages in my own `logread` output, however it doesn't seem to prevent the Pi from getting the correct time.

1activegeek, 2020-04-27:
So if that is the case, is there something else I should be looking into? It doesn't quite seem to make sense as to why I would see this scenario happening. I'm at home, connected to solid power and easily within ranges of wifi. DNS resolution is working fine. And when using commands such as:
```
sudo /etc/init.d/ntp stop
sudo ntpd -q -g
sudo /etc/init.d/ntp start
```
It will successfully set the time to the current time.

marcone, 2020-04-27:
> So if that is the case, is there something else I should be looking into? It doesn't quite seem to make sense as to why I would see this scenario happening. I'm at home, connected to solid power and [...]

Weird, that is pretty much exactly what teslausb does before it starts archiving.
I'm sorry, I don't know why it's not working for you.

sebnaf, 2020-04-27:
I am experiencing this issue as well now.
A manual ntp update runs fast.

Maybe you want to switch to chrony?

I`ll try to get some more logs over the next few days.

1activegeek, 2020-04-28:
So just an update, there seems to be something odd at play here. I decided to start from scratch thinking something was wrong in setup. Fresh image now, and at first it seems to be working ok. Just went to plug it into the car now, which still has ample Wifi coverage, and it's drifting more now. It's almost as though it will get within reason at some point during setup, but as it gets unplugged and/or loses real clock cycles, it will drift further and further but not recover.

marcone, 2020-04-28:
If you trigger an archive-cycle using `touch /tmp/archive_is_unreachable`, does time get corrected?

sebnaf, 2020-04-28:
No, it times out in this case. 
I do have screencast with all logs in case you are interested in.

1activegeek, 2020-05-01:
It looks like mine does, but it takes a little bit. I tried to finally run it again for some updates, and it looks like trying to re-configure wasn't working because the time was so far off:
```
Reading package lists... Done                              
E: Release file for http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease is not valid yet (invalid for another 3h 28min 44s). Updates for this repository will not be applied.
Failed, retrying
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
^Cading package lists... 88%                               
root@teslausb-model3:~# touch /tmp/archive_is_unreachable
root@teslausb-model3:~# date
Thu 30 Apr 15:25:37 EDT 2020
root@teslausb-model3:~# date
Thu 30 Apr 15:25:41 EDT 2020
root@teslausb-model3:~# date
Thu 30 Apr 15:25:45 EDT 2020
root@teslausb-model3:~# date
Thu 30 Apr 15:25:49 EDT 2020
root@teslausb-model3:~# date
Thu 30 Apr 15:25:53 EDT 2020
root@teslausb-model3:~# date
Thu 30 Apr 15:26:05 EDT 2020
root@teslausb-model3:~# date
Thu 30 Apr 15:26:08 EDT 2020
root@teslausb-model3:~# /root/bin/setup-teslausb ^C
root@teslausb-model3:~# date
Fri  1 May 19:17:08 EDT 2020
```