marcone/teslausb #100: Failure to go back to USB Drive mode

gkwok1, 2019-06-24:
I have a 2018 Model X with a custom wired in always on 12v connector inside the center console connected to the TeslaUSB. Every time I leave my house after the Pi connects to my wifi, I have to unplug and plug the USB back in for the dashcam icon to reappear.

Any thoughts on this?

kdvlr, 2019-06-24:
Can you follow this - https://github.com/marcone/teslausb/issues/35 and put the logs here? One of my PR was to specifically address this issue. I don't see that anymore on my side after that.

gkwok1, 2019-06-24:
Attached is the log you requested. I hope this helps.

[diagnostics.txt](https://github.com/marcone/teslausb/files/3322362/diagnostics.txt)

kdvlr, 2019-06-24:
That's interesting. It just looks like the Tesla is refusing to recognize the drive. You could try doing a 
```
ls -l /sys/devices/platform/soc/20980000.usb/gadget/
```
before the drive is attached but the dashcam icon does not appear and then once again after you have disconnected and reconnected the raspberry pi. I assume the pi does not restart when you do that (because you have it powered by 12v). 

Based on the log you have attached, I currently have no idea as to what's happening

marcone, 2019-06-24:
The diagnostics show that /backingfiles/cam_disk.bin is connected to lun0.
However, your dmesg seems wrong somehow:
```
[  529.074593] dwc2 20980000.usb: new device is high-speed
[  529.147457] dwc2 20980000.usb: new device is high-speed
[  529.205676] dwc2 20980000.usb: new address 16
[  529.218566] g_mass_storage gadget: high-speed config #1: Linux File-Backed Storage
[ 6093.312221] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 6096.077517] CIFS VFS: Server 172.30.1.2 has not responded in 120 seconds. Reconnecting...
[ 6115.552656] FAT-fs (loop0): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[11654.057276] Mass Storage Function, version: 2009/09/11
[11654.057295] LUN: removable file: (no medium)
[11654.057462] LUN: removable file: /backingfiles/cam_disk.bin
[11654.057471] Number of LUNs=1
[11654.068568] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/11
[11654.068588] g_mass_storage gadget: g_mass_storage ready
[11654.068603] dwc2 20980000.usb: bound driver g_mass_storage
[11654.419852] dwc2 20980000.usb: new device is full-speed
[11739.834677] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
```
the lines at timestamp 529.x are what I normally see when the car or a PC connects to the Pi's emulated drive.  In your dmesg you see some of those same lines again at timestamp 11654, but crucially not that final "g_mass_storage gadget: high-speed config #1: Linux File-Backed Storage" line. So while the Pi is exporting the emulated USB drive, it appears the car isn't actually connected to the Pi's USB port for some reason.

gkwok1, 2019-06-26:
> That's interesting. It just looks like the Tesla is refusing to recognize the drive. You could try doing a
> 
> ```
> ls -l /sys/devices/platform/soc/20980000.usb/gadget/
> ```
> 
> before the drive is attached but the dashcam icon does not appear and then once again after you have disconnected and reconnected the raspberry pi. I assume the pi does not restart when you do that (because you have it powered by 12v).
> 
> Based on the log you have attached, I currently don't know why it cannot see the drive.

I left my house right around 11:34am and got to my destination at 11:49am where I am not connected to my home wifi and SSHed into the Pi.

Here is the output before I disconnected:
```
root@teslausb:~# ls -l /sys/devices/platform/soc/20980000.usb/gadget/
total 0
drwxr-xr-x 3 root root    0 Jun 26 11:34 lun0
drwxr-xr-x 2 root root    0 Jun 26 11:49 power
-r--r--r-- 1 root root 4096 Jun 26 11:49 suspended
-rw-r--r-- 1 root root 4096 Jun 26 11:49 uevent
```

Here is the output after I disconnected and reconnected:
```
root@teslausb:~# ls -l /sys/devices/platform/soc/20980000.usb/gadget/
total 0
drwxr-xr-x 3 root root    0 Jun 26 11:49 lun0
drwxr-xr-x 2 root root    0 Jun 26 11:49 power
-r--r--r-- 1 root root 4096 Jun 26 11:49 suspended
-rw-r--r-- 1 root root 4096 Jun 26 11:49 uevent
```

Thanks.

boom-mug, 2019-06-27:
I just wanted to let you know that this is happening to me too to some extent. (Car not seeing the share after uploading)  Right now I've just been observing the behavior and it looks like sometimes it is able to recover after the car sits in the garage for 10-12 hours.  It is definitely fine no matter what until I get near my house wifi.

This makes me think (at least before reading all of this) it's when I drive off in the middle of it connected and uploading.  I also don't have a great 2.4Ghz where the car is parked.

kdvlr, 2019-06-29:
It's a known issue that it doesn't get reconnected if it loses wifi while archiving files. My PR doesn't work for me 100% of the time for CIFS. 

@gkwok1 - You mentioned that you are powering the pi from a 12v constant power, so next time, verify that lun0 is mounted and if the dashcam icon still does not show, reset your tesla and see if it mounts this time. Since the pi should not restart, this will confirm the issue is with Tesla trying to mount the drive. You can also see if you can restart the pi *while* connected to the USB to see if it mounts after a restart without requiring you to unplug and plug.

gkwok1, 2019-06-30:
> It's a known issue that it doesn't get reconnected if it loses wifi while archiving files. My PR doesn't work for me 100% of the time for CIFS more like about 80% of the time it works
> 
> @gkwok1 - You mentioned that you are powering the pi from a 12v constant power, so next time, verify that lun0 is mounted and if the dashcam icon still does not show, reset your tesla and see if it mounts this time. Since the pi should not restart, this will confirm the issue is with Tesla trying to mount the drive. You can also see if you can restart the pi _while_ connected to the USB to see if it mounts after a restart without requiring you to unplug and plug.

@kdvlr I have tried a few things.
1) Rebooted car with dash cam icon not showing - Results: Still no dash cam after reboot
2) Rebooted car with dash cam icon showing  - Results:  no dash cam after reboot (lost dash cam icon)
3) Rebooted pi with USB connected and no dash cam icon  - Results: dash cam icon reappears
4) Rebooted pi with USB connected and dash cam icon  - Results: dash cam icon appears

The issue might not be connected to it losing wifi while archiving since it happens to me first thing in the morning after the car has been asleep for 10+ hours and I have good reception in the garage for the pi.

marcone, 2019-07-02:
@gkwok1 since you said you're using permanent power, could you be running into the same problem as described in https://github.com/marcone/teslausb/issues/110#issuecomment-507429967 ?

gkwok1, 2019-07-02:
> @gkwok1 since you said you're using permanent power, could you be running into the same problem as described in [#110 (comment)](https://github.com/marcone/teslausb/issues/110#issuecomment-507429967) ?

I created a OTG cable per #110 and after rebooting the car, the dash icon is still there. I will conduct a few more test tomorrow and report back but it's promising.

For the record, I have a MCU 2 Model X with 2019.20.4.2.

gkwok1, 2019-07-03:
My issue seems to be resolved now by creating a data only cable. Thanks everyone for all their help.

closed 2019-07-03 by gkwok1