marcone/teslausb #98: Is it possible to use Tesla token?

bahree, 2019-06-24:
I am wondering if it is possible to use a Tesla auth token instead of the login credentials as part of the configuration? Also if I add those later (after the initial setup) - do I run the setup script again? Any suggestions? Thanks.

raushel, 2019-06-24:
I believe /root/bin/tesla_api.py checks the .conf file every time if there isn’t already a generated token. You could probably just manually create the token file instead.

If you don’t want the keep awake function (didn't need it on my 3), no need to enter your credentials at all.

marcone, 2019-06-24:
There is no support for specifying an auth token instead of login credentials in the setup conf file, but the tesla_api.py script retrieves such a token using the credentials, then stores it as JSON in /mutable/tesla_api.json, so you could create that file yourself.
For most changes you can update the .conf file and run /root/bin/setup-teslausb.

closed 2019-06-24 by marcone