marcone/teslausb #99: logread command not found

bahree, 2019-06-24:
I am trying to set this up and after changing some of the config file, was re-running the setup to get a log but I get the following error instead. 

```
root@teslausb:~# /root/bin/setup-teslausb diagnose > /mutable/diagnostics.txt
Read setup info from teslausb_setup_variables.conf (yes/no/cancel)? yes
/root/bin/setup-teslausb: line 388: logread: command not found
```

I was trying to understand where do I change the cifs vers from 3 to 2.0 or 1.0. I am trying to get this to copy to a Synology NAS which doesn't like v3 it seems.

Thanks.

marcone, 2019-06-24:
hmm, `/bin/logread` is part of the busybox-syslogd package, which is installed by default. Did you not use the prebuilt teslausb image?

marcone, 2019-06-24:
It's possible that logread wasn't installed yet if setup didn't complete, since it doesn't get installed until late in the setup process. I just added a fix for this. You can run `setup-teslausb selfupdate` to get the fix.

closed 2019-06-24 by marcone

bahree, 2019-06-24:
Appreciate the quick turnaround - I did use the prebuilt image. Let me try this and come back here. Thanks!

bahree, 2019-06-24:
OK, that fixed it. Just reporting here so if anyone else has an issue - they can find it too. Thanks again.