marcone/teslausb #102: OneStepSetup completes and even syncs music, but nothing for video

wwwb0n3zcom, 2019-06-25:
I have tried the OneStepSetup multiple times. I don't know where to turn to next. I'm not a pro at Linux or Pi but enough to be dangerous. Music is sync'ing fine. I have saved recording from TeslaCam under SavedClips (if I attached the Pi to a computer). But TeslaCam doesn't sync to my NAS like my music does.

Diagnostics log file:
https://gist.github.com/wwwb0n3zcom/e33bc472fbb42099e32a20ebc306ddfc

Hardware:
Raspberry Pi Zero W
200GB MicroSD card

Tesla firmware:
2019.20.4.1 currently

wwwb0n3zcom, 2019-06-25:
Using prebuilt image:
teslausb-20190615

marcone, 2019-06-25:
I'm surprised you say you have recordings under SavedClips, because the diagnostic log shows:
```
Thu  3 Nov 10:17:15 PDT 2016: /mnt/cam/TeslaCam/SavedClips does not exist, skipping
```

wwwb0n3zcom, 2019-06-25:
Remove from the car and attached it to my computer just now.

`E:\>dir /s
 Volume in drive E is CAM
 Volume Serial Number is 24F2-42A3

 Directory of E:\

06/24/2019  08:53 PM    <DIR>          TeslaCam
               0 File(s)              0 bytes

 Directory of E:\TeslaCam

06/25/2019  02:09 AM    <DIR>          .
06/25/2019  02:09 AM    <DIR>          ..
06/24/2019  08:53 PM    <DIR>          RecentClips
06/24/2019  07:33 PM    <DIR>          SavedClips
               0 File(s)              0 bytes

 Directory of E:\TeslaCam\RecentClips

06/24/2019  07:30 PM    <DIR>          .
06/24/2019  07:30 PM    <DIR>          ..
06/24/2019  08:35 PM               595 2019-06-24_20-34-42-front.mp4
06/24/2019  08:35 PM               595 2019-06-24_20-34-42-left_repeater.mp4
06/24/2019  08:35 PM               595 2019-06-24_20-34-42-right_repeater.mp4
06/24/2019  08:36 PM               595 2019-06-24_20-35-42-front.mp4
06/24/2019  08:36 PM               595 2019-06-24_20-35-42-left_repeater.mp4
06/24/2019  08:36 PM               595 2019-06-24_20-35-42-right_repeater.mp4
06/24/2019  08:37 PM               595 2019-06-24_20-36-42-front.mp4
06/24/2019  08:37 PM               595 2019-06-24_20-36-42-left_repeater.mp4
06/24/2019  08:37 PM               595 2019-06-24_20-36-42-right_repeater.mp4
06/24/2019  08:38 PM               595 2019-06-24_20-37-43-front.mp4
06/24/2019  08:38 PM               595 2019-06-24_20-37-43-left_repeater.mp4
06/24/2019  08:38 PM               595 2019-06-24_20-37-43-right_repeater.mp4
06/24/2019  08:39 PM               595 2019-06-24_20-38-43-front.mp4
06/24/2019  08:39 PM               595 2019-06-24_20-38-43-left_repeater.mp4
06/24/2019  08:39 PM               595 2019-06-24_20-38-43-right_repeater.mp4
06/24/2019  08:40 PM               595 2019-06-24_20-39-43-front.mp4
06/24/2019  08:40 PM               595 2019-06-24_20-39-43-left_repeater.mp4
06/24/2019  08:40 PM               595 2019-06-24_20-39-43-right_repeater.mp4
06/24/2019  08:41 PM               595 2019-06-24_20-40-43-front.mp4
06/24/2019  08:41 PM               595 2019-06-24_20-40-43-left_repeater.mp4
06/24/2019  08:41 PM               595 2019-06-24_20-40-43-right_repeater.mp4
06/24/2019  08:42 PM               595 2019-06-24_20-41-44-front.mp4
06/24/2019  08:42 PM               595 2019-06-24_20-41-44-left_repeater.mp4
06/24/2019  08:42 PM               595 2019-06-24_20-41-44-right_repeater.mp4
06/24/2019  08:43 PM               595 2019-06-24_20-42-44-front.mp4
06/24/2019  08:43 PM               595 2019-06-24_20-42-44-left_repeater.mp4
06/24/2019  08:43 PM               595 2019-06-24_20-42-44-right_repeater.mp4
06/24/2019  08:44 PM               595 2019-06-24_20-43-44-front.mp4
06/24/2019  08:44 PM               595 2019-06-24_20-43-44-left_repeater.mp4
06/24/2019  08:44 PM               595 2019-06-24_20-43-44-right_repeater.mp4
06/24/2019  08:45 PM               595 2019-06-24_20-44-44-front.mp4
06/24/2019  08:45 PM               595 2019-06-24_20-44-44-left_repeater.mp4
06/24/2019  08:45 PM               595 2019-06-24_20-44-44-right_repeater.mp4
06/24/2019  08:46 PM               595 2019-06-24_20-45-45-front.mp4
06/24/2019  08:46 PM               595 2019-06-24_20-45-45-left_repeater.mp4
06/24/2019  08:46 PM               595 2019-06-24_20-45-45-right_repeater.mp4
06/24/2019  08:47 PM               595 2019-06-24_20-46-45-front.mp4
06/24/2019  08:47 PM               595 2019-06-24_20-46-45-left_repeater.mp4
06/24/2019  08:47 PM               595 2019-06-24_20-46-45-right_repeater.mp4
06/24/2019  08:48 PM               595 2019-06-24_20-47-45-front.mp4
06/24/2019  08:48 PM               595 2019-06-24_20-47-45-left_repeater.mp4
06/24/2019  08:48 PM               595 2019-06-24_20-47-45-right_repeater.mp4
06/24/2019  08:49 PM               595 2019-06-24_20-48-45-front.mp4
06/24/2019  08:49 PM               595 2019-06-24_20-48-45-left_repeater.mp4
06/24/2019  08:49 PM               595 2019-06-24_20-48-45-right_repeater.mp4
06/24/2019  08:50 PM               595 2019-06-24_20-49-46-front.mp4
06/24/2019  08:50 PM               595 2019-06-24_20-49-46-left_repeater.mp4
[... 131 lines cut ...]
06/24/2019  07:31 PM           131,169 2019-06-24_19-30-16-front.mp4
06/24/2019  07:31 PM               595 2019-06-24_19-30-16-left_repeater.mp4
06/24/2019  07:31 PM               595 2019-06-24_19-30-16-right_repeater.mp4
06/24/2019  07:32 PM           130,882 2019-06-24_19-31-16-front.mp4
06/24/2019  07:32 PM               595 2019-06-24_19-31-16-left_repeater.mp4
06/24/2019  07:32 PM               595 2019-06-24_19-31-16-right_repeater.mp4
06/24/2019  07:33 PM           130,882 2019-06-24_19-32-17-front.mp4
06/24/2019  07:33 PM               595 2019-06-24_19-32-17-left_repeater.mp4
06/24/2019  07:33 PM               595 2019-06-24_19-32-17-right_repeater.mp4
06/24/2019  07:33 PM            65,866 2019-06-24_19-33-17-front.mp4
06/24/2019  07:33 PM               595 2019-06-24_19-33-17-left_repeater.mp4
06/24/2019  07:33 PM               595 2019-06-24_19-33-17-right_repeater.mp4
              15 File(s)        530,615 bytes

 Directory of E:\System Volume Information

06/24/2019  09:50 PM    <DIR>          .
06/24/2019  09:50 PM    <DIR>          ..
06/24/2019  09:50 PM                12 WPSettings.dat
06/24/2019  09:50 PM                76 IndexerVolumeGuid
               2 File(s)             88 bytes

     Total Files Listed:
             179 File(s)        627,093 bytes
              14 Dir(s)  138,468,327,424 bytes free`

marcone, 2019-06-25:
That's very odd. Can you try the following:
```
sudo -i
cd /backingfiles
/root/bin/mount_snapshot.sh cam_disk.bin snap.bin /tmp/snap
find /tmp/snap/
if [ -d /tmp/snap/TeslaCam/SavedClips ]; then echo dir; else echo not dir; fi
/root/bin/release_snapshot.sh /tmp/snap
rm snap.bin
```

wwwb0n3zcom, 2019-06-25:
Not completely sure what I did with these commands, but I'll study them and learn. Here is the output:

```
root@teslausb:~# sudo -i

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

root@teslausb:~# cd /backingfiles
root@teslausb:/backingfiles# /root/bin/mount_snapshot.sh cam_disk.bin snap.bin /tmp/snap
fsck from util-linux 2.29.2
fsck.fat 4.1 (2017-01-24)
Free cluster summary wrong (4225720 vs. really 4225718)
  Auto-correcting.
Performing changes.
/dev/loop0p1: 185 files, 199/4225917 clusters
root@teslausb:/backingfiles# find /tmp/snap/
/tmp/snap/
/tmp/snap/TeslaCam
/tmp/snap/TeslaCam/RecentClips
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-34-42-front.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-34-42-left_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-34-42-right_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-35-42-front.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-35-42-left_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-35-42-right_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-36-42-front.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-36-42-left_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-36-42-right_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-37-43-front.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-37-43-left_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-37-43-right_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-38-43-front.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-38-43-left_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-38-43-right_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-39-43-front.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-39-43-left_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-39-43-right_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-40-43-front.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-40-43-left_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-40-43-right_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-41-44-front.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-41-44-left_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-41-44-right_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-42-44-front.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-42-44-left_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-42-44-right_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-43-44-front.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-43-44-left_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-43-44-right_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-44-44-front.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-44-44-left_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-44-44-right_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-45-45-front.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-45-45-left_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-45-45-right_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-46-45-front.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-46-45-left_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-46-45-right_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-47-45-front.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-47-45-left_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-47-45-right_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-48-45-front.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-48-45-left_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-48-45-right_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-49-46-front.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-49-46-left_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-49-46-right_repeater.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-50-47-front.mp4
/tmp/snap/TeslaCam/RecentClips/2019-06-24_20-50-47-left_repeater.mp4
[... 113 lines cut ...]
/tmp/snap/TeslaCam/SavedClips/2019-06-24_19-33-31
/tmp/snap/TeslaCam/SavedClips/2019-06-24_19-33-31/2019-06-24_19-29-16-front.mp4
/tmp/snap/TeslaCam/SavedClips/2019-06-24_19-33-31/2019-06-24_19-29-16-left_repeater.mp4
/tmp/snap/TeslaCam/SavedClips/2019-06-24_19-33-31/2019-06-24_19-29-16-right_repeater.mp4
/tmp/snap/TeslaCam/SavedClips/2019-06-24_19-33-31/2019-06-24_19-30-16-front.mp4
/tmp/snap/TeslaCam/SavedClips/2019-06-24_19-33-31/2019-06-24_19-30-16-left_repeater.mp4
/tmp/snap/TeslaCam/SavedClips/2019-06-24_19-33-31/2019-06-24_19-30-16-right_repeater.mp4
/tmp/snap/TeslaCam/SavedClips/2019-06-24_19-33-31/2019-06-24_19-31-16-front.mp4
/tmp/snap/TeslaCam/SavedClips/2019-06-24_19-33-31/2019-06-24_19-31-16-left_repeater.mp4
/tmp/snap/TeslaCam/SavedClips/2019-06-24_19-33-31/2019-06-24_19-31-16-right_repeater.mp4
/tmp/snap/TeslaCam/SavedClips/2019-06-24_19-33-31/2019-06-24_19-32-17-front.mp4
/tmp/snap/TeslaCam/SavedClips/2019-06-24_19-33-31/2019-06-24_19-32-17-left_repeater.mp4
/tmp/snap/TeslaCam/SavedClips/2019-06-24_19-33-31/2019-06-24_19-32-17-right_repeater.mp4
/tmp/snap/TeslaCam/SavedClips/2019-06-24_19-33-31/2019-06-24_19-33-17-front.mp4
/tmp/snap/TeslaCam/SavedClips/2019-06-24_19-33-31/2019-06-24_19-33-17-left_repeater.mp4
/tmp/snap/TeslaCam/SavedClips/2019-06-24_19-33-31/2019-06-24_19-33-17-right_repeater.mp4
/tmp/snap/System Volume Information
/tmp/snap/System Volume Information/WPSettings.dat
/tmp/snap/System Volume Information/IndexerVolumeGuid
root@teslausb:/backingfiles# if [ -d /tmp/snap/TeslaCam/SavedClips ]; then echo dir; else echo not dir; fi
dir
root@teslausb:/backingfiles# /root/bin/release_snapshot.sh /tmp/snap
root@teslausb:/backingfiles# rm snap.bin
root@teslausb:/backingfiles#
```

wwwb0n3zcom, 2019-06-25:
BTW Marcone - I just wanted to pause and tell you how I appreciate you and others helping this project. Once I learn more, I would also like to help support others in learning how to use this project for themselves. If your ever in Orange, CA - I owe you a beer or something to say thank you.

marcone, 2019-06-26:
The output above confirms that there is a SavedClips folder, and that the Pi sees it.
The diagnostics you attached at the top of this issue showed "/mnt/cam/TeslaCam/SavedClips does not exist", however based on the timestamps that was *before* the files shown above were created (the "does not exist" message is from prior to 19:29:09, and the files shown above are from after 19:29:16), so there's no inherent contradiction there.
I would expect these files to be backed up on the next cycle.

wwwb0n3zcom, 2019-06-26:
Marcone - thanks to a friend on Discord (ScrawnyB - you rock!), I found out my permissions on my NAS box where the problem. Even though I had it set to R/W for the user account being used by TeslaUSB - it was still failing. I validated this using a PowerShell script from my PC and trying to connect using the same creds and via IP as the TeslaUSB is doing.

I have now SavedClips writing directly to my NAS share. I need to read up on how to move this to a TeslaCam\SavedClips directory and wish there was an option for saving ALL video (not just saved clips) and mimic the directory structure on the USB itself.

So maybe in the diagnostics log output - test permissions of the network share and output if it fails?

I really do appreciate your time Marcone and I'm sorry if I had you puzzled. I just didn't know where to turn. I learned a great deal from you and others - Thank you.

closed 2019-06-26 by wwwb0n3zcom

ScrawnyB, 2019-06-26:
This does bring up another semi-unrelated question for me, marcone. Do we have any scenario handling for target share being full, or what happens if let's say my share had 8GB free and I had 32GB of stuff on my Saved dashcam partition)?

marcone, 2019-06-26:
@wwwb0n3zcom I don't think that the non-writeable NAS was the original problem though. Or at least, the diagnostics simply showed there was nothing to back up at the time. In any case, I'll make the script handle and log a non-writeable destination. To mimic the structure on the USB drive, just create a TeslaCam/SavedClips folder on your NAS and specify that as the archive destination. Backing up RecentClips as well is on my list, but right now I don't think is very useful because it will only ever contain video from the most recent hour. I plan on first making it so that all video is retained, and then add an option to back it up. (and note that in that case you probably *don't* want to mimic the USB directory structure, because your archived RecentClips will become quite large.

@ScrawnyB There is nothing to handle that currently. Not really sure what to do in that case either. Currently, it will just fail to archive, the USB drive will fill up eventually, and finally the car will show the X icon, at which point you'll notice something is wrong.