marcone/teslausb #103: Have to modprobe for every reboot.

manosinistra, 2019-06-26:
M3 SR+ on 2019.20.2.1 5659e07.

Raspberry Pi Zero W with the latest Marcone/TeslaUsb image on a 128GB SanDisk Ultra.

Can SSH to it.

Ran interactive setup.

Computer can see USB volume "CAM" with directory TeslaCam in the root folder.

Can mount and see CIFS volume.

But when I plug into my 3 using the same USB cables, the DashCam icon never comes up. If I switch to a regular USB stick, the DashCam recording icon appears.

Only way to make it work is if I ssh and:

modprobe -r g_mass_storage

followed by:

modprobe g_mass_storage

I added those two lines at the end of my rc.local just above the exit(0). Did I miss a step somewhere?

marcone, 2019-06-26:
Mobprobe is run by the archiveloop script, and the fact that you see the "CAM" drive when connected to a computer seems to indicate that archiveloop is running. There shouldn't be anything different between a PC and the car, as far as the Pi is concerned.
Could be an issue with 2019.20, which I don't have yet.

marcone, 2019-06-27:
I just got 2019.20.4.2 on my S, and am still not seeing this problem. Sorry, nothing I can do here.

closed 2019-06-27 by marcone