marcone/teslausb #106: IFTTT notification at start of sync?

bahree, 2019-06-28:
Not really an issue, but rather a question / suggestion - I wanted to know if it is possible to send out a alert at the start when the pi connects to wifi and kicks off the copy? I have this configured to get the alert when it finishes, but at start would be good too. Thanks.

kdvlr, 2019-06-28:
I like the idea. I was planning to push a PR for making the push notification generic so they could be used elsewhere. I'll add it to my list.

kdvlr, 2019-06-28:
Added PR https://github.com/marcone/teslausb/pull/107 which should address this.

bahree, 2019-06-29:
Awesome thanks. So how can one get this? Would the self update work or if that once the PR is approved?

closed 2019-06-29 by bahree

kdvlr, 2019-06-29:
@marcone - Makes the final decision on what is merged to this repository. They are usually pretty good with it. However, you can get it from my repo if you're in a hurry.

bahree, 2019-06-29:
Ah ok, I can wait. So should I reopen this issue until @marcone decides, or that doesn't matter?