marcone/teslausb #110: DashCam icon not appearing

darrynlowe, 2019-06-30:
I've just created a teslausb-20190615 install and are having no luck getting the dashcam icon to appear in the car (Model X version 2019.20.4.2). 

Debug attempts have been:
- The TeslaCam folder appears when connecting the Pi to a USB port on a desktop computer
- The dashcam icon does appear when using a regular USB stick. 
- Running the Pi off seperate 12v power for the two usb cable approach
- Running the "modprobe" approach per issue #103 (no impact)
- Rebooting the pi with/without the USB cable connected per issue #100 (no impact - nothing makes the dashcam icon appear)

Bar the in vechile dashcam problem, everything else appears to be working great, including cifs.

Diagnostic log is attached.

Any ideas?
[diagnostics.txt](https://github.com/marcone/teslausb/files/3343035/diagnostics.txt)

paulio23, 2019-06-30:
I have the same problem. Never got it to work in my model S and gave up. Works great on my 3.

marcone, 2019-06-30:
Did you take those diagnostics while it was connected to the car? Everything looks fine at first glance.
Since you're using separate power, have you tried waiting for the Pi to be fully booted (flashing LED) before connecting it to the car's USB?

darrynlowe, 2019-06-30:
Those were taken while it was connected to the car. And I tried plugging/unplugging the USB cable after it had been fully booted.

One thought is that the car is an MCU1 Model X... I wonder if there is some differene in how the MCU1 and MCU2 Tesla mount USB drives that's causing the Pi to fail.

darrynlowe, 2019-07-01:
So I figured out the problem and it's working now:
- The Model X (at least the MCU1 version) doesn't have enough power to drive the Pi Zero W
- Fix this by using a seperate USB power cable from a DC adapter to the Pi Zero Power port
- Make sure that you create an OTG Cable to connect the Pi USB port to the car USB port (i.e., I made one by removing Pin 1 (5v) from a regular USB to Micro-USB cable)

The mistake I made originally was not removing the 5v pin correctly, which was confusing the car and causing it to isolate the port the Pi connected.

closed 2019-07-01 by darrynlowe