marcone/teslausb #113: Not Syncing Music

AceTheGmodGeek, 2019-07-04:
Hello!

I have a share for the USB named "tesla", and I didn't see the music share name until after setup, so I went back and edited the config. I have a folder in the NAS, named "music", so it would be "tesla/music". But, it doesn't seem to be syncing my music for whatever reason. 

Diagnostic: https://pastebin.com/HwYf33N2

Here is the part of my config that has the music share part- https://pastebin.com/002eKzu1

The videos sync from the pi to the NAS, but the music doesn't sync from the NAS to the pi.

marcone, 2019-07-04:
I see in your diagnostics that /etc/fstab does not contain an entry for your music share, so that's why it's currently not working.
Did you rerun setup after modifying the config file?
If not, there's your problem. Otherwise, there might be a bug in one the scripts.

AceTheGmodGeek, 2019-07-04:
Oh yeah, that is definitely the problem. How can I rerun setup?

marcone, 2019-07-04:
ssh into the Pi, then:
```
sudo -i
/root/bin/setup-teslausb
reboot
```

closed 2019-07-04 by marcone