marcone/teslausb #114: Archive all files instead of just saved/sentry clips

Pedals2Paddles, 2019-07-05:
I think a fairly simple additional option would be to archive all the clips in the TeslaCam directory, not just the saved clips.  Yes, it is only the last hour. But that could account for the majority of all drives for a lot of people.  It of course could quickly fill up storage at home, so maybe something you enable/disable as an option.  I'd deal with that issue locally on the server, but not everyone would want this.

I'm thinking this would be as simple as changing https://github.com/marcone/teslausb/blob/main-dev/run/cifs_archive/archive-clips.sh#L80 from `/TeslaCam/SavedClips` to just `/TeslaCam` and it will recurse through the current hour and all the subdirectories of saved clips?

I haven't dug through the scripts deep enough to find where there could be an EXPORT to enable/disable this yet.

gkwok1, 2019-07-07:
@Pedals2Paddles I made a few changes by adding additional variable for 2018/2019 firmware, sync start/end times, sync recentclips/savedclips flags and always sync flag to bypass the sync time. Please remove .txt extensions from these files.

@marcone please feel free to integrate this into your code. I added the sync times because I don't want it to sync when I am only stopping at my home for a few minutes and would prefer to sync the videos when I am asleep.

[archive-clips.sh,txt](https://github.com/marcone/teslausb/files/3365162/archive-clips.sh.txt)
[archiveloop.txt](https://github.com/marcone/teslausb/files/3365163/archiveloop.txt)
[teslausb.conf.txt](https://github.com/marcone/teslausb/files/3365164/teslausb.conf.txt)

bahree, 2019-07-07:
Maybe this could be a comfig driven option - will allow for one to have greater flexibility.
    	
    	

    	Device >>> BrevityAmit Bahree / @bahreeMobile: +1425-647-9961

bdoooh, 2019-08-07:
@marcone was there a reason to exclude `RecentClips` from archiving? Like @Pedals2Paddles said, that accounts for the majority of drives. I recently just thought that Sentry was overriding the RecentClips drives and I've been wondering if I should just be tapping the DashCam icon every 10 minutes.

marcone, 2019-08-08:
@bdoooh it only covers the majority of drives if the majority of drives return home within an hour, which I very much doubt.
The reason it's not enabled is mainly historical: it wasn't enabled in the original cimryan project. I haven't enabled it myself because personally I don't find it all that useful to only archive the last hour: if I'm going to archive everything, I'd like to archive *everything*, not just the trip home.
This is something I'm working on in the 'snapshots' branch: it will take continuous snapshots of the recordings, so they're all accessible when I get home. It lets me keep a record of several days worth of commuting. Once I'm done with that, I'll integrate it into the main branch and add an option to also archive everything.
Until then, if archiving only clips from the last hour is useful to you, it's trivial to add that functionaloty in the archive-clips.sh script by adding the RecentClips folder to the pattern of things to archive.

bdoooh, 2019-08-08:
@marcone that makes sense. That snapshots integration sounds awesome! Very good idea. 

I went ahead and added that line to the `archive-clips` file. The reason I had asked was the `SavedClips` directory never includes any driving (even in the last hour) unless a user hits the dashcam icon on the screen. 

Not sure if we should add this to the README, but I was personally confused at first at why didn't any of the recordings pushed to the NAS included any driving. Might help some folks who are Googling this problem to find. 

Thanks again for your efforts!

Pedals2Paddles, 2019-08-08:
That sounds awesome.  Sorry I haven't followed up on this. I posted it then got busy with a bunch of other stuff. Been running this (unmodified) for a month now with no problems. Hoping to try some these changes this weekend.

I agree having only the last hour is not really a complete solution.  The snapshot idea sounds like a great workaround!

crimsondr, 2019-08-09:
Would it be naive to add a background that rsync's the contents of recentclips into savedclips?

marcone, 2019-08-09:
You really don't want to be writing to the disk while the Tesla is also writing to it.

crimsondr, 2019-08-09:
Thanks.  If you're introducing a way to archive all clips, would you include some functionality to rotate or delete older clips?

marcone, 2019-08-09:
I don't know yet

acraigl, 2019-08-13:
> Thanks. If you're introducing a way to archive all clips, would you include some functionality to rotate or delete older clips?

I got around this by writing a simple scheduled task using a program called delage64. There's ways to do it natively using ForFiles but I was having trouble when I was first setting it up. The delage64 program worked wonderfully.
https://www.horstmuc.de/win/delage.htm

In short, it automatically removes any files and folders older than X days. I set it to 10 and I basically don't have to do any maintenance of files anymore, either on the USB or my NAS. It's perfect.

Here's the task command, FWIW:
X:\PathTo\delage64\delage64.exe "T:\*.mp4" 10 /quiet /rd /recurse

Where X is the location of the delage64 program and T is the letter drive share for your NAS with the TeslaCAM videos.

crimsondr, 2019-08-14:
I modified the archive-clips.sh script to include RecentClips and then my X started reporting that the drive is too slow and would not record dashcam video.  I reverted the change and it once again records dashcam video without issue.  Maybe I need a faster microsd card?  But it's not archiving while I'm driving so I don't know why it would affect anything.

miles267, 2019-10-24:
How do you also archive the RecentClips folder with CIFS?  I've added to archive-clips.sh:

connectionmonitor $$ &

# new file name pattern, firmware 2019.*
moveclips "$CAM_MOUNT/TeslaCam/SavedClips" '*'

# v10 firmware adds a SentryClips folder
moveclips "$CAM_MOUNT/TeslaCam/SentryClips" '*'

# this should add RecentClips folder
moveclips "$CAM_MOUNT/TeslaCam/RecentClips" '*'

and then to the archiveloop:

DIR_COUNT=$(cd "$CAM_MOUNT"/TeslaCam && find . -maxdepth 2 -path './SavedClips/*' -o -path './SentryClips/*' -o -path './RecentClips/*' -type d | wc -l)

FILE_COUNT=$(cd "$CAM_MOUNT"/TeslaCam && find . -maxdepth 3 -path './SavedClips/*' -o -path './SentryClips/*' -o -path './RecentClips/*' -type f | wc -l)

Yet still RecentClips folder doesn't appear on my CIFS file share?

miles267, 2019-10-24:
Shouldn't these two lines be sufficient to archive the RecentClips folder using CIFS?

archive-clips.sh:
moveclips "$CAM_MOUNT/TeslaCam/RecentClips" '*'

archiveloop:
DIR_COUNT=$(cd "$CAM_MOUNT"/TeslaCam && find . -maxdepth 2 -path './SavedClips/' -o -path './SentryClips/' -o -path './RecentClips/*' -type d | wc -l)

FILE_COUNT=$(cd "$CAM_MOUNT"/TeslaCam && find . -maxdepth 3 -path './SavedClips/' -o -path './SentryClips/' -o -path './RecentClips/*' -type f | wc -l)

sboesebeck, 2019-10-24:
the latest version of the cifs archive-clips file does use rsync. It is structured a lot simple. 

if you want do have all clips synced over, you need to change line 5 to this:

`num_files_moved=$(rsync -auvh --timeout=60 --remove-source-files --no-perms --stats --log-file=/tmp/archive-rsync-cmd.log $CAM_MOUNT/TeslaCam/ $ARCHIVE_MOUNT/ | awk '/files transferred/{print $NF}')`

this should sync all files in the TeslaCam folder to your cifs archive...

miles267, 2019-10-24:
@sboesebeck am running buster 0922 image that's working.  my archive-clips.sh file doesn't appear to be using rsync.  Do I need to run some sort of upgrade procedure to leverage this new way?

sboesebeck, 2019-10-24:
It is in main-dev. Running ./setup-teslausb selfupdate followed by ./setup-teslausb should do the trick...

miles267, 2019-10-24:
> It is in main-dev. Running ./setup-teslausb selfupdate followed by ./setup-teslausb should do the trick...
> […](#)

Thank you.  Yes, I've since run the update and got the rsync improvements in archive-clips.sh and made the modification to line 5 you had suggested, however even when simulating a disconnect (touch /tmp/archive_is_unreachable, my /backingfiles/TeslaCam/RecentClips are not appearing on my file server.  Even though I've confirmed there is content in the /RecentClips folder.  The other two are blank at moment (SentryClips, SavedClips).

miles267, 2019-10-24:
Thank you @sboesebeck. Yes, I've since run the update and got the rsync improvements in archive-clips.sh and made the modification to line 5 you had suggested, however even when simulating a disconnect (touch /tmp/archive_is_unreachable, my /backingfiles/TeslaCam/RecentClips are not appearing on my file server. Even though I've confirmed there is content in the /RecentClips folder. The other two are blank at moment (SentryClips, SavedClips).

sboesebeck, 2019-10-24:
@miles267  this is strange. Problem might be:
- you have a typo in the archive-clips.sh or some other error, then the script will exit. Have a look at /mutable/archiveloop.log... there you might see something
- if the rsync itself does not work properly, have a look at /tmp/archive-rsync-cmd.log... this should give more information on what happens.

how many files are there? It might take a while until the actual sync will start...

miles267, 2019-10-24:
I'm seeing this in the archiveloop.log.  Also it appears that under /backingfiles/TeslaCam/RecentClips there are a lot of files. But they're listed as some sort of file links?

*FROM ARCHIVELOOP.LOG*
Thu 24 Oct 14:42:07 CDT 2019: making links for /backingfiles/snapshots/newsnap/mnt, retargeted to /backingfiles/snapshots/snap-000000/mnt
Thu 24 Oct 14:42:07 CDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-24_13-40-44-back.mp4
Thu 24 Oct 14:42:07 CDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-24_13-40-44-front.mp4
Thu 24 Oct 14:42:08 CDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-24_13-40-44-left_repeater.mp4
Thu 24 Oct 14:42:08 CDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-24_13-40-44-right_repeater.mp4
Thu 24 Oct 14:42:08 CDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-24_13-41-45-back.mp4
Thu 24 Oct 14:42:08 CDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-24_13-41-45-front.mp4
Thu 24 Oct 14:42:09 CDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-24_13-41-45-left_repeater.mp4
Thu 24 Oct 14:42:09 CDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-24_13-41-45-right_repeater.mp4
Thu 24 Oct 14:42:09 CDT 2019: linking /backingfiles/snapshots/newsnap/mnt/TeslaCam/RecentClips/2019-10-24_13-42-45-back.mp4

*FROM /RECENTCLIPS/2019-10-24 subfolder:*
lrwxrwxrwx 1 root root 89 Feb 14  2019 2019-10-24_12-45-57-back.mp4 -> /backingfiles/snapshots/snap-000000/mnt/TeslaCam/RecentClips/2019-10-24_12-45-57-back.mp4
lrwxrwxrwx 1 root root 90 Feb 14  2019 2019-10-24_12-45-57-front.mp4 -> /backingfiles/snapshots/snap-000000/mnt/TeslaCam/RecentClips/2019-10-24_12-45-57-front.mp4
lrwxrwxrwx 1 root root 98 Feb 14  2019 2019-10-24_12-45-57-left_repeater.mp4 -> /backingfiles/snapshots/snap-000000/mnt/TeslaCam/RecentClips/2019-10-24_12-45-57-left_repeater.mp4
lrwxrwxrwx 1 root root 99 Feb 14  2019 2019-10-24_12-45-57-right_repeater.mp4 -> /backingfiles/snapshots/snap-000000/mnt/TeslaCam/RecentClips/2019-10-24_12-45-57-right_repeater.mp4
lrwxrwxrwx 1 root root 89 Feb 14  2019 2019-10-24_12-46-58-back.mp4 -> /backingfiles/snapshots/snap-000000/mnt/TeslaCam/RecentClips/2019-10-24_12-46-58-back.mp4
lrwxrwxrwx 1 root root 90 Feb 14  2019 2019-10-24_12-46-58-front.mp4 -> /backingfiles/snapshots/snap-000000/mnt/TeslaCam/RecentClips/2019-10-24_12-46-58-front.mp4
lrwxrwxrwx 1 root root 98 Feb 14  2019 2019-10-24_12-46-58-left_repeater.mp4 -> /backingfiles/snapshots/snap-000000/mnt/TeslaCam/RecentClips/2019-10-24_12-46-58-left_repeater.mp4
lrwxrwxrwx 1 root root 99 Feb 14  2019 2019-10-24_12-46-58-right_repeater.mp4 -> /backingfiles/snapshots/snap-000000/mnt/TeslaCam/RecentClips/2019-10-24_12-46-58-right_repeater.mp4

sboesebeck, 2019-10-24:
this is the snapshot functionality. The system creates snapshots of your directory using symbolic links in the filesystem, so that you do not copy files, that are created right now. You would end up having corrupted files.

You can switch that off setting SNAPSHOTS_ENABLED=false in your settings. (and rebooting the pi).

but that should not have an effect on the rsync mechanism.

miles267, 2019-10-24:
> this is the snapshot functionality. The system creates snapshots of your directory using symbolic links in the filesystem, so that you do not copy files, that are created right now. You would end up [...]

OK.  In that case, I do not have any data in my RecentClips folder to be backed up then.  If I get a directory listing under /RecentClips/2019-10-24, all it returns in an extensive list of those symbolic link files.  I suppose I must have it setup correctly but nothing is being written to my RecentClips folder.

sboesebeck, 2019-10-24:
that might be. If you want to test the sync functionality, just create a bunch of files... the file name does not matter for the sync to work - it takes any file in any directory in the TeslaCam folder.

So just for testing the sync, you could create a bunch of files in the teslacam-dir using 
`touch filename.txt`

this creates an empty file

if you want to create a whole bunch of files, you could do something like:

`for n in $(seq 10); do echo "File $i" > file_$i.tst; done`

this would create 10 files with the content "File 1" to "File 10"...

you can create directories for the sync if you like, do it in RecentClips or so.

maybe this helps a bit.

miles267, 2019-10-27:
@sboesebeck are you able to sync all 3 folders with your setup? /SentryClips, /SavedClips AND /RecentClips?  I've reimaged my card to start over again with buster 0922 and all the latest updates.  Cannot figure out how to get /RecentClips folder to archive to my CIFS file share.

sboesebeck, 2019-10-28:
to be honest, I never tried to have that all synced. As I said, the line i posted above _should_ work.

The rsync line i posted does not work as expected, try to add the RecentClips to the line in the script:

```
num_files_moved=$(rsync -auvh --timeout=60 --remove-source-files --no-perms --stats --log-file=/tmp/archive-rsync-cmd.log $CAM_MOUNT/TeslaCam/SentryClips $CAM_MOUNT/TeslaCam/SavedClips $CAM_MOUNT/TeslaCam/RecentClips $ARCHIVE_MOUNT/ | awk '/files transferred/{print $NF}')
```

but have a closer look at the logs. It seems like there is some other error, that is not related to the directory. I'd guess, there is a typo somewhere  in the script and hence id would not run at all...

miles267, 2019-10-28:
@sboesebeck sorry for the dumb question, but when I look inside my /RecentClips on my teslausb RP0, the only content is a dated folder (i.e. 2019-10-28) and, within that folder, there are a series of symbolic links to other files.  In that case, it wouldn't even be possible to archive symbolic linked files, correct?

sboesebeck, 2019-10-29:
hmm... it seems that snapshotting does not work with RecentClips - switch off the snapshot feature as I mentioned above and retry (SNAPSHOTS_ENABLED=false in your settings)

miles267, 2019-10-29:
I currently have snapshots enabled = true which I believe is the default setting.  And things are working.  As in, I'm not receiving USB slow errors.  Though I don't truly understand the function of snapshots so I've not disabled it.  Do people generally suggest keeping it on/true or off/false?

sboesebeck, 2019-10-29:
usually it is a good idea to keep the snapshot _on_ but you want to achieve something out of the ordenary. So we should reduce complexity. If it works with snapshots off, awesome. If not, we need to keep looking

jun3280net, 2019-11-03:
> Maybe this could be a config driven option - will allow for one to have greater flexibility?

Hi @bahree!  I remember you from the Teslamate GIT :)  Were you able to get this working? Or is yours only working for the Saved clips?

Cheers

clean-ev, 2019-11-17:
@sboesebeck - you mention rsync use in cifs archive-clips.sh, however looking at the code here at github and a newly built Pi0W this morning does not reflect rsync at all - where and what do i change to get this changes?
I rebuilt to make sure that I am getting the current code to no avail.
On one of your postings you mention that the rsync is part of main-dev however  from what I tell, setup.conf file mentions to change that specific export setting if  I plan to user other branch, so by default it is using main-dev, no?
Your guidance will be approciated

marcone, 2019-11-17:
@clean-ev I reverted the rsync-for-cifs change because rsync's "--timeout" option is broken for local copies, so it would hang for a long time when moving out of wifi range while archiving.

clean-ev, 2019-11-17:
Thanks @marcone.
After I posted I found that you committed changes on Oct 30.

So at this point will the below two changes be sufficient to archive RecentClips? 

___________________

Archive-clips.sh addition to include RecentClips in addition to SentryClips and SavedClips
——-
moveclips "$CAM_MOUNT/TeslaCam/RecentClips" '*

___________________
archiveloop changes
——-
DIR_COUNT=$(cd "$CAM_MOUNT"/TeslaCam && find . -maxdepth 2 -path ‘./RecentClips/*’ -type d -o -path './SavedClips/*' -type d -o -path './SentryClips/*' -type d | wc -l)
FILE_COUNT=$(cd "$CAM_MOUNT"/TeslaCam && find . -maxdepth 3 -path ‘./RecentClips/*’ -type f -o -path './SavedClips/*' -type f -o -path './SentryClips/*' -type f | wc -l)

miles267, 2019-11-17:
Am interest in this as well.  Have yet to find a way to archive RecentClips.

miles267, 2019-11-25:
@clean-ev have you confirmed these 2 entries above work to backup RecentClips?

clean-ev, 2019-11-26:
Hi @miles267,
Apologies for not updating you or others on this. The changes highlighted above did not work at all. I got no files whatsoever when I added above lines. Once I moved it back to default I was able to get Sentry and Saved archived. 

@marcone - can you guide us on how to enable Recent folder for archival, please? On other note, how far are you with implementing all archival you mentioned Aug 8? If there is anything I can do to help in testing do let me know. I have multiple Pi0W so can easily keep switching.

miles267, 2019-11-26:
Thanks @clean-ev you beat me to it!  I tried those entries without success and got not files syncing as a result.  Have also reverted back.  Functionality is restored.

Thank you also @marcone excited for this new feature.

clean-ev, 2019-11-26:
@miles267 - I added above lines to those files by remaking them with a #. I switched the remarks so default code is executed. Hope this helps.

Even more easier is to just rebuild from scratch. Take me less than 5 mins instead of making typos if you are comfortable rebuilding.

miles267, 2019-12-02:
Thanks @clean-ev; yes, I was able to revert back to the original lines of code and restore default functionality.

@marcone is there other syntax we should be using to archive all contents instead, including the RecentClips?

miles267, 2019-12-05:
@marcone would you please provide an update on the possibility of this?  Thank you.

marcone, 2019-12-05:
It's possible. I haven't done it yet. I don't know when I'll get to it.

miles267, 2019-12-05:
> It's possible. I haven't done it yet. I don't know when I'll get to it.

Understood.  Thank you for your consideration @marcone.

BeBuBi, 2019-12-08:
> Thanks @marcone.
> After I posted I found that you committed changes on Oct 30.
>
> So at this point will the below two changes be sufficient to archive RecentClips?
>
> Archive-clips.sh addition to include [...]

The possible reason your changes do not work because the path missed the '*' at the end of the directory.

The quick test on my MAC with the '*' appeared to work. Note: I just test the find cmd with my MAC not the Tesla.
DIR_COUNT=$(cd "$CAM_MOUNT"/TeslaCam && find . -maxdepth 2 -path './RecentClips/*' -type d -o -path './SavedClips/*' -type d -o -path './SentryClips/*' -type d | wc -l)
FILE_COUNT=$(cd "$CAM_MOUNT"/TeslaCam && find . -maxdepth 3 -path './RecentClips/*' -type f -o -path './SavedClips/*' -type f -o -path './SentryClips/*' -type f | wc -l)

Output:
TeslaCam % find . -maxdepth 2 -path './RecentClips/*' -type d -o -path './SavedClips/*' -type d -o -path './SentryClips/*'
./RecentClips/01-01-01
./RecentClips/02-02-02
TeslaCam % find . -maxdepth 3 -path './RecentClips/*' -type f -o -path './SavedClips/*' -type f -o -path './SentryClips/*' -type f
./RecentClips/01-01-01/1.mp4
./RecentClips/01-01-01/2.mp4

miles267, 2019-12-09:
I am no longer finding this thread from BeBuBi on Github for some reason?

marcone, 2019-12-09:
They probably deleted their comment.

BeBuBi, 2019-12-10:
> I am no longer finding this thread from BeBuBi on Github for some reason?
> […](#)

Sorry, I had to delete my comment because it was wrong.
I think the reason your "find . -maxdepth 2" command failed because the RecentClips directory does not organize clips into sub folder by times whereas the SentryClips directory does.

pi@pi-tm3:/mnt/cam/TeslaCam/RecentClips $ ls -ln
total 6248160
-rwxrwxrwx 1 0 0 29723511 Dec  9 13:26 2019-12-09_21-25-09-back.mp4
-rwxrwxrwx 1 0 0 29813173 Dec  9 13:26 2019-12-09_21-25-09-front.mp4
-rwxrwxrwx 1 0 0 29810996 Dec  9 13:26 2019-12-09_21-25-09-left_repeater.mp4
-rwxrwxrwx 1 0 0 29726854 Dec  9 13:26 2019-12-09_21-25-09-right_repeater.mp4

root@home-nc:/media/Momentus/SentryClips# ls -ln
total 32
drwxrwxrwx 2 1000 1000 4096 Dec  8 19:09 2019-12-08_10-55-28
drwxrwxrwx 2 1000 1000 4096 Dec  8 19:09 2019-12-08_12-35-40
drwxrwxrwx 2 1000 1000 4096 Dec  8 19:16 2019-12-08_13-50-42
 
I will see if i can hack the RecentClips to organize clips into sub folders by times and see if that will work.

BeBuBi, 2019-12-10:
The find cmd found the files in the directory but 0 sub directory so the archive did not kick off.

/mutable/archiveloop.log
Mon  9 Dec 23:07:05 EST 2019: Ensured cam file is mounted.
Mon  9 Dec 23:07:05 EST 2019: There are 0 event folder(s) with 208 file(s) to move in /mnt/cam.
Mon  9 Dec 23:07:06 EST 2019: Trimming free space in /mnt/cam, which has 620 extents
Mon  9 Dec 23:07:08 EST 2019: Trim complete, image now has  411 extents
Mon  9 Dec 23:07:08 EST 2019: Finished archiving.

In my thinking, the other option can be something like if archive RecentClips is enabled add 1 to DIR_COUNT ? Or wait for @marcone to complete his works on archiving all clips. :)

miles267, 2019-12-10:
Yes, unfortunately any attempt I've made to update archive-clips.sh and archiveloop, with any reference to /RecentClips files renders the archive process completely unable to run.  Once I back out my changes and reboot, archiving functionality is restored.

BeBuBi, 2019-12-11:
> Yes, unfortunately any attempt I've made to update archive-clips.sh and archiveloop, with any reference to /RecentClips files renders the archive process completely unable to run. Once I back out my changes and reboot, archiving functionality is restored.

@miles267 
[archive-clips.sh.gz](https://github.com/marcone/teslausb/files/3948093/archive-clips.sh.gz)
[archiveloop.gz](https://github.com/marcone/teslausb/files/3948094/archiveloop.gz)
The changes attached should work for you.
I tested just a while ago.
root@pi-tm3:~/bin# tail -f /mutable/archiveloop.log
Tue 10 Dec 21:17:25 EST 2019: Moved 'RecentClips/2019-12-10_19-59-29-left_repeater.mp4' to '/mnt/archive/RecentClips/.'
Tue 10 Dec 21:17:25 EST 2019: Moving 'RecentClips/2019-12-10_19-59-29-right_repeater.mp4'
Tue 10 Dec 21:17:35 EST 2019: Moved 'RecentClips/2019-12-10_19-59-29-right_repeater.mp4' to '/mnt/archive/RecentClips/.'
Tue 10 Dec 21:17:35 EST 2019: Moving 'RecentClips/2019-12-10_20-00-29-back.mp4'
Tue 10 Dec 21:17:43 EST 2019: Moved 'RecentClips/2019-12-10_20-00-29-back.mp4' to '/mnt/archive/RecentClips/.'
Tue 10 Dec 21:17:43 EST 2019: Moving 'RecentClips/2019-12-10_20-00-29-front.mp4'
Tue 10 Dec 21:17:53 EST 2019: Moved 'RecentClips/2019-12-10_20-00-29-front.mp4' to '/mnt/archive/RecentClips/.'
Tue 10 Dec 21:17:53 EST 2019: Moving 'RecentClips/2019-12-10_20-00-29-left_repeater.mp4'
Tue 10 Dec 21:18:03 EST 2019: Moved 'RecentClips/2019-12-10_20-00-29-left_repeater.mp4' to '/mnt/archive/RecentClips/.'
Tue 10 Dec 21:18:03 EST 2019: Moving 'RecentClips/2019-12-10_20-00-29-right_repeater.mp4'
Tue 10 Dec 21:18:13 EST 2019: Moved 'RecentClips/2019-12-10_20-00-29-right_repeater.mp4' to '/mnt/archive/RecentClips/.'
Tue 10 Dec 21:18:13 EST 2019: Moving 'RecentClips/2019-12-10_20-01-30-back.mp4'

miles267, 2019-12-11:
> @miles267
> [archive-clips.sh.gz](https://github.com/marcone/teslausb/files/3948093/archive-clips.sh.gz)
> [archiveloop.gz](https://github.com/marcone/teslausb/files/3948094/archiveloop.gz)
> The changes [...]

Thank you @BeBuBi.  I will try your modified files tonight.  I've reviewed the changes you've made in each file and can confirm they are not ones I've ever tried before.  Am excited :-)

BeBuBi, 2019-12-11:
> Thank you @BeBuBi. I will try your modified files tonight. I've reviewed the changes you've made in each file and can confirm they are not ones I've ever tried before. Am excited :-)

I reviewed my uploaded clips and found they are not very useful as @marcone  and others pointed out. The recent clips were about the drive home in the last hour and my awful skill at parking the car into the garage. :)

miles267, 2019-12-11:
> I reviewed my uploaded clips and found they are not very useful as @marcone and others pointed out. The recent clips were about the drive home in the last hour and my awful skill at parking the car into the garage. :)

@bebubi as crazy as it sounds, those are what I'm after... in the off chance something occurs that could prove valuable.  Thanks again.

miles267, 2019-12-12:
> I reviewed my uploaded clips and found they are not very useful as @marcone and others pointed out. The recent clips were about the drive home in the last hour and my awful skill at parking the car into the garage. :)

@BeBuBi I've just made this change last night.  Renamed original two files to *.bak in my /root/bin folder.  Copied over your updated versions with references to /RecentClips and rebooted the Pi.  After the reboot I tried to touch /tmp/archive_is_unreachable, but it didn't sync anything.  Tonight when my wife comes home and the Pi connects to wifi, will see whether it syncs a new /RecentClips folder to my file server.  Will post back whether it works or breaks the sync process altogether.

BeBuBi, 2019-12-12:
@miles267 send the file /mutable/archiveloop.log or output of "grep  RecentClips /mutable/archiveloop.log". i can see what could be the problem

miles267, 2019-12-12:
@BeBuBi OK - so I simply replaced the default archive-clips.sh and archiveloop files with the two modified versions you provided and rebooted the Pi. Since doing so the archive process has not run again.  Attached is my archiveloop.log file which shows data on Dec 11th before the 2 files were replaced, but nothing on Dec 12th after the new files were put in place and the Pi was rebooted.
[https://www.dropbox.com/s/aet4ixjm8jynit6/archiveloop.zip?dl=0](url)

miles267, 2019-12-14:
@BeBuBi confirming this is working great since chmod 755 applied to both files. However the /RecentClips do sync as expected, but are not organized into their respective date subfolders. Instead the /RecentClips folder contains a series of dated files. Not sure whether it’s even possible to organize it by dated subfolders?

@marcone would it be possible to add this working configuration to a future build? Maybe in the config to include RecentClips true, it adds these entries to archive-clips.sh and archiveloop files?

TivoRanger, 2019-12-17:
@miles267 I tried to emulate you by copying the scripts to /bin but, apparently, /bin is mounted on a read-only file system.  Did you unmount/remount as rw or what?
Sorry, but my Linux days are 20 years old :-)

miles267, 2019-12-17:
I changed file system to rw, copies both files to /root/bin then chmod 755 each file. Reboot the pi.

miles267, 2019-12-20:
@BeBuBi any idea how to modify those files to get RecentClips contents to archive in date folder like SentryClips and SavedClips? Currently it just copies all files to the root of /RecentClips folder.

[86 more replies]

closed 2021-03-28 by marcone