marcone/teslausb #116: Issue with archiving

lightdsn, 2019-07-07:
Hi, I have this setup and working fine on the car to record files on the drive but the auto archive doesn't work. It's a windows 7 machine and i changed the cifs version to 2.0 but still no files have copied over. Are there other things to check?

FlexyZ-xx, 2019-07-07:
try from CLI :

touch /mnt/archive/test
ls /mnt/archive/

and check if file is created

marcone, 2019-07-07:
Please see https://github.com/marcone/teslausb/issues/35 before filing issues.

closed 2019-07-07 by marcone

lightdsn, 2019-07-07:
Hi,

I actually arrived home and it started working. Thanks for the great instructions. Do you know a way for me to see the contents of the CAM folder over ssh? i just wanted to verify that the files got deleted as they got copied over? Thanks again!

marcone, 2019-07-07:
While archiving is in progress, the drive will be mounted at /mnt/cam
When archiving is not in progress you can manually mount it, though you should do that read-only since the car will be accessing it at the same time.