marcone/teslausb #118: Feature Request: allow for full sync on music drive.

CptanPanic, 2019-07-08:
So teslausb is working to sync down music, but I would like to be able to delete music by just deleting from the shared drive that it syncs from, also if I update tags in mp3's I would like that to get synced down also.

kdvlr, 2019-07-08:
I think overall, moving to rsync for CIFS and rsyncd operations will simplify this especially now that marcone has figured out how to kill syncing when wifi is lost. Keep the timeout on rsync less than the timeout on the connectionmonitor and it should work and we can get true bidirectional sync with delta.

gwacker, 2019-07-12:
Came to request the same. Sync of updates and deletes would be a great feature.

cyrilst, 2019-07-12:
Here is a dirty hack if you use rsync for the cam, and want to use it also for music:

In /root/bin/archiveloop, line 310, change:
`if timeout 5 [ -d "$MUSIC_ARCHIVE_MOUNT" -a -d "$MUSIC_MOUNT" ]`
to
`if timeout 5 [ -d "$MUSIC_MOUNT" ]`

In /root/.teslaCamRsyncConfig add:
`music=/path/to/your/music/`
(change the path accordingly)

Create the file /root/bin/copy-music.sh:
```sh
#!/bin/bash -eu

log "Copying music through rsync..."

source /root/.teslaCamRsyncConfig

num_files_moved=$(rsync -auvh --timeout=60 --delete --no-perms --stats --log-file=/tmp/archive-rsync-cmd.log $user@$server:$music/ $MUSIC_MOUNT/ | awk '/files transferred/{print $NF}')

if (( $num_files_moved > 0 ))
then
  log "Successfully synced music through rsync."
  /root/bin/send-push-message "TeslaUSB:" "Moved $num_files_moved music files"
else
  log "No music to copy through rsync."
fi
```
And don't forget to `chmod +x /root/bin/copy-music.sh`

Reboot, and you're good to go.

Of course, archiveloop should be changed to be able to detect whether the user wants to use CIFS or rsync for the music… but well, that's a dirty hack :)

marcone, 2019-11-07:
This is now implemented (both deletion of files that were removed on the server, and syncing of files that were updated on the server, as determined by time stamp)

closed 2019-11-07 by marcone

cpbl, 2020-03-06:
@marcone Just to be clear, when you say "this is implemented", you are not talking about rsync of music, are you? There is still no copy-music.sh for rsync as far as I can see.

@cyrilst Are you successfully using rsync for music? Have you written up documentation at all more for how to do this?

marcone, 2020-03-06:
Music sync is only implemented for CIFS