marcone/teslausb #120: Setup seemed OK, but TeslaCam doesn't work in car

tmhalen, 2019-07-09:
I went through the setup, which took a long time but appeared to complete correctly.  headless-setup.log shows 'All done'.  I've looked through the diagnostics, dmesg and archiveloop logs, but I can't see anything glaringly obvious.

I tried several different cables in the car, and have plugged into both of the front USB ports.  When I connect to my PC, the Cam drive comes up.  Connected to the PC, the LED on the Pi flashes twice every second, as I believe it should.  When I plug into the car, I get two flashes, then a burst of flashing of the LED (about 17secs by my rough count), then a pause and another two flashes.  This loop goes on as long as I leave it plugged in, and I never get the TeslaCam icon on the screen.  If I plug in my previous USB drive, the TeslaCam icon comes up within a few seconds.

Any help is appreciated!

[dmesg.txt](https://github.com/marcone/teslausb/files/3374813/dmesg.txt)
[archivelooplog.txt](https://github.com/marcone/teslausb/files/3374814/archivelooplog.txt)
[diagnostics.txt](https://github.com/marcone/teslausb/files/3374815/diagnostics.txt)

marcone, 2019-07-10:
The symptoms you describe sound like it's stuck in a reboot loop because it keeps losing power. Some people have had luck with using two usb cables (one from each port in the car to each port on the Pi). You could also try an alternate power source like a battery.

tmhalen, 2019-07-10:
With a little more troubleshooting, I have to agree.  Connecting the power-only port on the Pi, it seems to boot OK, but of course no data connection.  If I hook up both USB ports on the car to the ports on the Pi, then it works, but that leads to a mess of wires.

Don't suppose there's a Apollo-13-esque way to rearrange the boot to let it get through?

(that's mostly tongue-in-cheek)

Appreciate your response.

tmhalen, 2019-07-11:
I had the brilliant idea that perhaps the cable length was causing enough of a voltage drop to cause the reboot issue. 

Ordered a six-inch (15cm) cable off Amazon and just tried it - no luck. Still gets stuck in the loop.

tmhalen, 2019-07-12:
OK, I'm re-opening this in case anyone can help.

I found the instructions to disable HDMI  ( [https://www.jeffgeerling.com/blogs/jeff-geerling/raspberry-pi-zero-conserve-energy](url) ) and the LED [https://www.jeffgeerling.com/blogs/jeff-geerling/controlling-pwr-act-leds-raspberry-pi](url) ), and after taking those steps, the Pi seems to boot OK in the car, but the Teslacam icon never comes up on the MCU.

The drive also does not mount in Windows if I plug the Pi into the computer.  Making the changes I did required changing some directories from read-only to read-write.  Is that something that would break the drive mounting properly?

Attaching the output of the mount command in case that helps in any way.

Again - any assistance is very appreciated!

[mount.txt](https://github.com/marcone/teslausb/files/3387972/mount.txt)

marcone, 2019-07-12:
Mounting read-write should not affect the cam/music drives.
The output of the mount command is not helpful here, because the cam/music drives are not mounted on the Pi when they're exposed to the car/PC.
`setup-teslausb diagnose` might show more helpful info.

tmhalen, 2019-07-12:
Hi - thanks for replying so quickly.

diagnostics.txt attached.

[diagnostics.txt](https://github.com/marcone/teslausb/files/3388056/diagnostics.txt)

marcone, 2019-07-12:
The diagnostics show that the drive image was not exported (i.e. USB mass storage mode was not enabled) at the time the diagnostics were generated.

dmesgs shows this as the probably cause, but I don't know what it means:
```
[   37.956652] udc-core: couldn't find an available UDC - added [g_mass_storage] to list of pending drivers
[   91.828351] udc-core: couldn't find an available UDC - added [g_mass_storage] to list of pending drivers
```

tmhalen, 2019-07-12:
I created my own problem.  In editing config.txt to add the lines which disable the LED, I noticed these lines (which were the last two lines in the file previously):

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
dtoverlay=dwc2

Trying to be smart, I thought "well, I don't need audio", and so I commented them out.

Having fixed that problem, my drive now loads on the computer, but this puts me back into the reboot loop in the car. :-(

I guess, if nothing else, that narrows down somewhat where the issue is cropping up.

marcone, 2019-07-13:
Did you put both lines back, or just the "dtoverlay=dwc2" line?
Asking because audio is not needed, so it's safe to comment out (or replace "on" with "off", I suppose), but the dwc2 line is required.

tmhalen, 2019-07-13:
I did uncomment both lines. I’ll try getting rid of the audio tomorrow. I’m doubtful, but I really want to get this working!

tmhalen, 2019-07-13:
Tried commenting out the dtparam=audio=on line.  As expected, sadly, this didn't fix the issue.

marcone, 2019-07-14:
Closing, since it's a duplicate of https://github.com/marcone/teslausb/issues/9

closed 2019-07-14 by marcone