marcone/teslausb #122: Specify which notifications are sent

Phontana, 2019-07-11:
It would be nice to be able to specify when notifications are sent. For example, I would like to receive notifications when files are copied/moved, but I don’t like the ‘Archiving started at x’ notification every time the car wakes up.

kdvlr, 2019-07-13:
I pushed the change for a notification at start and end of sync. There was a feature request and I liked the option. I'll monitor this issue to see if others feel the notifications should be changed.

cgram23, 2019-07-15:
@kdvlr, instead of creating an option to specify WHICH notifications are sent, couldn't this request be accomplished by changing WHEN the notifications are sent?  

For example, if the initial notification is sent AFTER /mnt/cam is mounted, we could run a quick check on the directory to see if there are any files to copy.  If not, don't send a  notification.  This would resolve the issue of sending backup notifications whenever the car wakes up, as the directory would already be empty.

Phontana, 2019-07-15:
@cgram23 that would be a perfect solution.

marcone, 2019-07-16:
@kdvlr https://github.com/marcone/teslausb/issues/129#issuecomment-511622719 has another request to not send a notification immediately when archiving starts.

SemoTech, 2019-07-16:
Yes, exactly, best to know when/if video was uploaded, rather then just that teslausb has connected to home WiFi.

SemoTech, 2019-08-03:
@marcone any updates on the notification options? Thanks in advance.

marcone, 2019-08-03:
It's not a priority for me since I don't use notifications. Happy to accept contributions though.

kdvlr, 2019-08-04:
These requests are reasonable. I will look into it. I am traveling the next 2 weeks, but I'll try to submit a fix. 
I like the notifications when starting sync. What I'll try to do is send the start notification only when there is something to copy.

avinashjoshi, 2019-08-06:
I too would to suppress notifications sent right when archiving starts. I like what @kdvlr proposed, and would like to make another suggestion:

Create a comma separated variable in `teslausb_setup_variables.conf` (for instance `NOTIFICATION_SUBSCRIPTIONS`) that has a list of notifications that a user wants to "subscribe to". Example:
```
export NOTIFICATION_SUBSCRIPTIONS=ARCHIVE_START,ARCHIVE_SUCCESS,COPY_MUSIC_START,COPY_MUSIC_SUCCESS
```

We can potentially add more pre-defined subscriptions to this list. We could also have a boolean variable for each notification type.

Just a thought, and I'm happy to contribute / collaborate on this feature.

kdvlr, 2019-08-13:
Looking into this, sending notifications only when there is a change is not very straightforward. I actually like the archiving notifications. Admittedly, I  don't get as much because my  pi is on a battery pack, so it rarely runs out of power. 

It'll be great if someone else figures it out and creates a PR, but I will not be able to fix it.

bahree, 2019-08-22:
And I personally prefer the way the current notification is - it does given me a sense of how often does the car wake up and when. I can get some more insights on the overall behavior of how Tesla is talking to the car - not that I can make much sense of it - that is another conversation. :)

raushel, 2019-08-22:
I mean it’s a bit chatty when the car is plugged in:
[Image]

SemoTech, 2019-08-22:
Makes sense and I agree @bahree however it would be _ideal_ to have control over what notifications are sent, via the config file.  I.e.:
1) Wakeup & Connected
2) Video Data Saved to NAS
3) Music Was Synched ( <XXX> files Added/Removed )
4) USB Storage FS Repaired
5) etc...

it would be amazing if someone skilled enough could develop a Swift iOS and Android App to allow management of the Pi via Bluetooth, configuration and customization of the notifications compete with a badge system of events.  It can then be side-loaded until Apple approves it.  (Hey, one can dream, no?)

marcone, 2019-09-03:
There's always going to be room for change/improvement, but since it now doesn't send the start notification unless there's actually something to archive, I think the biggest annoyance is fixed, so I'm going to close this.

closed 2019-09-03 by marcone

Phontana, 2019-09-03:
The solution implemented is great and fixes my issue. When will it be released?

marcone, 2019-09-03:
It's already released in the sense that if you re-run
/root/bin/setup-teslausb you will get the latest code. There is no need to
flash a new prebuilt image.