marcone/teslausb #124: Wrong indication on instruction...

mickymix, 2019-07-11:
Please fix the information on this document:

https://github.com/marcone/teslausb/blob/main-dev/doc/GetShellWithoutMonitorOnWindows.md

It took me more than 20 minute to recognize that the "ssh pi@raspberrypi.local" should have been "ssh pi@teslausb.local"...

or at least, integrate with information for those using the prebuilt archive.

Thx

marcone, 2019-07-11:
I think I'll just remove that document, and the scripts that it refers to. I've always used Windows to set up my pi, but never used those instructions or those scripts. They appear to be unnecessary.

mickymix, 2019-07-12:
> I think I'll just remove that document, and the scripts that it refers to. I've always used Windows to set up my pi, but never used those instructions or those scripts. They appear to be unnecessary.

Well, I would not do this... it actually helped me a lot understanding the way a pi can be used/configured/so on...
I recognized after my message that I also made a mistake following partially the instructions to set it up from scratch and the instructions to use the prebuilt archive...
I'll close this...
Thank you for your great work ! ;-)

closed 2019-07-12 by mickymix