marcone/teslausb #125: Only copying one file, then stopping

cgram23, 2019-07-11:
I have everything up and running, and for the first day or so, all we fine.  Today, I noticed that I would receive the 'start' pushover notification, but never received a 'completed' notification.  I checked my NAS, and only 1 file was copied over.  I mounted the /mnt/cam directory on the pi, and SavedClips had several files waiting.  I pulled the diagnostics.txt, archiveloop.log, and dmesg files, but I'm not quite sure what to think of them.  Any help would be appreciated.

[archiveloop.log](https://github.com/marcone/teslausb/files/3384353/archiveloop.log)
[diagnostics.txt](https://github.com/marcone/teslausb/files/3384354/diagnostics.txt)
[dmesg.txt](https://github.com/marcone/teslausb/files/3384355/dmesg.txt)

kdvlr, 2019-07-12:
From archiveloop log, it looks like the connection monitor did not get a response when checking if wifi was connected and then killed the archive process. Check your wifi connection to the car.

marcone, 2019-07-12:
Yes, that's what the log shows (though more precisely, the connection monitor doesn't monitor the wifi connection itself, but checks whether it can reach the archive server). The connection monitor kills the archive process if it can't contact the server within 5 seconds, two times in a row. Perhaps it needs to wait longer or try more times.

cgram23, 2019-07-12:
I think the interval or window needs to be bumped up.  I'm monitoring my wifi connection from my AP, and I've  had a solid connection the whole time.  If there was any interference that delayed a response, it couldn't have been long.

In any case, I rebooted and reconnected.  All files are moving  now.  Thanks for the responses, guys!

closed 2019-07-12 by cgram23