marcone/teslausb #127: cam directory corruption

ChrisABailey, 2019-07-12:
I just got what I believe was a working setup for my model3 but came in this morning (I have it syncing to work computer) after saving two clips on my drive and one of the two directories is complete gibberish.  I only know enough Linux to be dangerous so I'm not sure how far up the chain this are broken.

1) All of the files in /mnt/TeslaCam/SavedClips/2019-07012_09_10_11 have garbage names:
<
pi@teslausb:/mnt/cam/TeslaCam/SavedClips/2019-07-12_09-10-11 $ ls -l
ls: cannot access '╕[σ]&@│!.,╓w': Input/output error
ls: cannot access ''$'\036''╔∙┬'\''╖,j.'$'\023''oA': Input/output error
ls: cannot access 'ΘNAσßg'$'\020''U.#8ü': Input/output error
ls: cannot access '┌÷⌠¡╪²º'$'\a''.φ╥': Input/output error
...
ls: cannot access 'ö'$'\a''«╚k'$'\177''╫ .v≥≥': Input/output error
total 123938904
-r-xr-xr-x 1 root root 2661178886 Dec 12  1968 #╜·≈{÷╖▒.┤╣?
d????????? ? ?    ?             ?            ? ╛½²kƒ╡ⁿ╕.▐╕═
-rwxrwxrwx 1 root root 1208819364 Jun 17  1984 ¼╞▓5╘│▄o.▌'8
-rwxrwxrwx 1 root root 1210236021 Jun 29  1906 <0`Θµº°?.å₧¬
-r-xr-xr-x 1 root root 2133385015 Sep  1  1950 ½ *Ωxv??.╝Jñ
>
2) I looked at the ArchiveLoop.log and it apears that there is a bigger problem even in the other directories.  I have edited the error log for brevity but  I believe I have a couple examles of each error message:
<
Sending IFTTT message for moved files.
Fri 12 Jul 09:16:42 EDT 2019: Ensuring cam file is mounted...
Fri 12 Jul 09:16:42 EDT 2019: Mounting /mnt/cam...
Fri 12 Jul 09:16:42 EDT 2019: Mounted /mnt/cam.
Fri 12 Jul 09:16:42 EDT 2019: Ensured cam file is mounted.
Fri 12 Jul 09:16:42 EDT 2019: Running fsck on /mnt/cam...
Fri 12 Jul 09:16:43 EDT 2019: | fsck from util-linux 2.29.2
Fri 12 Jul 09:16:43 EDT 2019: | fsck.fat 4.1 (2017-01-24)
Fri 12 Jul 09:16:43 EDT 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Fri 12 Jul 09:16:43 EDT 2019: |  Automatically removing dirty bit.
Fri 12 Jul 09:16:43 EDT 2019: | /TeslaCam/RecentClips/2019-07-12_08-47-49-front.mp4
Fri 12 Jul 09:16:43 EDT 2019: |   Contains a free cluster (823296). Assuming EOF.
Fri 12 Jul 09:16:43 EDT 2019: | /TeslaCam/RecentClips/2019-07-12_08-47-49-front.mp4
Fri 12 Jul 09:16:43 EDT 2019: |   File size is 29354669 bytes, cluster chain length is 1097728 bytes.
Fri 12 Jul 09:16:43 EDT 2019: |   Truncating file to 1097728 bytes.
...
Fri 12 Jul 09:16:45 EDT 2019: | /TeslaCam/RecentClips/2019-07-12_08-58-54-right_repeater.mp4
Fri 12 Jul 09:16:45 EDT 2019: |   File size is 29351424 bytes, cluster chain length is 0 bytes.
Fri 12 Jul 09:16:45 EDT 2019: |   Truncating file to 0 bytes.
Fri 12 Jul 09:16:45 EDT 2019: | /TeslaCam/SavedClips/2019-07-12_09-10-11
Fri 12 Jul 09:16:45 EDT 2019: |   Has a large number of bad entries. (241/254)
Fri 12 Jul 09:16:45 EDT 2019: |   Not dropping it in auto-mode.
Fri 12 Jul 09:16:45 EDT 2019: | /TeslaCam/SavedClips/2019-07-12_09-10-11/▀²f^A^W^T│╛.╢╕^L
Fri 12 Jul 09:16:45 EDT 2019: |   Bad short file name (▀²f^A^W^T│╛.╢╕^L).
Fri 12 Jul 09:16:45 EDT 2019: |   Auto-renaming it.
Fri 12 Jul 09:16:45 EDT 2019: |   Renamed to FSCK0000.000
Fri 12 Jul 09:16:45 EDT 2019: | /TeslaCam/SavedClips/2019-07-12_09-10-11/▓^G⌡^Xóö^[^].h├₧
Fri 12 Jul 09:16:45 EDT 2019: |   Bad short file name (▓^G⌡^Xóö^[^].h├₧).
Fri 12 Jul 09:16:45 EDT 2019: |   Auto-renaming it.
Fri 12 Jul 09:16:45 EDT 2019: |   Renamed to FSCK0000.001
...
Fri 12 Jul 09:16:52 EDT 2019: | /TeslaCam/SavedClips/2019-07-12_09-10-11
Fri 12 Jul 09:16:52 EDT 2019: |   "." is missing. Can't fix this yet.
Fri 12 Jul 09:16:52 EDT 2019: | /TeslaCam/SavedClips/2019-07-12_09-10-11
Fri 12 Jul 09:16:52 EDT 2019: |   ".." is missing. Can't fix this yet.
Fri 12 Jul 09:16:52 EDT 2019: | /TeslaCam/SavedClips/2019-07-12_09-10-11/FSCK0000.000
Fri 12 Jul 09:16:52 EDT 2019: |   Start cluster beyond limit (2717091573 > 1091353). Truncating file.
Fri 12 Jul 09:16:52 EDT 2019: | /TeslaCam/SavedClips/2019-07-12_09-10-11/FSCK0000.000
Fri 12 Jul 09:16:52 EDT 2019: |   File size is 2588508667 bytes, cluster chain length is 0 bytes.
Fri 12 Jul 09:16:52 EDT 2019: |   Truncating file to 0 bytes.
Fri 12 Jul 09:16:52 EDT 2019: | /TeslaCam/SavedClips/2019-07-12_09-10-11/FSCK0000.001
Fri 12 Jul 09:16:52 EDT 2019: |   Start cluster beyond limit (1873605054 > 1091353). Truncating file.
Fri 12 Jul 09:16:52 EDT 2019: | /TeslaCam/SavedClips/2019-07-12_09-10-11/FSCK0000.001
Fri 12 Jul 09:16:52 EDT 2019: |   File size is 76998045 bytes, cluster chain length is 0 bytes.
Fri 12 Jul 09:16:52 EDT 2019: |   Truncating file to 0 bytes.
Fri 12 Jul 09:16:52 EDT 2019: | /TeslaCam/SavedClips/2019-07-12_09-10-11/zy=_-╫NR.^MR₧
Fri 12 Jul 09:16:52 EDT 2019: |   Start cluster beyond limit (955267895 > 1091353). Truncating file.
Fri 12 Jul 09:16:52 EDT 2019: | /TeslaCam/SavedClips/2019-07-12_09-10-11/zy=_-╫NR.^MR₧
Fri 12 Jul 09:16:52 EDT 2019: |   File size is 4075831198 bytes, cluster chain length is 0 bytes.
Fri 12 Jul 09:16:52 EDT 2019: |   Truncating file to 0 bytes.
[... 42 lines cut ...]
Fri 12 Jul 09:25:02 EDT 2019: Unmounting /mnt/cam...
Fri 12 Jul 09:25:03 EDT 2019: Unmounted /mnt/cam.
Fri 12 Jul 09:25:03 EDT 2019: Finished archiving.
Fri 12 Jul 09:25:03 EDT 2019: Copying music...
Fri 12 Jul 09:25:04 EDT 2019: Starting music sync...
Fri 12 Jul 09:25:04 EDT 2019: Ensuring music backing file is mounted...
Fri 12 Jul 09:25:04 EDT 2019: Mounting /mnt/music...
Fri 12 Jul 09:25:04 EDT 2019: Mounted /mnt/music.
Fri 12 Jul 09:25:04 EDT 2019: Ensured music drive is mounted.
Fri 12 Jul 09:25:04 EDT 2019: Running fsck on /mnt/music...
Fri 12 Jul 09:25:04 EDT 2019: | fsck from util-linux 2.29.2
Fri 12 Jul 09:25:05 EDT 2019: | fsck.fat 4.1 (2017-01-24)
Fri 12 Jul 09:25:05 EDT 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Fri 12 Jul 09:25:05 EDT 2019: |  Automatically removing dirty bit.
Fri 12 Jul 09:25:05 EDT 2019: | Performing changes.
Fri 12 Jul 09:25:05 EDT 2019: | /dev/loop0: 113 files, 193980/934384 clusters
Fri 12 Jul 09:25:05 EDT 2019: Finished fsck on /mnt/music.
Fri 12 Jul 09:25:05 EDT 2019: Copying music from  archive...
Fri 12 Jul 09:31:03 EDT 2019: Copied 32 music file(s), skipped 1 previously-copied files, encountered 0 errors.
Fri 12 Jul 09:31:03 EDT 2019
Sending IFTTT message for moved files.
Fri 12 Jul 09:31:04 EDT 2019: Unmounting /mnt/music...
>

Any ideas on the issue and how to correct.  Do I need to destroy and recreate the camera partition?  or do I need to completely recreate the drive?

marcone, 2019-07-12:
I'd first check dmesg to see if there are any errors on the sd card itself. If that seems fine, then recreate the cam drive by deleting it and rerunning setup-teslausb.

ChrisABailey, 2019-07-16:
Thanks.
I ended up having to delete the partition and start over.  After getting it back going, and using it for only two trips, there were disk errors caught by the fsck step.  While this time they were successfully fixed, I have a feeling that to make this reliable in the long term, I'll need to use one of the battery backup options on the pi.

marcone, 2019-09-03:
Nothing actionable here

closed 2019-09-03 by marcone