marcone/teslausb #128: USB mounts on computer but not in Tesla

optaz, 2019-07-12:
Very similar to issue #66 

When I plug the device into my computer (same cable) it mounts as a USB drive.  As soon as I plug it into my car, it no longer works. 

SSH/WIFI works fine, I've attached the diagnostics file.   I've tried 2 cables as someone suggested it may not be getting enough power. 

I have a model S, same as issue #66.   Wonder if there's something related to the S and the newer firmware? It seems Model 3 users don't have issues.

optaz, 2019-07-12:
[diagnostics.txt](https://github.com/marcone/teslausb/files/3388185/diagnostics.txt)

marcone, 2019-07-13:
The diagnostics file you attached appears to have been copied from https://github.com/marcone/teslausb/issues/120#issuecomment-511042530

optaz, 2019-07-13:
My bad, sorry about that.   Attached.
[diagnostics.txt](https://github.com/marcone/teslausb/files/3388453/diagnostics.txt)

marcone, 2019-07-13:
Your diagnostics appear to show that the car isn't connected to the Pi's drive.
Normally the kernel log will show "g_mass_storage gadget: high-speed config #1: Linux File-Backed Storage" when the car (or PC) connects to the drive, and your diagnostics don't show that.
The usual reason for that to happen is using a charge-only cable, or connecting the cable to the wrong port on the Pi. You said it works with the same cables on a PC though, so no idea why it wouldn't work on the car.

optaz, 2019-07-13:
Thanks for checking.

closed 2019-07-13 by optaz