marcone/teslausb #129: OneStepSetup - No Reboot, No Shares Creation

SemoTech, 2019-07-13:
OK, this is embarrassing but here goes:

After some pain, I was eventailly successful in getting a 32GB microSD card setup correctly using OneStepSetup & 2019-06-15 firmware, with help from @marcone, see [HERE](https://github.com/marcone/teslausb/issues/70)

I then tried the SAME thing ([OneStepSetup](https://github.com/marcone/teslausb/blob/main-dev/doc/OneStepSetup.md)) with both 2019-05-05 and 2019-06-15 firmwares, and **several** 128GB & 256GB microSD cards, but I am unable to get the teslausb to create the partitions, reboot and show the CAM folder.   

- SSH works fine for access, so WiFi connection is good.
- "WIFI_ENABLED" and "TESLAUSB_SETUP_STARTED" files are created in /boot but "TESLAUSB_SETUP_FINISHED" never appears, it just gets stuck.

I have linked the "diagnostics.txt" file [HERE](https://pastebin.com/STvkqAmK)
The config file used (sanitized) can be found [HERE](https://pastebin.com/dhrZwnpj)
The [/root/bin/archiveloop] folder does not get created and the "archiveloop" process is not running.

Any Idea whats going on?  
Thanks in advance!

marcone, 2019-07-14:
I'm not sure if that "Failed to start /etc/rc.local Compatibility" message is relevant, since (1) I don't know what it even means and (2) it did start the setup-teslausb script.
Looks like setup got stuck after that though. You could try running setup-teslausb from the command line and see what it does.

SemoTech, 2019-07-16:
Thanks @marcone 

I 1st flashed your [2019-06-15 firmware](https://github.com/marcone/teslausb/releases/tag/v1.3), copied my config over, and booted it in the Pi.  Config stared and stopped short, as before.  

I then ran "apt-get update" and "apt-get upgrade" which upgraded quite a few things. Not sure if the update/upgrade had anything to do with it but it looks like Rasbian Buster was just released so I think I upgraded to that...

Then I manually ran ./setup-teslausb from the /root/bin folder as you suggested, and it worked. [This is the output](https://pastebin.com/3FUhMaXn)

Rebooted manually and got the CAM folder! Success!

**One more question**:  To setup IFTTT push notifications I followed [THIS IFTTT Tutorial](https://community.ipswitch.com/s/article/How-to-create-the-IFTTT-action).  **It works**, but seems to trigger every time the teslausb connects to the WiFi.  

_Would it be possible for notifications to trigger only if a successful actual video upload occurs?_

Thanks again!

marcone, 2019-07-16:
I did an install myself yesterday and got Stretch, but `cat /etc/os-release` will tell you if it actually ended up installed Buster. I haven't tested teslausb with Buster.

SemoTech, 2019-07-16:
Hmmm...  Looks like Stretch it is:

> pi@ELECTRA-TM3_USB:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@ELECTRA-TM3_USB:~ $

marcone, 2019-07-16:
I just ran into the same issue, and it's repeatable.
My /var/log/boot/.log ends with:
```
Detecting whether to update wpa_supplicant.conf
Grabbing main setup file.
[   27.741891] rc.local[509]:   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
[   27.754736] rc.local[509]:                                  Dload  Upload   Total   Spent    Left  Speed
100 13012  100 13012    0     0  22524      0 --:--:-- --:--:-- --:--:-- 22512
Starting setup.
reading config from /root/teslausb_setup_variables.conf
[FAILED] Failed to start /etc/rc.local Compatibility.
See 'systemctl status rc-local.service' for details.
         Starting Hold until boot process finishes up...
         Starting Terminate Plymouth Boot Screen...
```
and then systemctl says:
```
root@teslausb:~# systemctl status rc-local.service
● rc-local.service - /etc/rc.local Compatibility
   Loaded: loaded (/lib/systemd/system/rc-local.service; static; vendor preset: enabled)
  Drop-In: /lib/systemd/system/rc-local.service.d
           └─debian.conf
        /etc/systemd/system/rc-local.service.d
           └─ttyoutput.conf
   Active: failed (Result: exit-code) since Sat 2019-06-15 22:27:42 BST; 4 weeks 2 days ago
  Process: 509 ExecStart=/etc/rc.local start (code=exited, status=22)

Jun 15 22:27:40 teslausb systemd[1]: Starting /etc/rc.local Compatibility...
Jun 15 22:27:40 teslausb rc.local[509]:   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Cu
Jun 15 22:27:40 teslausb rc.local[509]:                                  Dload  Upload   Total   Spent    Left  Sp
Jun 15 22:27:41 teslausb rc.local[509]: [237B blob data]
Jun 15 22:27:42 teslausb systemd[1]: rc-local.service: Control process exited, code=exited status=22
Jun 15 22:27:42 teslausb systemd[1]: Failed to start /etc/rc.local Compatibility.
Jun 15 22:27:42 teslausb systemd[1]: rc-local.service: Unit entered failed state.
Jun 15 22:27:42 teslausb systemd[1]: rc-local.service: Failed with result 'exit-code'.
```
No idea yet what any of it means.

marcone, 2019-07-17:
This should be fixed now. The problem was that the setup script tried to download additional scripts before the network was up, resulting in all kinds of fun failure cases.

closed 2019-07-17 by marcone