marcone/teslausb #132: CIFS Sharing with Mac

luzer, 2019-07-14:
i had been dealing with CIFS share issues for a while. the local share is on a Mac. some tricks that might help others

- my WiFi password has non-standard characters that needed to be escaped with a \
- my Mac share settings (Yosemite) required setting to "SMB" (see below) that allows
- CIFS version should actually be 3.0, not 2.1 or 2.0 {pending confirmation}
[image]

closed 2019-07-14 by luzer

dragonxlwang, 2019-12-17:
might worth for @marcone to help put it on readme for people with a os x server:

need to select all the three boxes like in above picture -- i was selecting only smb (as default) and struggling to figure out how to login from linux (and hence raspberry pi).

marcone, 2019-12-17:
@dragonxlwang Pretty sure that "Share files and folders using AFP" is not required for the Pi to access the share

DMBlakeley, 2019-12-17:
I concur that "Share files and folders using AFP" is not required. Found that the CIFS 3.0 is correct for the Mac also.