marcone/teslausb #133: rclone configuration move

withinfocus, 2019-07-15:
All the setup scripts seem to expect rclone configuration to be in `/root/.config/rclone` but it's now defaulting to `/root/.rclone.conf`. Setup doesn't catch this and you end up with no rclone configuration, causing any archive operations to fail.

marcone, 2019-07-15:
The docs for rclone seem to be saying it's the other way around: "Normally the config file is in your home directory as a file called .config/rclone/rclone.conf (or .rclone.conf if created with an older version)" (from https://rclone.org/docs/)
Which version of rclone are you actually using?  Running "apt install rclone" on the Pi installs 1.35 for me, which is quite old. If that's what you did, you probably want to get the latest version of rclone instead from rclone.org or the rclone4pi project as described in https://github.com/marcone/teslausb/blob/main-dev/doc/SetupRClone.md

withinfocus, 2019-07-16:
Used the linked documentation so rclone was installed via https://raw.github.com/pageauc/rclone4pi/master/rclone-install.sh and still expected `.rclone.conf`. When the Pi boots and I SSH in I don't see configs in the expected location (https://github.com/marcone/teslausb/blob/main-dev/run/rclone_archive/archive-clips.sh#L12). Seems that explicit config was added in June and I don't see why that was done nor why the symbolic link from the root's config location isn't in `/mutable/...`.

marcone, 2019-07-16:
The change to rclone that was made in June was related to push notifications. The line you linked to was added in October 2018, in the upstream project: https://github.com/marcone/teslausb/blame/main-dev/run/rclone_archive/archive-clips.sh#L12

That's not to say that the current code is correct (I don't use rclone myself, so I don't know), but the current documentation for rclone I linked to earlier does say that /root/.config/rclone/rclone.conf is the correct location, and the setup script moves that to /mutable and replaces it with a link: https://github.com/marcone/teslausb/blob/main-dev/run/rclone_archive/configure-archive.sh

withinfocus, 2019-07-16:
I think I'll try to use `apt` to install rclone and report back.

marcone, 2019-07-16:
You'll probably get version 1.35 then, which is over 2 years old.

skiman6010, 2019-07-21:
I'm noticing the same thing on a fresh setup trying to use rclone.

withinfocus, 2019-07-23:
It didn't, and I ended up just symlinking to the (new?) config location. rclone seems totally ready to go but the archive loop isn't triggering correctly or in the right context for me. If I manually run a loop and rclone once SSHed in things are fine.

skiman6010, 2019-07-31:
Can confirm that symlinking the new config location works for me. I did a `touch /tmp/archive_is_unreachable` to simulate driving into wifi range and all my clips are syncing now. I haven't confirmed whether driving into wifi range and parking works yet.

dsestes, 2019-08-12:
Also having this issue after following the instructions in the readme. What was the symlink you set up? Noob here - must be missing something.

DMBlakeley, 2019-08-12:
i am just really stumped as I cannot get files to transfer from the pi cam folder to the rclone drive! I have gone through the process a number of times to confirm my steps with the same results.

I have carefully followed the instruction to setup rclone. In my case I was using OneDrive. I confirmed RcloneSetup instructions that I was indeed connected to the OneDrive folder. I even confirmed that files would transfer from the pi to the OneDrive folder (rclone sync -v ...).

Returning to the main instructions I set the CAM size (export camsize=100%) and then followed the "Set up the USB storage functionality" steps allowing the pi to process to completion.

Upon rebooting (unplug and replug) the CAM drive mounts and the pi goes into a 2 blink loop (my understanding is that 2 blinks means connected to wifi). To test if the file transfer works I dropped some picture files CAM/SavedClips folder. If I then ssh to the pi, sudo -i, and run "touch /tmp/archive_is_unreachable" the LED blinks rapidly, unmounts the CAM drive, goes back to 2 blinks and then the CAM drive is remounted. Unfortunately, no files are transferred.

Any suggestions?

dsestes, 2019-08-12:
@DMBlakeley There is a log file at /root/mutable/archiveloop.log that has some solid error descriptions. You'll likely discover the issue in here.

In my case, I was able to discover that the config file was not in the location that the teslausb scripts were looking for it. I struggled with the symlink but was able to successfully copy the rclone.conf file into the folder where teslausb was looking for it and the archiving kicked off from there.

DMBlakeley, 2019-08-13:
@dsestes Thanks for pointing me to the archiveloop.log file. After triggering "/tmp/archive_is_unreachable" I was able to see the problem as you noted.

teslausb is looking for the rclone.conf at /root/.config/rclone/rclone.conf

Following this path I found that rclone -> /mutable/configs/rclone, however, /mutable/configs/rclone directory did not exist. I created this directory.

Then found .teslaCamRcloneConfig in the root directory. Copied this to create the missing rclone.conf file.

Triggering "/tmp/archive_is_unreachable" found that the rclone.conf file was incomplete as it was missing entries for token, drive_id, and drive_type.  I reran "rclone config" to get these entries, updated the rclone.conf file and now all seems to work.

Detailing all of this info for @marcone info. FYI, at one point in the many times trying to get the rclone configuration to work I could have sworn that the .teslaCamRcloneConfig file contained all of the correct information.

DMBlakeley, 2019-08-15:
I ran into another interesting issue with Rclone and OneDrive. This may not be unique to OneDrive but not sure.

I started getting the following errors in the archiveloop.log:

Thu 14 Feb 10:13:03 GMT 2019: Moving /mnt/cam/TeslaCam/SavedClips/2019-08-13_13-45-56 ...
2019/02/14 10:13:04 NOTICE: Time may be set wrong - time from "graph.microsoft.com" is -4360h22m15.19676s different from this computer
2019/02/14 10:13:04 Failed to create file system for "tesla3:TeslaCam": failed to get root: InvalidAuthenticationToken: CompactToken validation failed with reason code: 80049228.
Thu 14 Feb 10:13:04 GMT 2019: Moved /mnt/cam/TeslaCam/SavedClips/2019-08-13_13-45-56.

Although the entry seemed to indicate that the file transferred I found that it had not.

After a bit of searching and trial and error found that the problem was not related to the Authentication Token but was due to the time error. As the pi zero does not maintain time when unpowered it resets to the default start time each time it is powered up.

With a bit more digging I found this article: https://raspberrypi.stackexchange.com/questions/8231/how-to-force-ntpd-to-update-date-time-after-boot. Installing the ntpdate package on the pi and executing the commands to set the clock I found that “touch /tmp/archive_is_unreachable” caused file archive to resume without to completion without errors.

I believe that an update to “archiveloop” is required to set the time once the pi is connected to the home WiFi network. Is this the right place or is there another / better file which should be updated?

marcone, 2019-08-15:
@DMBlakeley have you tried updating? I added some code last month to set the current time before archiving starts.

DMBlakeley, 2019-08-15:
I created the pi image in the last couple of days. Also checked that the archiveloop file on the pi looks to be the same that is posted in your archive. See the set_time code.

If I issue these commands from the pi shell the clock sets to the correct time:
sudo /etc/init.d/ntp stop
sudo ntpd -q -g
sudo /etc/init.d/ntp start

withinfocus, 2019-08-15:
The NTP issues were indeed another variable I fought with for Google Drive sync as the stored OAuth token would be seen as invalid due to the time being off by about 5 years. The recent forced `ntp` sync added did help but I found the device booting and still having the incorrect time (after later SSH login). Something like that really should be mandated on boot.

DMBlakeley, 2019-08-15:
One thing that I realized is that ntp sync only works if the pi is connected to the internet. Boot occurs when you wake up your Tesla. Internet connection is only realized when you enter your home WiFi.

dsestes, 2019-08-15:
This might be a stupid question -- but is there a way to edit the code to change where the scripts are looking for these files without forking the entire project and re-installing? As in, are the .sh files in the run folder sitting anywhere in the SD card that we can modify them?

DMBlakeley, 2019-08-16:
Most of what I have learned was through opening a secure shell (ssh pi@teslusb.local) with the pi via the terminal program on my Mac. I am learning as going but yes you can edit the files on the pi and if necessary do a reboot to test.

dsestes, 2019-08-16:
I think I'm 100% up and running now. My most recent error from the archiveloop.log file was:

**Failed to save config after 10 tries: Failed to create temp file for new config: open /root/.config/rclone/rcl$** (the $ was more often a long number string?)

However, if I SSH'd in, **remounted as read/write** (as shown at the end of the Readme), and triggered the archive, it would go through. 

Eventually, I added a line of code to the **archive-clips.sh** file to remount as rw prior to starting the archive. Not sure what the side affects will be, but it seems to be working. :)

marcone, 2019-08-16:
You probably want to create an rclone directory on /mutable and then link /root/.config/rclone to it

DMBlakeley, 2019-08-18:
I created a fork from you archive (DMBlakeley/teslausb) and worked on the rclone archive and time set issue. I made updates as follows:

- teslausb/run/rclone_archive/verify-and-configure-archive.sh - found that /mutable/configs/rclone need to first be created. Update mv command to include source filename.

- teslausb/run/archiveloop - found that ntp process needed to be stopped prior to forcing time set and then restarted.

I can now create an image which runs unassisted. Triggered "archive_is_unreachable" a number of times and test files transferred from the CAM folder to my OneDrive folder.

A bit new to this but believe that I have made the appropriate pull requests against your archive.

marcone, 2019-08-19:
I'm going to close this because current rclone stores the config in /root/.config/rclone/rclone.conf, not in /root/.rclone.conf as claimed above:
```
root@teslausb:~# rclone version
rclone v1.48.0
- os/arch: linux/arm
- go version: go1.12.6
root@teslausb:~# rclone config file
Configuration file doesn't exist, but rclone will use this path:
/root/.config/rclone/rclone.conf
```
This is true both for rclone installed directly from rclone.org, and rclone installed from the rclone4pi project (no surprise, since they appear to install the same binary).

closed 2019-08-19 by marcone

withinfocus, 2019-08-19:
I would call this solved by 996e4bba750d3dbf2598c1c1bad93349fe0089a7 actually. We're seeing what we posted as several different users but the key was in getting the right configuration move.