marcone/teslausb #134: how to update Tesla password

jhtown, 2019-07-16:
I just changed my tesla password, and I'm wondering is there a way I can update the tesla password saved in teslausb through ssh? 
I checked the /root/bin/tesla_api.py, but it seems that the password is not saved there. I also tried to edit the /root/teslausb_setup_variables.conf file, but it's a read-only file system that even sudo cannot edit it.
Any suggestion what the right way to update it? (Also, how to edit other configure info like share folder ip/password?)

AceTheGmodGeek, 2019-07-16:
Hey!

I'm not marcone, but, heres what should work-

ssh into the pi
type the following:
sudo -i
/root/bin/remountfs_rw

make your edits, in the teslausb_rsetup_variables.conf file, then type
/root/bin/setup-teslausb

then, reboot the raspberry pi

marcone, 2019-07-16:
I don't use it myself so I'm not sure but for the tesla password I think you either need to delete /mutable/tesla_api.json in order for it to be recreated with the new password from teslausb_setup_variables.conf, or just update it directly in the json file.

jhtown, 2019-07-17:
I updated the teslausb_setup_variables.conf with corrected password, deleted both /mutable/tesla_api.json and tesla_api.py, then ran /root/bin/setup-teslausb
still not working.
Actually, I didn't see the email and password appeared anywhere in the tesla_api.py, how does it load the email and password from the conf file?
@coredumperror can you educate me? Thank you very much,
-------------------
root@teslausb:~# /root/bin/tesla_api.py --debug get_odometer
Wed 17 Jul 05:09:42 BST 2019: Mutable data didn't exist, writing defaults...
Wed 17 Jul 05:09:42 BST 2019: Writing /mutable/tesla_api.json...
Wed 17 Jul 05:09:42 BST 2019: Retrieving new API token...
Wed 17 Jul 05:09:45 BST 2019: Unable to create access token:
Wed 17 Jul 05:09:45 BST 2019: {'response': 'authorization_required_for_txid_``'}
-----------------------

marcone, 2019-07-17:
"still not working" is not helpful. Please run "setup-teslausb diagnose" as described here: https://github.com/marcone/teslausb/issues/35

jhtown, 2019-07-18:
A quick update. I was able to use Postman to manually generate a token following this [step](https://medium.com/@jhuang5132/a-beginners-guide-to-the-unofficial-tesla-api-a5b3edfe1467), and update tesla_api.json using the access and refresh token I got from the POST. Now the tesla_api.py works.
It seems that previously although I deleted the tesla_api.json file, running the python script was still not able to generate correct token using the tesla password from the config file. I'm a newbie in coding. I'll spend some time tomorrow to look into this.

mrrross, 2019-07-18:
I've been wondering about the teslausb_setup_variables.conf file.  Can this be deleted after setup?  Concern is anyone breaking into car could plug into a USB port and get access to credentials. I don't seem to recall ha IMG to log in through a USB connection. Pardon the newbie question and thanks for your response.

jhtown, 2019-07-18:
> I've been wondering about the teslausb_setup_variables.conf file. Can this be deleted after setup? Concern is anyone breaking into car could plug into a USB port and get access to credentials. I don't seem to recall having to log in through a USB connection. Pardon the newbie question and thanks for your response.

I'm still learning this, but I think you can delete the tesla username and password from Config file after the token has been generated. The tesla API should operate normally with token. 
But of course, if the thief is really good at this, he can extract the token and use it to get access to your tesla too. But at least, you didn't expose your email and password. And if you do lose your tesla USB, change your password immediately.

dprus, 2019-08-03:
I'm having this exact same problem.

dopeghoti, 2019-09-29:
Have you tried deleting the token JSON file, updating the `.conf` file with your new credentials, and re-running the teslausb-setup script?

bahree, 2019-10-09:
Assuming this happen - someone breaks in, is a geek like many of us and wants to go after the token - the best way to handle that would be for the owner to login and change their Tesla credentials. When you do that, it will automatically invalidate all tokens. So if someone does get their hands on it, it wont be any use (from that threat vector perspective).

marcone, 2019-11-02:
Since this was a question rather than an issue, I'm going to close this.

closed 2019-11-02 by marcone