marcone/teslausb #135: Raspberry Pi randomly stops working

AceTheGmodGeek, 2019-07-16:
Hello!

So, recently I have been using this, and its been amazing. However, this has happened twice, where I'm on the road, and it's blinking rapidly, like it's uploading something; but I'm not at home, so it's not even connected to WiFi. I have to unplug and replug it for it to be recognized again.

I'm not sure if the diagnostics would be useful since I had to unplug it from the car and replug it in, then unplug to plug into the computer.

[diagnostics.txt](https://github.com/marcone/teslausb/files/3395070/diagnostics.txt)

gkwok1, 2019-07-16:
I actually have similar issues as well but never got around figuring out what caused it.

marcone, 2019-07-17:
Weird. The fast blinking is only started after it detects that the archive is reachable. Are you using a mobile hotspot or something?

AceTheGmodGeek, 2019-07-17:
Nope, I don’t use a mobile hotspot. I’m guessing that it got tricked that there was something to upload, and it was trying to upload, and was stuck maybe?

I use a portable battery if that changes anything.

marcone, 2019-07-17:
Just to be clear, does it *start* the fast blink pattern while you're on the road, or was it possibly fast blinking the whole time, but you didn't notice it until later?
(it only checks whether there is something to upload after it has detected that the archive destination is reachable, so I it should never start blinking rapidly while you're away from the network)

AceTheGmodGeek, 2019-07-17:
It does start blinking before I leave my house; but it's definitely already been at least 10 hours since I have used the car, and it is connected to a portable battery, therefore, it should have uploaded what is on it by then.

emerout, 2019-07-17:
> It does start blinking before I leave my house; but it's definitely already been at least 10 hours since I have used the car, and it is connected to a portable battery, therefore, it should have uploaded what is on it by then.

Same behavior for me : when I park at home in the evening, archiving starts end ends successfully. When I leave home in the morning, the drive does not mount and the pi is blinking (until I reboot it).

marcone, 2019-07-17:
If it's fast-blinking in the morning, that either means it was fast-blinking the entire night (i.e. it didn't finish from the previous day), or the Pi restarted when you started the car (possible even with a portable battery if you're not using a data-only cable for the USB connection).
Then if it continues blinking for a long time after you've moved out of wifi range, there's either a problem in the code that detects and aborts the archive operation, or you haven't updated since I added that code.

AceTheGmodGeek, 2019-07-17:
I installed this about 25 days ago, so unless there is an update I didn’t get, that might be the solution?

As for the data only cable, I have a power only going into the power bank, which feeds into the raspberry pi, and I have a data and power that comes directly from the car- I don’t have a data only cable.

marcone, 2019-07-17:
The code I'm referring to was added 17 days ago, so you might not have it then. You should be able to just run `setup-teslausb selfupdate; setup-teslausb` to get the latest.

As for power-only and data-only cabling: when I was using a battery and a full USB cable for the data connection, I noticed that when the car was off, the car was actually pulling power from the battery: I had a USB power meter between the battery and the Pi, and it showed power draw going up to 400 mA when the car was turned off. Needless to say that made the battery drain a lot faster than it should. I made a data-only cable by putting a small strip of packing tape over the 5V contact in the USB cable.

emerout, 2019-07-17:
I have only one USB cable for power and data.

I have updated few days ago. Only this commit is missing : https://github.com/marcone/teslausb/commit/77a1c44cca17536bf979461e81e6bb2528ef9b32

The archiving was completed before the car went asleep. 
(I have moved the archive notification to the end of the archiving process, and I received this notification).

I think the Pi boot up went I opened the car, I leave home and  I waited a couple of minutes before I restarted it.

Maybe a bug with a code that detects wifi loss, I can help in debugging.

hokiegir1, 2019-07-19:
I've got a similar issue, except that mine starts the transfer, and then fails at some point (I've tried with Sentry on, and now I've added the username/password for the Tesla API to see if that solved it). We used to leave Sentry on at home and when it would fail with that enabled, I'd have to either unplug the drive or reboot the car to get it to connect/function again and it would work fine until it connected to wifi again, when I'd have the same issues -- transfer start, bomb out in the middle, fail to restart properly. 

Now, with Sentry off at home but the API info entered, the drive did start this morning and was working -- so that's that part of the problem solved. It's just keeping it connected to finish the transfer, apparently, that is the problem. I don't currently have a battery hooked up. I did *just* add the API info, so that's only been a 1 day test. I entered it just like in the sample -- no parentheses or anything. Did I need to add those? I can check the archive log tonight to see what that says. It did do the fast "transfer" blink when I left this morning, but fixed itself within a couple minutes after I was out of range.

closed 2019-10-25 by marcone