marcone/teslausb #137: Pi Zero not recognized

jhaltd, 2019-07-19:
I've been banging my head against the wall for the last two weeks...  I think that I've followed all the instructions, the image has been successfully loaded and looks fine from both my Windows and Linux boxes.  But, my Model 3 will not acknowledge the Raspberry Pi Zero.  Here's my diagnostics, via GIST, if anyone could please offer some insight:  https://gist.github.com/jhaltd/dae9434b4f1cb58a99a6440883ed0380

Thank you so much,

Dunc

marcone, 2019-07-20:
Does it work (the "CAM" drive shows up) when you connect it to a PC?
Your diagnostics seem to show the car isn't attempting to connect, which is usually due to either using a charge-only cable, or plugging the cable into the power-only port on the Pi.

Nashable, 2019-07-20:
I'm having the same issue as @jhaltd. Model 3 (3 weeks old with latest software). The partitions work fine when plugged into a PC. Tried multiple cables and tried both single cable in USB port (not power port) and then dual cable (both directly into the second front port and then into a battery/charging device)

I experience sync of music files to the PI and have no issues SSHing when its on the wi-fi. The car never sees the drive and I can't record dashcam footage or play USB music.

The card is 512gb with 25% CAM and 75% Music.