marcone/teslausb #141: Music sync slow

ken830, 2019-07-21:
Marcone -- I've been following for awhile. Thanks for your great work.

Finally moved over from cimryan/teslausb and setting up a new SD card with the latest v1.3 image. This is the first boot with the RaspberryPi plugged into my Windows laptop. No USB drives appear in Windows and LED is constantly flashing (about 4 flash/sec), but archiveloop.log indicates that it started copying music. That was over 2 hours ago. I checked disk usage on /mnt/music and I'm at 6.2GB. My music share is over 13GB in size. Another 3 hours to go. I didn't even attempt to include my audiobooks because that would be another 20GB!

Two questions:

1) Why is WiFi performance so slow? The Pi is about 2 feet away from my Ubiquiti AP which has giabit connection to my Synology NAS where the music share is located. Synology network monitor indicates the transfer rate is under 1MB/s, which is consistent with what I'm seeing so far (<3GB/hour)

2) Is there a way to safely prevent the initial music sync and instead mount the music drive and make it available to the laptop via USB so that I can copy the music folder over USB? I might want to do this the next time I have to redo the setup.

marcone, 2019-07-21:
No idea why it's so slow for you. The Pi is not very fast, but I'm getting about 2.6 MB/sec while 30 ft away from my wifi access point, and "iperf" reports 28.5 Mbits/sec between the Pi and my server from that same distance (and 36 Mbit/sec when right next to the AP).

There's no "nice" way to avoid the initial sync, but you could do one of the following:
- disconnect your NAS when the Pi boots, so it can't find it and skips music sync
- ssh into the Pi while setup is still in progress, and delete the `/mnt/musicarchive` folder after it is created. When the Pi reboots, it will fail to find the folder and not sync music. You can then copy your music manually and recreate /mnt/musicarchive
- configure it without music sync first, then manually copy the music, then configure music sync and rerun setup-teslausb

ken830, 2019-07-22:
Even at 2.6 MB/sec, it would have still taken 1.5 hours. Mine just finished, but your suggestions are good for the next time I rebuild it. Thanks!

closed 2019-07-22 by ken830