marcone/teslausb #142: Cannot Mount Apple Time Capsule with Attached Storage

THESHULK, 2019-07-22:
I have an apple time capsule with an attached hard drive (named WD Easystore 4TB). I am able to connect to the attached drive from my Mac using smb protocol by hitting Cmd+K from a finder window and entering in smb://<time capsule IP address>/WD Easystore 4TB. However, no matter what I try, the pi cannot successfully mount the drive. I keep getting the following error:

Mon 22 Jul 14:39:37 CDT 2019 : verify-archive-configuration: The archive server is reachable.
Mon 22 Jul 14:39:37 CDT 2019 : verify-archive-configuration: Verifying that the archive share is mountable...
Mon 22 Jul 14:39:37 CDT 2019 : verify-archive-configuration: Mount command-line:
Mon 22 Jul 14:39:37 CDT 2019 : verify-archive-configuration: mount -t cifs //<IP address>/WD Easystore 4TB /tmp/archivetestmount -o vers=2.0,credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777
Mon 22 Jul 14:39:37 CDT 2019 : verify-archive-configuration: STOP: The archive couldn't be mounted with CIFS version 2.0. Try specifying a lower number for the CIFS version like this:
Mon 22 Jul 14:39:37 CDT 2019 : verify-archive-configuration:   export cifs_version=2.1
Mon 22 Jul 14:39:37 CDT 2019 : verify-archive-configuration: Other versions you can try are 2.0 and 1.0

These are the settings I'm using:

export ARCHIVE_SYSTEM=cifs
export archiveserver=IP address
export sharename=WD\ Easystore\ 4TB
export shareuser=user
export sharepassword=password
export cifs_version="2.0"       #I've tried 3.0, 2.1, 2.0 and 1.0
export camsize=100%

One thing I know is that time capsule uses smb1, so I tried setting the cifs version to 3.0, 2.1, 2.0 and 1.0 but I always get the same result. I'm wondering if maybe TeslaUSB doesn't support smb1?

Thanks in advance!

marcone, 2019-07-22:
I assume that connecting to a Time Capsule is similar to connecting to a shared drive on a Mac, and I've never been able to successfully connect to a Mac.
In any case this isn't really a teslausb issue, but more of a Raspbian or CIFS issue, so check relevant forums I suppose.

THESHULK, 2019-07-22:
Thanks for the prompt reply. I'll check some other forums as well. Can you tell me what ARCHIVE_SYSTEM values teslausb supports? I tried both smb and smbfs and it didn't accept either one.

marcone, 2019-07-22:
valid values are : cifs, rsync, rclone and none

THESHULK, 2019-07-23:
I was able to successfully mount the drive by manually running this command:

mount -t cifs "//[IP address]/WD Easystore 4TB" /tmp/archivetestmount -o vers=1.0,user=username,pass=password,iocharset=utf8,file_mode=0777,dir_mode=0777,sec=ntlm

This is nearly identical to the command that the headless setup runs with the exception of the credentials being passed in directly (the credentials file didn't get created for some reason), the double quotes around the server directory, and the sec=ntlm option. Is it possible to update the setup to allow for the sec=ntlm option and to surround the server directory in double quotes?

marcone, 2019-07-23:
Apparently quotes are note allowed in in /etc/fstab, but instead spaces should be escaped/replaced with `\040`, so perhaps putting:
```
export sharename='192.168.1.1/WD\040Easystore\0404TB'
```
in the setup conf file will work? Would be great if you could try that and report back.
Also, is the `sec=ntlm` actually required, and if so, is that a Mac/TimeCapsule thing, or is it specific to your server?

THESHULK, 2019-07-23:
I'll give that a try. From what I found in forums,"sec" is the security mode that determines how passwords are encrypted between server and client. ntlm used to be the default but the default is now ntlmssp. So if you don't specify the sec then it gets defaulted to ntlmssp. My time capsule is about 6 years old, so I tried sec=ntlm and that worked.

THESHULK, 2019-07-23:
Replacing the spaces with \040 did not work. It ended up looking like this:

`mount -t cifs //192.168.1.70/WD040External0404TB`

So I just renamed my external drive to remove the spaces and it still doesn't work. I'm convinced that sec=ntlm is needed to mount the drive. Is there any way I can update the mount command to add sec=ntlm? I downloaded the source code and I see that it's called in verify-archive-configure.sh but I don't know how I can update the code in the prebuilt image.

marcone, 2019-07-23:
> Replacing the spaces with \040 did not work. It ended up looking like this:
> 
> `mount -t cifs //192.168.1.1/WD040External0404TB`
 
Did you use single quotes around the string in the teslausb_setup_variables.conf file, like I did in my comment above?

> So I just renamed my external drive to remove the spaces and it still doesn't work. I'm convinced that sec=ntlm is needed to mount the drive. Is there any way I can update the mount command to add [...]

verify-archive-configuration.sh and configure-archive.sh are downloaded at install time, they are not part of the prebuilt image. You would have to create your own fork, modify the scripts, then point the teslausb_setup_variables.conf at your fork.

THESHULK, 2019-07-23:
I may have forgotten the single quotes.

I'll create my own fork and let you know how it goes. Thanks.

THESHULK, 2019-07-23:
Looks like I don't have permission to commit my new branch with the change. Is this something you can grant to me?

marcone, 2019-07-23:
I shouldn't have to grant anything. If you click on the "fork" button at the top right you create your own fork, independent of this one. Your fork will be at https://github.com/THESHULK/teslausb, where you can then make your own changes.

THESHULK, 2019-07-23:
Oh ok. I was trying to create a branch. Thanks!

THESHULK, 2019-07-23:
Good news! It worked.

`Tue 23 Jul 18:01:12 CDT 2019 : verify-archive-configuration: Mount command-line: 
Tue 23 Jul 18:01:12 CDT 2019 : verify-archive-configuration: mount -t cifs //192.168.1.1/External4TB /tmp/archivetestmount -o vers=1.0,credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,sec=ntlm
Tue 23 Jul 18:01:25 CDT 2019 : verify-archive-configuration: The archive share is mountable.
Tue 23 Jul 18:01:25 CDT 2019 : configure-archive: Configuring the archive...
Tue 23 Jul 18:01:25 CDT 2019 : configure-archive: Configured the archive.`

I'll have to add in a variable to the conf to make the sec parameter configurable. Do you think you'll merge this into your fork?

marcone, 2019-07-24:
Yes, happy to accept contributions.

THESHULK, 2019-07-24:
So I was able to get headless setup to complete successfully and got the "All done" output in the teslausb-headless-setup.log. However, my saved clips from the cam partition never got synced. I checked the boot.log and I found that the following error:

`[^[[0;1;31mFAILED^[[0m] Failed to mount /mnt/archive.^M
See 'systemctl status mnt-archive.mount' for details.`

Running `grep archive /etc/fstab` shows:
`//192.168.1.1/External4TB /mnt/archive cifs vers=1.0,credentials=/root/.teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777 0`

So I have two questions:

1. Where is 'systemctl status mnt-archive.mount' and how do I access it to see additional details?
2. What's responsible for writing to /etc/fstab? I'd like to update it to include sec=ntlm.

marcone, 2019-07-24:
> 1. Where is 'systemctl status mnt-archive.mount' and how do I access it to see additional details?

It's a command you run on the Pi

marcone, 2019-07-24:
I just modified configure-archive.sh so it handles spaces in share names.

THESHULK, 2019-07-24:
Awesome, thanks for making that change. I update configure-archive.sh to use ntlm but I'm still getting this error:

`[   23.470487] rc.local[437]: /etc/rc.local: line 24: /boot/teslausb-headless-setup.log: Read-only file system^M
can't write to /boot/teslausb-headless-setup.log^M
Detecting whether to update wpa_supplicant.conf^M
[^[[0;1;31mFAILED^[[0m] Failed to mount /mnt/archive.^M
See 'systemctl status mnt-archive.mount' for details.^M
[^[[0;1;33mDEPEND^[[0m] Dependency failed for Remote File Systems.^M
         Starting rng-tools.service...^M
         Starting Permit User Sessions...^M
         Starting LSB: Switch to ondemand cpu…or (unless shift key is pressed)...^M
         Starting LSB: Start NTP daemon...^M
         Starting LSB: Starts syslogd...^M
Powering off HDMI^M
[^[[0;32m  OK  ^[[0m] Started /etc/rc.local Compatibility.^M
[^[[0;32m  OK  ^[[0m] Started rng-tools.service.^M
[^[[0;32m  OK  ^[[0m] Started Permit User Sessions.^M
[^[[0;32m  OK  ^[[0m] Started OpenBSD Secure Shell server.^M
[^[[0;32m  OK  ^[[0m] Started LSB: Switch to ondemand cpu …rnor (unless shift key is pressed).^M
         Starting Hold until boot process finishes up...^M
         Starting Terminate Plymouth Boot Screen...^M`

Running `systemctl status mnt-archive.mount` returns the following:

`mnt-archive.mount - /mnt/archive
   Loaded: loaded (/etc/fstab; generated; vendor preset: enabled)
   Active: inactive (dead) since Tue 2019-07-23 22:52:24 CDT; 2h 18min ago
    Where: /mnt/archive
     What: //192.168.1.1/External4TB
     Docs: man:fstab(5)
           man:systemd-fstab-generator(8)
  Process: 439 ExecMount=/bin/mount //192.168.1.70/External4TB /mnt/archive -t cifs -o vers=1.0,credentials=/root/.teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,sec=ntlm (code=exited, status=32)`

So I see that it is inactive (which I'm not sure what that refers to) and I'm getting an exit code 32. I'll do some digging to see what I can find but any tips you might have would be appreciated.

marcone, 2019-07-24:
`setup-teslausb diagnose` might give more info

marcone, 2019-07-25:
@THESHULK what happens if you leave out the "vers=1.0" part of the mount options entirely? Does it still work then? I'm wondering if teslausb should even be explicitly specifying a version, or if it should just rely on the auto-negotiation that's part of the protocol.

THESHULK, 2019-07-25:
I will give that a try. I still can’t get backups to work though and the
dash cam won’t stay connected to the car. I did manage to get one backup to
work after I reflashed and manually copied dash cam clips to the cam mount.
It’s been pretty frustrating to tell you the truth.

THESHULK, 2019-07-25:
@marcone removing the version didn't work:

> Thu 25 Jul 12:22:36 CDT 2019 : verify-archive-configuration: mount -t cifs //192.168.1.1/External4TB /tmp/archivetestmount -o credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,sec=ntlm
> Thu 25 Jul 12:22:37 CDT 2019 : verify-archive-configuration: STOP: The archive couldn't be mounted with CIFS version 1.0. Try specifying a lower number for the CIFS version like this:
> Thu 25 Jul 12:22:37 CDT 2019 : verify-archive-configuration:   export cifs_version=2.1
> Thu 25 Jul 12:22:37 CDT 2019 : verify-archive-configuration: Other versions you can try are 2.0 and 1.0

THESHULK, 2019-07-25:
@marcone I have good news. I got the archive backup working now. I saw in another thread that you have your pi powered by a power bank, so I decided to give that a try and that did the trick. The only change I had to make to the source code was to add `sec=ntlm` to the mount options in the configure-archive.sh and the verify-archive-configuration.sh files. If you could incorporate this change to your fork that would be great. I think you'd need to add another configuration option in the teslausb_setup_variables.conf for people to set the sec to ntlm if they need to override the default value of ntlmssp.

For your power bank that you have plugged into your pi, are you just bringing it inside every so often to recharge it or do you have the power bank plugged into the other USB port in the car?

marcone, 2019-07-25:
I don't use a powerbank anymore, but now have an adapter that provides permanent(*) power from the ODB2 port. That's not an option for everyone, since apparently the Model 3 does not have that. When I was still using the battery I would bring it in to recharge every so often.

(*) sometimes even the supposed-to-be-always-powered OBD2 port appears to lose power while the car is parked. I've been considering getting a battery with passthrough charging instead.

THESHULK, 2019-07-25:
Ok got it. Thanks so much! One last question: is it possible to add a subdirectory to the sharename in the conf (i.e. `export sharename=External4TB/TeslaCam`)? And can I do this without reflashing?

marcone, 2019-07-25:
Yes, you can specify a path after the share name, and after initial setup you can simply add that to the relevant /etc/fstab entry

THESHULK, 2019-07-25:
Awesome. Thanks again. You've been such a great help!

marcone, 2019-07-27:
I just pushed a change to the "cifsmount" branch that lets you specify the "sec=" option (using `export cifs_sec=<option>` in the setup .conf file, but also attempts to automate the process of trying to find what works by trying several combinations of "vers" and "sec" mount options, if you don't specify any in the setup .conf. Hopefully that will work for all but the most exotic cases.
Give it a try if you want (in the setup .conf, set `export BRANCH=cifsmount`).

BadPirate, 2019-07-27:
Marcone, fresh install from OSX to try and point at a Time Capsule -- Tried using the cifsmount branch, and the script banged it's head on lots of options before failing out.  Manually copying and pasting `mount -t cifs '//AirportName.local/Data' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,vers=1.0,sec=ntlm'` was able to mount however.

After mounting and viewing files at /tmp/archivetestmount, I tried running setup script again and same result...
```
verify-archive-configuration: mount -t cifs '//HNL.local/Data' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,vers=1.0,sec=ntlm'
Unable to find suitable address.
verify-and-configure-archive: STOP: no working combination of vers and sec mount options worked
verify-archive-configuration: STOP: no working combination of vers and sec mount options worked
```

marcone, 2019-07-27:
Huh, so it didn't work as part of the script, but just copy/pasting the command that the script used worked?

BadPirate, 2019-07-27:
@marcone Yeah.. Strange huh?  This is for the built in drive on the time capsule, not an external plugged in one, but yeah.  Copying the command after `verify-archive-configuration:` and pasting into terminal mounted /tmp/archivetestmount and it was usable :/

BadPirate, 2019-07-27:
@marcone - Okay, seems that the .local version of the address is unreliable.  Using IP address worked and setup correctly.

THESHULK, 2019-07-27:
@marcone I haven't tried out the new branch you made because I'm actually having issues and I don't want to lose any logs before I troubleshoot. I'm noticing that the car stops detecting the dash cam after about 10-15 minutes of driving. I thought that I had fixed this problem by supplying the pi with an external power source but it appears that's not the case. Do you think this is related to the mounting changes I made? Is there a particular log file I can check that might reveal the problem? Thanks so much.

marcone, 2019-07-27:
I doubt it has anything with the cifs mount option changes, but "setup-teslausb diagnose" prints most relevant info.

THESHULK, 2019-07-27:
Do I need to run that command from a specific directory? I'm getting the following:

```
pi@teslausb:~ $ setup-teslausb diagnose
-bash: setup-teslausb: command not found
```

THESHULK, 2019-07-27:
Ok so I found these logs occurred while I was driving and approximately when the dash cam became unavailable to the car. A couple of things I find strange. 

1. The date and time is completely wrong at the beginning but then it fixed itself even though it says "failed to set time."
2. It's trying to run the archive while I'm driving. Shouldn't it only do that when it's connected to the network? I was no where near my house.

> Thu  3 Nov 12:17:04 CDT 2016: Starting archiveloop...
> Thu  3 Nov 12:17:04 CDT 2016: Ensuring cam file is mounted...
> Thu  3 Nov 12:17:05 CDT 2016: Mounting /mnt/cam...
> Thu  3 Nov 12:17:05 CDT 2016: Mounted /mnt/cam.
> Thu  3 Nov 12:17:05 CDT 2016: Ensured cam file is mounted.
> Thu  3 Nov 12:17:05 CDT 2016: Running fsck on /mnt/cam...
> Thu  3 Nov 12:17:06 CDT 2016: | fsck from util-linux 2.29.2
> Thu  3 Nov 12:17:09 CDT 2016: | fsck.fat 4.1 (2017-01-24)
> Thu  3 Nov 12:17:09 CDT 2016: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
> Thu  3 Nov 12:17:09 CDT 2016: |  Automatically removing dirty bit.
> Thu  3 Nov 12:17:09 CDT 2016: | Performing changes.
> Thu  3 Nov 12:17:10 CDT 2016: | /dev/loop0: 95 files, 97/3785199 clusters
> Thu  3 Nov 12:17:10 CDT 2016: Finished fsck on /mnt/cam.
> Thu  3 Nov 12:17:10 CDT 2016: Unmounting /mnt/cam...
> Thu  3 Nov 12:17:10 CDT 2016: Unmounted /mnt/cam.
> Thu  3 Nov 12:17:10 CDT 2016: Connecting usb to host...
> Thu  3 Nov 12:17:10 CDT 2016: Connected usb to host.
> Thu  3 Nov 12:17:10 CDT 2016: Waiting for archive to be reachable...
> Thu  3 Nov 12:17:10 CDT 2016: Archive is reachable.
> Thu  3 Nov 12:17:10 CDT 2016: Waiting for time to be set by ntpd
> Fri 26 Jul 18:15:00 CDT 2019: Time still not set, attempting to force it
> Fri 26 Jul 18:15:00 CDT 2019: Failed to set time
> Fri 26 Jul 18:15:21 CDT 2019: Archiving...
> Fri 26 Jul 18:15:21 CDT 2019: Disconnecting usb from host...
> Fri 26 Jul 18:15:21 CDT 2019: Disconnected usb from host.
> Fri 26 Jul 18:15:21 CDT 2019: Ensuring cam archive is mounted...
> Fri 26 Jul 18:15:21 CDT 2019: Mounting /mnt/archive...
> mount error(101): Network is unreachable
> Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
> Fri 26 Jul 18:15:26 CDT 2019: Failed to mount /mnt/archive.
> Fri 26 Jul 18:15:26 CDT 2019: Sleeping before retry...
> Fri 26 Jul 18:15:27 CDT 2019: Retrying...
> Fri 26 Jul 18:15:27 CDT 2019: Mounting /mnt/archive...
> mount error(101): Network is unreachable
> Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
> Fri 26 Jul 18:15:27 CDT 2019: Failed to mount /mnt/archive.
> Fri 26 Jul 18:15:27 CDT 2019: Sleeping before retry...
> Fri 26 Jul 18:15:28 CDT 2019: Retrying...
> Fri 26 Jul 18:15:28 CDT 2019: Mounting /mnt/archive...
> mount error(101): Network is unreachable
> Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
> Fri 26 Jul 18:15:28 CDT 2019: Failed to mount /mnt/archive.
> Fri 26 Jul 18:15:28 CDT 2019: Sleeping before retry...
> Fri 26 Jul 18:15:29 CDT 2019: Retrying...
> Fri 26 Jul 18:15:29 CDT 2019: Mounting /mnt/archive...
> mount error(101): Network is unreachable
> Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
> Fri 26 Jul 18:15:29 CDT 2019: Failed to mount /mnt/archive.
> Fri 26 Jul 18:15:29 CDT 2019: Sleeping before retry...
> Fri 26 Jul 18:15:30 CDT 2019: Retrying...
> Fri 26 Jul 18:15:30 CDT 2019: Mounting /mnt/archive...
> mount error(101): Network is unreachable
> Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
> Fri 26 Jul 18:15:30 CDT 2019: Failed to mount /mnt/archive.
> Fri 26 Jul 18:15:30 CDT 2019: Sleeping before retry...
> Fri 26 Jul 18:15:31 CDT 2019: Retrying...
> Fri 26 Jul 18:15:32 CDT 2019: Mounting /mnt/archive...
> mount error(101): Network is unreachable
> Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
> Fri 26 Jul 18:15:32 CDT 2019: Failed to mount /mnt/archive.
> Fri 26 Jul 18:15:32 CDT 2019: Sleeping before retry...
> Fri 26 Jul 18:15:33 CDT 2019: Retrying...
> Fri 26 Jul 18:15:33 CDT 2019: Mounting /mnt/archive...
> mount error(101): Network is unreachable
> Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
[... 29 lines cut ...]
> Fri 26 Jul 18:15:37 CDT 2019: Connected usb to host.
> Fri 26 Jul 18:15:38 CDT 2019: Waiting for archive to be unreachable...
> Fri 26 Jul 18:15:38 CDT 2019: Sleeping before retry...
> Fri 26 Jul 18:15:39 CDT 2019: Retrying...
> Fri 26 Jul 18:15:39 CDT 2019: Sleeping before retry...
> Fri 26 Jul 18:15:40 CDT 2019: Retrying...
> Fri 26 Jul 18:15:40 CDT 2019: Sleeping before retry...
> Fri 26 Jul 18:15:41 CDT 2019: Retrying...
> Fri 26 Jul 18:15:41 CDT 2019: Sleeping before retry...
> Fri 26 Jul 18:15:42 CDT 2019: Retrying...
> Fri 26 Jul 18:15:42 CDT 2019: Sleeping before retry...
> Fri 26 Jul 18:15:43 CDT 2019: Retrying...
> Fri 26 Jul 18:15:43 CDT 2019: Sleeping before retry...
> Fri 26 Jul 18:15:44 CDT 2019: Retrying...
> Fri 26 Jul 18:15:44 CDT 2019: Sleeping before retry...
> Fri 26 Jul 18:15:45 CDT 2019: Retrying...
> Fri 26 Jul 18:15:46 CDT 2019: Sleeping before retry...
> Fri 26 Jul 18:15:47 CDT 2019: Retrying...
> Fri 26 Jul 18:15:47 CDT 2019: Sleeping before retry...
> Fri 26 Jul 18:15:48 CDT 2019: Retrying...
> Fri 26 Jul 18:15:48 CDT 2019: Sleeping before retry...
> Fri 26 Jul 18:15:49 CDT 2019: Retrying...
> Fri 26 Jul 18:15:49 CDT 2019: Attempts exhausted.
> Fri 26 Jul 18:15:49 CDT 2019: Archive is unreachable.
> Fri 26 Jul 18:15:49 CDT 2019: Waiting for archive to be reachable...

marcone, 2019-07-27:
That first line:
```
Thu 3 Nov 12:17:04 CDT 2016: Starting archiveloop...
```
is printed shortly after the Pi boots up. At this point you presumably just turned on the car and are still in wifi range, which is why it starts archiving:
```
Thu 3 Nov 12:17:10 CDT 2016: Waiting for archive to be reachable...
Thu 3 Nov 12:17:10 CDT 2016: Archive is reachable.
Thu 3 Nov 12:17:10 CDT 2016: Waiting for time to be set by ntpd
Fri 26 Jul 18:15:00 CDT 2019: Time still not set, attempting to force it
Fri 26 Jul 18:15:00 CDT 2019: Failed to set time
Fri 26 Jul 18:15:21 CDT 2019: Archiving...
```
The "failed to set time" message is likely because at that point ntpd just started running as well, and locked out the instance of ntpd that the script tried to run to force time to be set. It doesn't matter, since the time was set anyway.
At this point it tries to start archiving, but fails because now you're out of wifi range:
```
Fri 26 Jul 18:15:21 CDT 2019: Mounting /mnt/archive...
mount error(101): Network is unreachable
```
it retries a few more times, then gives up:
```
Fri 26 Jul 18:15:37 CDT 2019: Attempts exhausted.
Fri 26 Jul 18:15:37 CDT 2019: Failed to mount cam archive.
Fri 26 Jul 18:15:37 CDT 2019: Couldn't connect archive, skipping archive step
```
and then it connects the drives to the car:
```
Fri 26 Jul 18:15:37 CDT 2019: Connecting usb to host...
Fri 26 Jul 18:15:37 CDT 2019: Connected usb to host.
```

Based on this section of log, what you should have seen in the car is that after turning on the car,  the dashcam icon wasn't showing at first, and then started showing about 40 seconds after you moved out of wifi range.

THESHULK, 2019-07-27:
Ok looking back at my driving history you are correct. That's the time I left the house and it did take a minute for the dash cam icon to show up. So then there are no other logs from the rest of my drive last night. The next entry after this section of logs is from today. Is there anything else I can check?

marcone, 2019-08-03:
I think the original issue reported here has been fixed, and the remaining issue is https://github.com/marcone/teslausb/issues/149

closed 2019-08-03 by marcone