marcone/teslausb #143: Tesla fails to write files

andersbjarby, 2019-07-22:
The Tesla recognize the mounted volume and folders. However it seems that it fails to write complete files so when reading /mutable/archiveloop.log I find this:

`Mon 22 Jul 13:54:46 CEST 2019: '2019-07-22_10-24-04/2019-07-22_10-14-55-left_rep
eater.mp4' is only 595 bytes
Mon 22 Jul 13:54:46 CEST 2019: '2019-07-22_10-24-04/2019-07-22_10-14-55-right_re
peater.mp4' is only 595 bytes
Mon 22 Jul 13:54:46 CEST 2019: '2019-07-22_10-24-04/2019-07-22_10-15-55-front.mp
4' is only 595 bytes
Mon 22 Jul 13:54:46 CEST 2019: '2019-07-22_10-24-04/2019-07-22_10-15-55-left_rep
eater.mp4' is only 595 bytes
Mon 22 Jul 13:54:46 CEST 2019: '2019-07-22_10-24-04/2019-07-22_10-15-55-right_re
peater.mp4' is only 595 bytes
Mon 22 Jul 13:54:46 CEST 2019: '2019-07-22_10-24-04/2019-07-22_10-16-56-front.mp
4' is only 595 bytes`

Could it have anything to do with this that I also find in the same log:

`Mon 22 Jul 18:23:05 CEST 2019: Running fsck on /mnt/cam...
Mon 22 Jul 18:23:06 CEST 2019: | fsck from util-linux 2.29.2
Mon 22 Jul 18:23:08 CEST 2019: | fsck.fat 4.1 (2017-01-24)
Mon 22 Jul 18:23:08 CEST 2019: | 0x41: Dirty bit is set. Fs was not properly unm
ounted and some data may be corrupt.`

marcone, 2019-07-22:
The "Dirty bit is set" message is fairly standard. You could try plugging the Pi into a PC and see if you can write files to the CAM drive.

BadPirate, 2019-07-28:
+1 Also getting this error, sync creates all the folders but copies no files.
`Fri 15 Feb 06:54:24 GMT 2019: '2019-07-27_17-31-04/2019-07-27_17-22-19-front.mp4' is only 595 bytes`
I'll try and see if I can write to USB and update.

BadPirate, 2019-07-28:
Okay, verified that folders are in fact filled with files from Tesla, but that they are all 595 bytes.  I am able to copy files from computer onto CAM drive without issue. :/

marcone, 2019-07-28:
I'd start by rebooting the car.
If the problem persists, check dmesg and see if there's anything in there. Clearly the car is able to write to the card (otherwise there would be no files or folders at all), it's just not doing any actual recording.

andersbjarby, 2019-07-28:
I could verify that both I and the Tesla could write to the folder so the small files had to be Tesla related. A reboot of the car did not fix it. However, when updating to 2019.24.4 the problem was resolved and everything seems to work.

ppamidimarri, 2019-07-29:
I have the same problem on 2019.24.4 with a freshly built Pi with the latest release prebuilt image. I built the Pi after I had 24.4 in the car. Rebooting the car and the Pi won't fix it.

ppamidimarri, 2019-07-30:
I tried to diagnose the issue, and here is what I found. 

- When I connect the Pi to a PC, it shows the CAM and MUSIC drives. I can write to the CAM/TeslaCam folder from the PC.
- The Pi has write access to the CIFS share. I can see that there are empty folders within the CIFS share for each timestamp. 
- After the archiveloop has run, the TeslaCam/SavedClips folder is empty. Looks like the archiveloop deletes those files. 

Here is are some excerpts from /mutable/archiveloop.log:

```
Thu  3 Nov 10:17:34 PDT 2016: Waiting for archive to be reachable...
Thu  3 Nov 21:21:21 PDT 2016: Archive is reachable.
Thu  3 Nov 21:21:21 PDT 2016: Waiting for time to be set by ntpd
Mon 29 Jul 17:50:42 PDT 2019: Time still not set, attempting to force it
Mon 29 Jul 17:50:42 PDT 2019: Failed to set time
Mon 29 Jul 17:51:02 PDT 2019: Archiving...
Mon 29 Jul 17:51:02 PDT 2019: Disconnecting usb from host...
Mon 29 Jul 17:51:02 PDT 2019: Disconnected usb from host.
Mon 29 Jul 17:51:02 PDT 2019: Ensuring cam archive is mounted...
Mon 29 Jul 17:51:02 PDT 2019: Mounting /mnt/archive...
Mon 29 Jul 17:51:03 PDT 2019: Mounted /mnt/archive.
Mon 29 Jul 17:51:03 PDT 2019: Ensured cam archive is mounted.
Mon 29 Jul 17:51:03 PDT 2019: Starting recording archiving...
Mon 29 Jul 17:51:03 PDT 2019: Ensuring cam file is mounted...
Mon 29 Jul 17:51:03 PDT 2019: Mounting /mnt/cam...
Mon 29 Jul 17:51:03 PDT 2019: Mounted /mnt/cam.
Mon 29 Jul 17:51:03 PDT 2019: Ensured cam file is mounted.
Mon 29 Jul 17:51:04 PDT 2019: Running fsck on /mnt/cam...
Mon 29 Jul 17:51:04 PDT 2019: | fsck from util-linux 2.29.2
Mon 29 Jul 17:51:05 PDT 2019: | fsck.fat 4.1 (2017-01-24)
Mon 29 Jul 17:51:05 PDT 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Mon 29 Jul 17:51:05 PDT 2019: |  Automatically removing dirty bit.
Mon 29 Jul 17:51:05 PDT 2019: | Free cluster summary wrong (3027896 vs. really 3027899)
Mon 29 Jul 17:51:05 PDT 2019: |   Auto-correcting.
Mon 29 Jul 17:51:05 PDT 2019: | Performing changes.
Mon 29 Jul 17:51:05 PDT 2019: | /dev/loop0: 255 files, 255/3028154 clusters
Mon 29 Jul 17:51:05 PDT 2019: Finished fsck on /mnt/cam.
Mon 29 Jul 17:51:05 PDT 2019: wake up car process is 11631
Mon 29 Jul 17:51:05 PDT 2019: Moving clips to archive...
Mon 29 Jul 17:51:05 PDT 2019: Creating output directory ''
Mon 29 Jul 17:51:06 PDT 2019: Creating output directory '2019-07-28_22-09-45'
Mon 29 Jul 17:51:06 PDT 2019: '2019-07-28_22-09-45/2019-07-28_21-59-18-front.mp4' is only 595 bytes
Mon 29 Jul 17:51:06 PDT 2019: '2019-07-28_22-09-45/2019-07-28_21-59-18-left_repeater.mp4' is only 595 bytes
Mon 29 Jul 17:51:06 PDT 2019: '2019-07-28_22-09-45/2019-07-28_21-59-18-right_repeater.mp4' is only 595 bytes
Mon 29 Jul 17:51:06 PDT 2019: '2019-07-28_22-09-45/2019-07-28_22-00-19-front.mp4' is only 595 bytes
[... lots of other lines like this ...]
Mon 29 Jul 17:51:08 PDT 2019: '2019-07-29_03-50-32/2019-07-29_03-46-45-front.mp4' is only 595 bytes
Mon 29 Jul 17:51:08 PDT 2019: '2019-07-29_03-50-32/2019-07-29_03-46-45-left_repeater.mp4' is only 595 bytes
Mon 29 Jul 17:51:08 PDT 2019: '2019-07-29_03-50-32/2019-07-29_03-46-45-right_repeater.mp4' is only 595 bytes
Mon 29 Jul 17:51:08 PDT 2019: Moved 0 file(s), failed to copy 0, deleted 66.
Mon 29 Jul 17:51:08 PDT 2019: Finished moving clips to archive.
Mon 29 Jul 17:51:08 PDT 2019: killing PID 11631
Mon 29 Jul 17:51:08 PDT 2019: Unmounting /mnt/cam...
Mon 29 Jul 17:51:09 PDT 2019: Unmounted /mnt/cam.
Mon 29 Jul 17:51:09 PDT 2019: Finished archiving.
Mon 29 Jul 17:51:09 PDT 2019: Music archive not configured or unreachable
Mon 29 Jul 17:51:09 PDT 2019: unmounting /mnt/archive
Mon 29 Jul 17:51:09 PDT 2019: Connecting usb to host...
Mon 29 Jul 17:51:09 PDT 2019: Connected usb to host.
Mon 29 Jul 17:51:09 PDT 2019: Waiting for archive to be unreachable...
Mon 29 Jul 17:55:52 PDT 2019: Sleeping before retry...
Mon 29 Jul 17:55:53 PDT 2019: Retrying...
```
If it matters: I built my Pi from the prebuilt image using one-step setup instructions. I use a CIFS share, with no pushover/gotify/ifttt notifications. I do have the Tesla API thing enabled to keep the car alive. My car software is 2019.24.4, and I first plugged this freshly-built Pi into the car with this software version. 

Any ideas what I could try to get this to work again? (I used to have a working teslausb with rclone until a few days ago.)

gautamvasudevan, 2019-07-30:
Doesn't look related to TeslaUSB - 

https://www.reddit.com/r/teslamotors/comments/bz4pnz/teslacamsentry_not_working_keeps_writing_595_byte/

ppamidimarri, 2019-07-30:
> Doesn't look related to TeslaUSB -
> 
> https://www.reddit.com/r/teslamotors/comments/bz4pnz/teslacamsentry_not_working_keeps_writing_595_byte/

Thank you, this is helpful.

ppamidimarri, 2019-07-31:
This problem is fixed by doing the following the steps in the reddit post. 

- Disconnect the Pi from the car
- Turn off sentry mode
- Reboot the car
- Then park the car for a couple of hours 
- Connect the Pi again, enable sentry mode

marcone, 2019-10-25:
This seems obsolete by now

closed 2019-10-25 by marcone