marcone/teslausb #144: rsync configuration parameters

natrlhy, 2019-07-22:
I was wondering what parameters in the teslausb_setup_variables.conf I would need to use to try and get the rsync method a go? I see that the readme says only CIFS has been tested and working but seems others may have gotten rsync working as well? Are entries in the teslausb_setup_variables.conf the only things that are needed to get it working? If not any guidance on fully configuring rsync would be of great help.

Thanks!

natrlhy, 2019-07-22:
Review this information now:

https://github.com/marcone/teslausb/blob/main-dev/doc/SetupRSync.md

edalquist, 2019-07-23:
I have rsync based backup setup. I believe I just followed those instructions, I think the one change may be needed in those docs is running `/root/bin/remountfs_rw` before doing the keygen

natrlhy, 2019-07-23:
How do I go about disabling CIFS and try out rsync?

edalquist, 2019-07-23:
I think you follow those instructions and then re-run `/root/bin/setup-teslausb`

My rsync config is:
```
# Export files via rsync
export ARCHIVE_SYSTEM=rsync
export RSYNC_USER=tesla
export RSYNC_SERVER=backups
export RSYNC_PATH=/data/backups/teslacam
```

`backups` is the LAN DNS name of my linux server, you could put in the IP address
`tesla` is the username on the linux machine that the teslausb pi scripts will use when connecting

Here is what I think I did for the setup:

```
# switch to root
$ sudo -i

# Remount SDCard read-write so the ssh key can be saved
$ /root/bin/remountfs_rw

# Generate a ECDSA key. Hit enter when prompted for a password
$ ssh-keygen -t ecdsa

# Get the public key, copy it into the ~/.ssh/authorized_keys file on your target server
# Remember that the .ssh director needs 700 permissions and authorized_keys file needs 600 permissions
$ cat /root/.ssh/id_ecdsa.pub

# SSH to your target server so that the pi saves the server's key fingerprint
$ ssh tesla@backups

# Reboot cleanly to get back to a read-only file system
$ reboot now
```

If you verify that I can send a pull request to update the SetupRSync.md doc

natrlhy, 2019-07-24:
Thanks for this!

So I did the following:
Read and followed: https://github.com/marcone/teslausb/blob/main-dev/doc/SetupRSync.md
Edited the /root/teslausb_setup_variables.conf file.
Commented out the cifs settings and pasted in my lines:

export ARCHIVE_SYSTEM=rsync
export RSYNC_USER=user1
export RSYNC_SERVER=server1
export RSYNC_PATH=/volume1/Storage/TeslaCam

ran /root/bin/setup-teslausb
rebooted

I found the log file /mutable/archiveloop.log and it's retrying. I am staging old SavedClips data there but I just need a way on my PC to simulate losing wifi connection but still have SSH. I tried raspberrypi.local but that doesn't seem to work.

edalquist, 2019-07-24:
So if you do:

```
$ sudu -i
$ ssh user1@server1
```

Do you successfully connect to your server without any prompts?

natrlhy, 2019-07-24:
Yep passwordless SSH is setup and working fine from the pi to my NAS

edalquist, 2019-07-24:
Oh sorry, are you saying this is working for you for you now?

natrlhy, 2019-08-02:
Got my Model 3 back today after getting some Ceramic coating done, etc. I'll be testing this more now with real data

natrlhy, 2019-08-02:
So went out and came back. Not much going on and I saved some SavedClips while in the car:

Fri  2 Aug 11:53:53 PDT 2019: Unmounted /mnt/cam.
Fri  2 Aug 11:53:53 PDT 2019: Finished archiving.
Fri  2 Aug 11:53:53 PDT 2019: Music archive not configured or unreachable
Fri  2 Aug 11:53:53 PDT 2019: Connecting usb to host...
Fri  2 Aug 11:53:53 PDT 2019: Connected usb to host.
Fri  2 Aug 11:53:54 PDT 2019: Waiting for archive to be unreachable...
Fri  2 Aug 11:54:04 PDT 2019: Sleeping before retry...
Fri  2 Aug 11:54:05 PDT 2019: Retrying...
Fri  2 Aug 11:54:07 PDT 2019: Sleeping before retry...
Fri  2 Aug 11:54:08 PDT 2019: Retrying...

natrlhy, 2019-08-03:
So I disconnected my Pi from my Model 3... Not sure why files aren't getting synced to my NAS via rsync... 

It's mounting it fine on the Pi:

//10.10.10.10/Storage\TeslaCam   15T   13T  2.1T  86% /mnt/archive

from the archiveloop.log file here is a section of it:

Fri  2 Aug 22:45:45 PDT 2019: Ensured cam file is mounted.
Fri  2 Aug 22:45:45 PDT 2019: Running fsck on /mnt/cam... 
Fri  2 Aug 22:45:45 PDT 2019: | fsck from util-linux 2.29.2
Fri  2 Aug 22:45:47 PDT 2019: | fsck.fat 4.1 (2017-01-24)
Fri  2 Aug 22:45:47 PDT 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Fri  2 Aug 22:45:47 PDT 2019: |  Automatically removing dirty bit.
Fri  2 Aug 22:45:47 PDT 2019: | Performing changes.
Fri  2 Aug 22:45:47 PDT 2019: | /dev/loop0: 344 files, 356159/1841695 clusters
Fri  2 Aug 22:45:47 PDT 2019: Finished fsck on /mnt/cam.
Fri  2 Aug 22:45:47 PDT 2019: Archiving through rsync...
Fri  2 Aug 22:45:48 PDT 2019: No files to archive through rsync.
Fri  2 Aug 22:45:48 PDT 2019: Unmounting /mnt/cam... 
Fri  2 Aug 22:45:49 PDT 2019: Unmounted /mnt/cam. 
Fri  2 Aug 22:45:49 PDT 2019: Finished archiving.
Fri  2 Aug 22:45:49 PDT 2019: Music archive not configured or unreachable
Fri  2 Aug 22:45:49 PDT 2019: Connecting usb to host...
Fri  2 Aug 22:45:49 PDT 2019: Connected usb to host.
Fri  2 Aug 22:45:49 PDT 2019: Waiting for archive to be unreachable...
==============================================
Thu  3 Nov 10:17:06 PDT 2016: Starting archiveloop...
Thu  3 Nov 10:17:06 PDT 2016: Waiting for time to be set by ntpd
Fri  2 Aug 22:51:40 PDT 2019: Time still not set, attempting to force it
Fri  2 Aug 22:51:40 PDT 2019: Failed to set time
Fri  2 Aug 22:51:40 PDT 2019: Archiving...
Fri  2 Aug 22:51:40 PDT 2019: Disconnecting usb from host...
Fri  2 Aug 22:51:40 PDT 2019: Disconnected usb from host.
Fri  2 Aug 22:51:40 PDT 2019: Starting recording archiving... 
Fri  2 Aug 22:51:41 PDT 2019: Ensuring cam file is mounted...
Fri  2 Aug 22:51:41 PDT 2019: Mounting /mnt/cam...
Fri  2 Aug 22:51:41 PDT 2019: Mounted /mnt/cam.
Fri  2 Aug 22:51:41 PDT 2019: Ensured cam file is mounted.
Fri  2 Aug 22:51:41 PDT 2019: Running fsck on /mnt/cam...
Fri  2 Aug 22:51:41 PDT 2019: | fsck from util-linux 2.29.2
Fri  2 Aug 22:51:43 PDT 2019: | fsck.fat 4.1 (2017-01-24)
Fri  2 Aug 22:51:43 PDT 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Fri  2 Aug 22:51:43 PDT 2019: |  Automatically removing dirty bit.
Fri  2 Aug 22:51:43 PDT 2019: | Reclaimed 3443 unused clusters (112820224 bytes) in 3 chains.
Fri  2 Aug 22:51:43 PDT 2019: | Free cluster summary wrong (1485558 vs. really 1485521)
Fri  2 Aug 22:51:43 PDT 2019: |   Auto-correcting.
Fri  2 Aug 22:51:43 PDT 2019: | Performing changes.
Fri  2 Aug 22:51:43 PDT 2019: | /dev/loop0: 347 files, 356174/1841695 clusters
Fri  2 Aug 22:51:43 PDT 2019: Finished fsck on /mnt/cam.
Fri  2 Aug 22:51:43 PDT 2019: Archiving through rsync...
Fri  2 Aug 22:51:44 PDT 2019: No files to archive through rsync.
Fri  2 Aug 22:51:44 PDT 2019: Unmounting /mnt/cam...
Fri  2 Aug 22:51:44 PDT 2019: Unmounted /mnt/cam.
Fri  2 Aug 22:51:44 PDT 2019: Finished archiving.
Fri  2 Aug 22:51:44 PDT 2019: Music archive not configured or unreachable
Fri  2 Aug 22:51:44 PDT 2019: Connecting usb to host...
Fri  2 Aug 22:51:44 PDT 2019: Connected usb to host.
Fri  2 Aug 22:51:44 PDT 2019: Waiting for archive to be unreachable...
Fri  2 Aug 22:52:58 PDT 2019: Sleeping before retry...
Fri  2 Aug 22:52:59 PDT 2019: Retrying...
Fri  2 Aug 22:53:15 PDT 2019: Sleeping before retry...
Fri  2 Aug 22:53:16 PDT 2019: Retrying...
Fri  2 Aug 22:54:27 PDT 2019: Sleeping before retry...
Fri  2 Aug 22:54:28 PDT 2019: Retrying...
Fri  2 Aug 22:59:02 PDT 2019: Sleeping before retry...
Fri  2 Aug 22:59:03 PDT 2019: Retrying...
Fri  2 Aug 22:59:45 PDT 2019: Sleeping before retry...
Fri  2 Aug 22:59:46 PDT 2019: Retrying...
Fri  2 Aug 23:04:42 PDT 2019: Sleeping before retry...

There are files on the TeslaCam that are not synced:

natrlhy, 2019-08-03:
I:\TeslaCam\SavedClips>dir
 Volume in drive I is CAM
 Volume Serial Number is 8BD1-CA3A

 Directory of I:\TeslaCam\SavedClips

07/22/2019  05:01 PM    <DIR>          .
07/22/2019  05:01 PM    <DIR>          ..
08/02/2019  10:07 AM    <DIR>          2019-08-02_10-07-51
07/19/2019  04:01 PM    <DIR>          2019-07-19_10-42-01
08/02/2019  11:55 AM    <DIR>          2019-08-02_11-55-12
08/02/2019  12:03 PM    <DIR>          2019-08-02_12-03-20
08/02/2019  12:04 PM    <DIR>          2019-08-02_12-04-24
08/02/2019  12:05 PM    <DIR>          2019-08-02_12-05-32
08/02/2019  12:06 PM    <DIR>          2019-08-02_12-06-40
08/02/2019  12:07 PM    <DIR>          2019-08-02_12-07-46
08/02/2019  08:23 PM    <DIR>          2019-08-02_20-23-39
08/02/2019  09:45 PM    <DIR>          2019-08-02_21-45-01
               0 File(s)              0 bytes
              12 Dir(s)  48,677,552,128 bytes free

natrlhy, 2019-08-03:
where do I go from here?

marcone, 2019-08-03:
Are there actually any files in any of those directories?
The fact that your server mounts OK is irrelevant, since that's a CIFS mount, and you're trying to use rsync (in fact, there shouldn't even *be* a mount, unless you created it manually). Does your server actually support rsync?

natrlhy, 2019-08-03:
Yes There is valid data in all the directories. I didn’t create a mount manually and if I did it wouldn’t be persistent right?

My NAS server is a Synology server and does support rsync

marcone, 2019-08-03:
Looks like you originally configured this for cifs backup (per comments above), then reconfigured for rsync. The /mnt/archive mount is a leftover from when it was configured for cifs.

You could try manually running the rsync command and see if it prints anything useful.

natrlhy, 2019-08-03:
Yes I did have CIFS to start but wanted to try rsync. What's the best way to cleanup the configuration? Also how do I find the fully rsync command that is to be run?

natrlhy, 2019-08-04:
Ok, so I've started over again... I wasn't sure how to clean up from the original CIFS setup.

1. Reflashed with latest teslausb-20190615.zip image.

Modified the tesla_setup_variables.conf file for my needs:

# Archive files via rsync
export ARCHIVE_SYSTEM=rsync
export RSYNC_USER=me
export RSYNC_SERVER=server
export RSYNC_PATH=/volume1/Storage/TeslaCam
export camsize=100%

Pi comes up fine after about 6 minutes and looks fine. Mounts CAM on my PC fine.

ssh to Pi
sudo -i
/root/bin/remountfs_rw
ssh-keygen -t ecdsa
cat /root/.ssh/id_ecdsa.pub

I copied the output of the .pub file to my NAS server account's authorized_keys file. Permissions are fine on the Pi and NAS side.

Able to ssh me@server without password prompt

Put the Pi in my Tesla, drove around, tapped to save some clips. Got home, nothing sycned, went out again came home nothing synced.

Here's a snippet. of the /mutable/archiveloop.log file
`Sat  3 Aug 20:51:31 PDT 2019: Retrying...
Sat  3 Aug 20:51:31 PDT 2019: Attempts exhausted.
Sat  3 Aug 20:51:31 PDT 2019: Archive is unreachable.
Sat  3 Aug 20:51:31 PDT 2019: Waiting for archive to be reachable...
Sat  3 Aug 21:17:12 PDT 2019: Archive is reachable.
Sat  3 Aug 21:17:12 PDT 2019: Waiting for time to be set by ntpd...
Sat  3 Aug 21:17:14 PDT 2019: Time now set
Sat  3 Aug 21:17:34 PDT 2019: Archiving...
Sat  3 Aug 21:17:34 PDT 2019: Disconnecting usb from host...
Sat  3 Aug 21:17:34 PDT 2019: Disconnected usb from host.
Sat  3 Aug 21:17:34 PDT 2019: Starting recording archiving...
Sat  3 Aug 21:17:34 PDT 2019: Ensuring cam file is mounted...
Sat  3 Aug 21:17:34 PDT 2019: Mounting /mnt/cam...
Sat  3 Aug 21:17:34 PDT 2019: Mounted /mnt/cam.
Sat  3 Aug 21:17:34 PDT 2019: Ensured cam file is mounted.
Sat  3 Aug 21:17:34 PDT 2019: Running fsck on /mnt/cam...
Sat  3 Aug 21:17:34 PDT 2019: | fsck from util-linux 2.29.2
Sat  3 Aug 21:17:35 PDT 2019: | fsck.fat 4.1 (2017-01-24)
Sat  3 Aug 21:17:35 PDT 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Sat  3 Aug 21:17:35 PDT 2019: |  Automatically removing dirty bit.
Sat  3 Aug 21:17:35 PDT 2019: | Performing changes.
Sat  3 Aug 21:17:35 PDT 2019: | /dev/loop0: 86 files, 87999/1841695 clusters
Sat  3 Aug 21:17:35 PDT 2019: Finished fsck on /mnt/cam.
Sat  3 Aug 21:17:35 PDT 2019: Archiving through rsync...
Sat  3 Aug 21:17:36 PDT 2019: No files to archive through rsync.
Sat  3 Aug 21:17:36 PDT 2019: Unmounting /mnt/cam...
Sat  3 Aug 21:17:37 PDT 2019: Unmounted /mnt/cam.
Sat  3 Aug 21:17:37 PDT 2019: Finished archiving.
Sat  3 Aug 21:17:37 PDT 2019: Music archive not configured or unreachable
Sat  3 Aug 21:17:37 PDT 2019: Connecting usb to host...
Sat  3 Aug 21:17:37 PDT 2019: Connected usb to host.
Sat  3 Aug 21:17:37 PDT 2019: Waiting for archive to be unreachable...
Sat  3 Aug 21:21:02 PDT 2019: Sleeping before retry...
Sat  3 Aug 21:21:03 PDT 2019: Retrying...
Sat  3 Aug 21:28:02 PDT 2019: Sleeping before retry...
Sat  3 Aug 21:28:03 PDT 2019: Retrying...
Sat  3 Aug 21:52:38 PDT 2019: Sleeping before retry...
Sat  3 Aug 21:52:39 PDT 2019: Retrying...
Sat  3 Aug 21:52:39 PDT 2019: Sleeping before retry...
Sat  3 Aug 21:52:40 PDT 2019: Retrying...
Sat  3 Aug 21:52:40 PDT 2019: Sleeping before retry...
Sat  3 Aug 21:52:41 PDT 2019: Retrying...
[... 33 lines cut ...]
Sat  3 Aug 22:41:22 PDT 2019: | Performing changes.
Sat  3 Aug 22:41:22 PDT 2019: | /dev/loop0: 243 files, 266045/1841695 clusters
Sat  3 Aug 22:41:22 PDT 2019: Finished fsck on /mnt/cam.
Sat  3 Aug 22:41:22 PDT 2019: Archiving through rsync...
Sat  3 Aug 22:41:23 PDT 2019: No files to archive through rsync.
Sat  3 Aug 22:41:23 PDT 2019: Unmounting /mnt/cam...
Sat  3 Aug 22:41:23 PDT 2019: Unmounted /mnt/cam.
Sat  3 Aug 22:41:23 PDT 2019: Finished archiving.
Sat  3 Aug 22:41:23 PDT 2019: Music archive not configured or unreachable
Sat  3 Aug 22:41:23 PDT 2019: Connecting usb to host...
Sat  3 Aug 22:41:23 PDT 2019: Connected usb to host.
Sat  3 Aug 22:41:23 PDT 2019: Waiting for archive to be unreachable..`

What am I doing wrong?
What can I do test the rsync?

I see this log file in /tmp:

root@teslacam:/tmp# more archive-rsync-cmd.log
2019/08/03 20:50:28 [1107] rsync: connection unexpectedly closed (0 bytes received so far) [sender]
2019/08/03 20:50:28 [1107] rsync error: error in rsync protocol data stream (code 12) at io.c(235) [sender=3.1.2]
2019/08/03 21:17:36 [7308] rsync: connection unexpectedly closed (0 bytes received so far) [sender]
2019/08/03 21:17:36 [7308] rsync error: error in rsync protocol data stream (code 12) at io.c(235) [sender=3.1.2]
2019/08/03 22:41:23 [23696] rsync: connection unexpectedly closed (0 bytes received so far) [sender]
2019/08/03 22:41:23 [23696] rsync error: error in rsync protocol data stream (code 12) at io.c(235) [sender=3.1.2]

natrlhy, 2019-08-04:
Ok so all along I thought I had rsync on my NAS enabled but I didn't 👍 

I powered off my Pi, after it booted and connected....

pi@teslacam:/tmp $ tail -F /tmp/archive-rsync-cmd.log 
2019/08/03 23:33:28 [998] building file list
2019/08/03 23:33:28 [998] rsync: link_stat "/mnt/cam/TeslaCam/saved*" failed: No such file or directory (2)
2019/08/03 23:33:28 [998] cd+++++++++ 2019-08-03_21-00-39/
2019/08/03 23:33:59 [998] <f+++++++++ 2019-08-03_21-00-39/2019-08-03_20-50-31-front.mp4
2019/08/03 23:34:29 [998] <f+++++++++ 2019-08-03_21-00-39/2019-08-03_20-50-31-left_repeater.mp4
2019/08/03 23:34:57 [998] <f+++++++++ 2019-08-03_21-00-39/2019-08-03_20-50-31-right_repeater.mp4
2019/08/03 23:35:21 [998] <f+++++++++ 2019-08-03_21-00-39/2019-08-03_20-51-32-front.mp4
2019/08/03 23:35:42 [998] <f+++++++++ 2019-08-03_21-00-39/2019-08-03_20-51-32-left_repeater.mp4

Yay! Glad it seems to be working!!!!

closed 2019-08-04 by marcone