marcone/teslausb #146: Stuck on initial setup.

cman2011, 2019-07-24:
Posted earlier, didn't provide enough info.  Sorry, this is all new to me.  Let me try again.  I couldn't find where the diagnostics.txt was saved to, I used pastebin instead: https://pastebin.com/RbMEtS1Q

LED is pulsing twice.

Hope this is enough to troubleshoot. Thank you.

marcone, 2019-07-24:
The main cause of failure is that the mount command failed:
```
Wed 24 Jul 14:42:14 EDT 2019 : verify-archive-configuration: mount -t cifs //192.168.1.1/hdd2teslacam /tmp/archivetestmount -o vers=3.0,credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777
Wed 24 Jul 14:42:15 EDT 2019 : verify-archive-configuration: STOP: The archive couldn't be mounted with CIFS version 3.0. Try specifying a lower number for the CIFS version like this:
```
for which there could be several reasons, such as:
- the share doesn't actually exist
- the credentials are wrong
- wrong cifs version
- missing mount options

Before playing with the version, please verify that your server actually has IP address 192.168.1.1, that it is actually publishing a share named hdd2teslacam, and that you specified the correct login credentials. If the share password contains special characters, make sure to escape it properly, as described in https://github.com/marcone/teslausb/blob/main-dev/doc/OneStepSetup.md
Also see issue https://github.com/marcone/teslausb/issues/142, which describes that in some cases additional mount options may need to be specified.

It's easiest to experiment with various options if you ssh into the Pi and try to use the mount command manually from the command line, so you don't have to modify the .conf file and restart setup every time.

cman2011, 2019-07-25:
Thanks for the feedback. I’ll work on your suggestions. I noticed that the share path does not display the same as how I have it in the config file. The actual share path is 192.168.1.1/hdd2/teslacam. It seems like it has omitted the /

marcone, 2019-07-25:
That's odd, I use / myself in the share name, and it works fine for me. Did you by any chance accidentally use a backward slash instead of a forward slash?

cman2011, 2019-07-25:
I've corrected the / by reflashing
The server is confirmed to be 192.168.1.1 (and is reachable)
The share path is hdd2/teslacam on that server, and can be reached via Win10
There are no credentials required by the server and there are none set in the config file (left blank)

Here is an updated log:
https://pastebin.com/62J2EXJs

Thanks again.

marcone, 2019-07-25:
Still the mount command that's failing, but unfortunately cifs mount doesn't provide good diagnostics, so you'll just have to try different options and see what works.
You could try something like this: (untested)
```
sudo -i
mkdir -p /tmp/archivetestmount
for vers in 3.0 2.1 2.0 1.0
do
  for sec in ntlm ntlmv2 ntlmssp
  do
    mount -t cifs //192.168.1.1/hdd2/teslacam /tmp/archivetestmount -o vers=$vers,user=,pass=,iocharset=utf8,file_mode=0777,dir_mode=0777,sec=$sec
  done
done
mount | grep '/tmp/archivetestmount'
```
and see if any of them succeed.

marcone, 2019-07-27:
I just pushed a change to the "cifsmount" branch that does pretty much the above. Please give it a try and see if it works for you (add `export BRANCH=cifsmount` to the setup .conf file)

cman2011, 2019-07-28:
Thanks I will try and report back

marcone, 2019-07-28:
@cman2011 it's a little odd that it complains about "host is down" until it gets to "vers=1.0" mount option which succeeds, but other than that it looks like a successful install to me.

cman2011, 2019-07-29:
I see what you mean.  It does appear to be successful after switching to cifs 1.0.

I've tried to manually test the functionality by placing video files in the drive, then powering the Pi via USB (power only), but the files don't transfer and the Pi's LED is indicating a 2 pulse.

Jonkheer73, 2019-07-29:
Hi,
First the installation script would not complete and was stuck at step 2.
I found out that it was trying to connect to my Synology Nas with DSM 6.2 but failed with cifs 3.0.
As long that it could not connect it will not proceed.
Synology DSM 6.2 does not support cifs 3.0 of 2.1.
When I added export cifs_version="2.0" at the beginning of the install script I succesfully finished the installation.

marcone, 2019-08-01:
@Jonkheer73 did you try the "cifsmount" branch as noted above?

@cman2011 did you place and name the files the same as the car would? Because the archive script doesn't just copy any file it encounters. /mutable/archiveloop.log will tell you if the archive script found any files but failed to archive them, or whether it did not find any files to archive.

cman2011, 2019-08-02:
@marcone you were right.  For my test, I didn't properly mimic the directory structure.  I fixed that and the files transferred over.  Thank you for the cifs fix, it seems that resolved the issue I was having.  You can mark this issue closed.

closed 2019-08-02 by marcone

Jonkheer73, 2019-08-07:
> 
> 
> @Jonkheer73 did you try the "cifsmount" branch as noted above?
> 
@marcone sorry, I did not. It works flawless rightnow.