marcone/teslausb #148: HTTPS transfer fails due to Pi time being behind

shutton, 2019-07-30:
If your Pi, like mine, boots up thinking it's last February, the setup scripts will fail to download the raw GitHub script links due to GitHub's certificate being "not yet valid" (it was updated in June).

Minimally, the curl command lines need to be updated to relax this restriction.  Even better would be time syncing ahead of doing self-updates.

marcone, 2019-07-31:
Does this actually cause setup to fail, and if so, can you post some logs that show what happens?
(all of the downloads now have a retry, so even if a download fails at first because the date is wrong, I'd expect it to succeed once ntpd has set the correct date)

shutton, 2019-07-31:
It appeared to -- I had to re-run setup later.  I'm going to rebuild mine soon, so I can record what I was seeing (provided it's reproducible).

totszwai, 2019-07-31:
I haven't start using teslausb yet (but soon!)
About the time problem, would this help if you add ntpupdate into the said setup script?
https://raspberrypi.stackexchange.com/questions/8231/how-to-force-ntpd-to-update-date-time-after-boot

shutton, 2019-07-31:
Yep, ran into it again, though my date jumped around a bit for some reason (up to July 10th?):

```
WARNING: main-dev contains a different version of ./setup-teslausb. It is recommended to run './setup-teslausb selfupdate' to update to that version
Updating package index files...
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian buster InRelease [25.1 kB]
Reading package lists... Done
E: Release file for http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease is not valid yet (invalid for another 21d 13h 47min 26s). Updates for this repository will not be applied.
E: Release file for http://archive.raspberrypi.org/debian/dists/buster/InRelease is not valid yet (invalid for another 20d 11h 38min 57s). Updates for this repository will not be applied.
root@raspberrypi:~/setup/bin# cat /boot/cmdline.txt
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=17869b7d-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait modules-load=dwc2,g_ether quiet
root@raspberrypi:~/setup/bin# date
Wed 10 Jul 03:57:25 BST 2019
```

marcone, 2019-08-01:
I see. Since you mentioned curl and github, I figured you were talking about the additional teslausb setup scripts being downloaded from github by the main teslausb setup script using curl, but this is "apt-get update" failing.
I see you're trying to install on top of buster though, which I haven't tested, and which some people have reported problems with. I recommend using the teslausb prebuilt release image instead.

closed 2019-08-01 by marcone