marcone/teslausb #149: Pi Keeps Disconnecting from Model 3

THESHULK, 2019-07-31:
I've spent a significant amount of time trying to get TelsaUSB to work with my setup and I feel like I'm really close but I have one issue remaining that I can't figure out. For some reason, the pi keeps disconnecting from my Model 3. It will work fine for maybe 5-10 minutes at a time while driving but then the dashcam icon will disappear and it won't come back until I reboot the pi. The problem then repeats. Works fine for 5 minutes then disconnects until I reboot. I've tried supplying the pi with an external power from a power bank and I've tried different USB cables, but it's always the same outcome. I've attached the diagnostics log. Any help is appreciated.
[diagnostics.txt](https://github.com/marcone/teslausb/files/3449286/diagnostics.txt)

marcone, 2019-07-31:
Nothing stands out to me in the diagnostics, but they were taken on a freshly booted Pi, so presumably not when the problem was actually happening.
It might be more useful to take the diagnostics while the Pi is actually in the failed state.

THESHULK, 2019-07-31:
Ok I thought it kept a record of the past day. I'll take my laptop with me
on my next drive and fetch the logs when it's disconnected.

marcone, 2019-07-31:
It does keep a log of past days, but that's just the archiveloop log, which in this case doesn't show anything. The kernel log is probably more relevant here, but that's not kept across reboots.

THESHULK, 2019-07-31:
Good to know. Thanks for clearing that up.

THESHULK, 2019-08-02:
Ok I was able to reproduce the issue. With the new software update the car gave me a message saying that the USB drive is too slow to save and that I need to use a drive with a higher write speed. But this can't be the problem. I'm using a [Samsung PRO Endurance](https://www.amazon.com/Samsung-Endurance-128GB-Micro-Adapter/dp/B07B984HJ5/ref=sr_1_3?crid=2CX2HMM2ZS0VR&keywords=samsung+endurance+micro+sd&qid=1564761877&s=gateway&sprefix=Samsung+en%2Caps%2C154&sr=8-3) with a 100 MB/s write speed. So I tried to SSH to the pi but it doesn't appear to be powered on. The LED isn't flashing. I'm not sure what to do at this point. Could my pi be faulty? Could it be overheating and turning off? My case is 3D printed and it has vents over the processor but it doesn't have a heatsink.

marcone, 2019-08-02:
Maybe the card developed a bad sector or something? It definitely shouldn't be too slow. 100MB/sec is about 60 times faster than it needs to be :)

THESHULK, 2019-08-02:
That's possible. I can try taking the pi out of the equation and just use a
USB adapter to see if I get the same error. I also ordered a case with a
heatsink in case overheating is the problem. I'll post an update when I
know more.

THESHULK, 2019-08-06:
Update: The new case with the heatsink did not help. So I tried a full format of the SD card to fix any bad sectors and I've been using it without the pi as a regular dash cam USB. So far it's been working just fine. My next step is to try TeslaUSB again.

Alaric323, 2019-08-06:
I recently experienced this as well, but for me, it only occurs during music playback from the second image.  The music froze, and when I tried to skip to the next song (under the assumption that it was just a transfer error over CIFS), it dropped out completely, and both dashcam and USB disappeared from the car.  A replug fixed it, after which I tried only dashcam for the rest of my trip, and I arrived home (after two hours) without it crashing again.

THESHULK, 2019-08-07:
I tested again with teslausb and it's still disconnecting from the car. I'm
thinking it may be a hardware issue. I just purchased a brand new USB cable
to see if that will fix the problem. If not then I will try to replace the
pi.

THESHULK, 2019-08-11:
New cable didn't fix it but a new pi did. I guess I got a faulty pi.

closed 2019-08-11 by THESHULK