marcone/teslausb #150: Feature Request: ext4 support for TeslaCam & Music

Vipercat, 2019-07-31:
@marcone 
Hi Marcone, since recent versions of the Tesla firmware (not sure which one exactly) there is support for ext4 filesystem for both the TeslaCam folder as well as the Music Folder. Is there any benefit in using ext4 instead of fat32 and would it be possible to convert from fat32 to ext4 or does that require a rebuilt for these partitions?

Thanks,
Vipercat

marcone, 2019-07-31:
Not sure if there's any benefit (less prone to corruption, perhaps?), but a drawback would be reduced compatibility, e.g. you wouldn't be able to access the cam and music drives from Windows anymore. I tried getting ext4 to work once, but the car wouldn't recognize it.
You'd have to recreate the partitions as ext4 if you wanted to try this, there's no support for converting existing partitions.

Vipercat, 2019-07-31:
@marcone, Thanks for your prompt reply. Agree with your remark on the incompatibility with Windows, although I no longer have the need for connecting to a Windows system since your solution works perfectly and access remotely (in case of) does the job. Of course others might need this, so okay with leaving on fat32.

closed 2019-07-31 by Vipercat