marcone/teslausb #151: Drives not recognized

bladec59, 2019-08-01:
[teslausb-headless-setup 2.log](https://github.com/marcone/teslausb/files/3457263/teslausb-headless-setup.2.log)
Have done the one-step-setup and all appears to have finished correctly (setup log looks clean), but the usb isn't being recognized by PC/MAC or Tesla.  Did 100% cam, fyi.

bladec59, 2019-08-01:
My bad...wrong cable.

bladec59, 2019-08-01:
correction: recognized by PC/Mac but not by Tesla 2019.24; ordered new cables, will see how that goes.

bladec59, 2019-08-01:
Found out it works fine in my wife's X (2018 MCU2) and doesn't in my S (2017 MCU1), both on 2019.24.  At this point I'd say MCU1 is the core problem.

marcone, 2019-08-02:
Does a regular USB drive work with your MCU1 car?

bladec59, 2019-08-02:
Yes

marcone, 2019-08-02:
If you ssh into the Pi and then
```
sudo -i
modprobe -r g_mass_storage
modprobe g_mass_storage
```
does that change anything?

bladec59, 2019-08-02:
While connected to the S/MCU1?

marcone, 2019-08-02:
Yes, while connected to the car

bladec59, 2019-08-02:
No luck.

DMBlakeley, 2019-08-03:
Have this issue with CAM and MUSIC drives not be mounted when connected to my Mac. If I perform the steps of the previous post the drives mount and are accessible. As soon as I issue a halt to the ssh session the drives are unmounted.

The-Lor, 2019-08-03:
I have found that the g_ether in /boot/cmdline.txt caused my issues, once I removed it everything worked just fine except that windows didn't like to wait for the g_mass_storage to load, used the pwr in usb let the device boot then plugged in the data usb and it came up fine, but didn't have to do that in the car just took a min before it was showing available.  Apparently there is a g_multi that will do both ether and mass storage but once wifi is working then the g_eather wasn't needed https://www.raspberrypi.org/forums/viewtopic.php?t=238280 pointed me in this direction.

marcone, 2019-08-04:
@DMBlakeley what do you mean by "issue a halt to the ssh session" ?
If you're typing `halt` into the command line, you're shutting down the Pi, so of course the drives will be unmounted.

bdoooh, 2019-08-04:
I'm having the exact same issue. Tried multiple cables but the green light keeps doing the double blinks.

marcone, 2019-08-04:
See https://github.com/marcone/teslausb/issues/35

bladec59, 2019-08-04:
Per request. Moved from X, where it was working to S, where it was not working.  Ran diagnostics.
[diagnostics.txt](https://github.com/marcone/teslausb/files/3465032/diagnostics.txt)

marcone, 2019-08-04:
Can you do that again, but wait a bit longer for the Pi to start before running diagnostics? It looks like archiveloop was still in the middle of checking the disk image for errors for this report.

bdoooh, 2019-08-04:
@marcone looks like I'm an idiot. I didn't realize I needed to plug in both the USB data cable to the car and the USB charging port to a charging port. Splitter galore here now.

bladec59, 2019-08-04:
So I tried a different SD card; reflashed and configured and set up rpi.  Same (2 cables, no luck) result in the Model S.  Diags attached.
[diagnostics2.txt](https://github.com/marcone/teslausb/files/3465083/diagnostics2.txt)

marcone, 2019-08-04:
Diagnostics file shows the Pi has published the cam drive and that the car has connected to it. No idea why it's not actually working for you.

closed 2019-08-12 by marcone