marcone/teslausb #152: SMB transfers being flagged as SMBLoris

pushc6, 2019-08-04:
My Tesla-USB syncs across a VLAN to my storage device. I run an IPS before it routes the traffic and it seems that Snort is picking up the traffic from the Pi as a trojan, more specifically SMBLoris. 

ET DOS SMBLoris NBSS Length Mem Exhaustion Attempt (PoC Based)

Has anyone else seen this? I wonder if there is a bug in the SMB implementation?

marcone, 2019-08-04:
You got me... This is all an elaborate attempt to hack Tesla owners' computer systems.

More seriously, could this be a bad/overzealous snort rule?
There doesn't appear to be an official "community rule" on the snort site for this (perhaps it's only available to registered/paying users?), but I found this one:
```
alert tcp any any -> $HOME_NET [139,445] (msg:"ET DOS SMBLoris NBSS Length Mem Exhaustion Attempt (PoC Based)"; flow:established,to_server; content:"|00 01 ff ff|"; depth:4; threshold:type both,track by_dst,count 30, seconds 300; metadata: former_category DOS; reference:url,isc.sans.edu/forums/diary/SMBLoris+the+new+SMB+flaw/22662/; classtype:trojan-activity; sid:2024511; rev:2; metadata:affected_product Windows_XP_Vista_7_8_10_Server_32_64_Bit, attack_target Client_and_Server, deployment Internal, signature_severity Major, created_at 2017_08_02, performance_impact Significant, updated_at 2017_08_03;)
```
on https://github.com/jpalanco/alienvault-ossim/blob/master/snort-rules-default-open/rules/2.9.2/emerging.rules/emerging-dos.rules which seems to simply flag any sequence of more than 30 packets starting with a specific 4-byte sequence.
I'm not familiar with the SMB/CIFS protocol, but it doesn't seem that unexpected to me that when transferring even a single 30-megabyte file, you'll see many packets in a row that have the length set to the maximum allowed (128KB).

In any case, this doesn't have anything to do with teslausb itself. You'd have to take that up with the smb or snort maintainers.

closed 2019-08-04 by marcone

pushc6, 2019-08-04:
I hope that issue didn’t come off accusatory, I knew it wasn’t a deliberate attack. I was just wondering if anyone had experienced something similar. I didn’t know if there was anything you may have setup other than a straight ‘mv’ that might cause it to flag, since I don’t get any other false positives with other transfers I do. 

Looking at the script nothing jumped out at me that would trigger it. I’m going to keep the rule on for now and just see if I notice any kind of pattern or if it was just me testing it too much and it took it as an attack. 

If I had to guess I’m thinking maybe the connection monitor may trip it if it’s called over a certain duration. I’ll keep an eye out and see if I can correlate if it fires again and what it was doing. 

Thanks for this app, my many dead USBs thank you. Lol

marcone, 2019-08-04:
> I hope that issue didn’t come off accusatory, I knew it wasn’t a deliberate attack.

I should have added that smiley face after all... :)

> I was just wondering if anyone had experienced something similar.

Probably not that many people are running their internal traffic through snort.

> I didn’t know if there was anything you may have setup other than a straight ‘mv’ that might cause it to flag, since I don’t get any other false positives with other transfers I do.

There are multiple steps to the backup: a find, and then some mkdir and mv. Maybe try each of those separately and see if any of them trigger it? Regardless, the scripts are not doing anything bad, so I'm not about to change anything to appease snort.

> Looking at the script nothing jumped out at me that would trigger it. I’m going to keep the rule on for now and just see if I notice any kind of pattern or if it was just me testing it too much and [...]

pushc6, 2019-08-04:
Yea, no worries at this point it's just kind of a expedition. With Tesla owners being more "tech savy" I was hoping I may get lucky lol. The easy fix would be to just ignore this SID for teslausb but now I'm more curious than anything what made it trip. If it's an easy enough fix I'll throw up a pull request and you can merge it in or ignore it!