marcone/teslausb #153: Noob unable to get started

afgoody, 2019-08-04:
This is my second RPi project, so not a complete noob, but I'm having a hard time getting my Raspberry Pi Zero WH connected to my WiFi.

I've tried:
- The Headless (Automatic) instructions by setting up the .conf file with the SSID and password (they're definitely correct). I've attached the file with the actual values zeroed out. I never see the RPi on the network (via Netgear Orbi)
- Directly connecting to the RPi to watch the startup. Please see the attached log and screenshot.

I would like to try escaping from the startup log and just going into raspi-config and change the WiFi settings, but don't know the right commands to get to a prompt.

I'm sure I'm missing something obvious, but need a little help getting connected so I can finish configuring headless via PuTTY and WinSCP.

Thanks!
[teslausb_setup_variables.conf.txt](https://github.com/marcone/teslausb/files/3465147/teslausb_setup_variables.conf.txt)
[teslausb-headless-setup.log.txt](https://github.com/marcone/teslausb/files/3465148/teslausb-headless-setup.log.txt)
[image]

marcone, 2019-08-04:
Well, your wifi is clearly not working. Does your wifi password contain any special characters, and did you escape them properly?

afgoody, 2019-08-04:
Agreed (of course). No special characters (all alphanumerics) in SSID and password.

gkwok1, 2019-08-04:
Might be a dumb question but do you have a 2.4 GHz network and the Pi in range of the wifi? Maybe try using your phone as a hotspot to see if it works.

afgoody, 2019-08-05:
I'm running an Orbi with another RPi on it and the 2.4 and 5 GHz networks are active. I can see 35 other devices connected to the Orbi. About half are on 2.4 GHz.
[image]

gkwok1, 2019-08-06:
My only other suggestion would be to set up the Pi using a normal raspbian build to see if you got a bad wifi module.

afgoody, 2019-08-08:
I just reformatted and reimaged. It completed the setup and is blinking away at 1 Hz now and I'm able to SSH into it via PuTTY. So, I'm getting there. I have not completed the rest of the setup (e.g. Tesla API, etc.) or tested that I set up the fileshare with the right variables, so I need to edit the teslausb_setup_variables.conf file. It appears that the setup deletes that file. Do I just replace it with an updated one? I was hoping to be able to edit via nano.

marcone, 2019-08-08:
Setup moves it to /root, you can edit it there

afgoody, 2019-08-09:
Thanks. To be sure I'm doing this the right way...to edit the configuration parameters, do I do this:

ssh pi@teslausb
sudo -i
/root/bin/remountfs_rw

...then edit the /root/teslausb_setup_variables.conf file and leave it in /root and then just plug it back in?

afgoody, 2019-08-09:
Also...for the fileshare setup. I created a new user on Windows 10 named TeslaPi with a login password (let's say it's 1234). Does this look right?

export archiveserver="ShouldThisBeTheSystemNameonWindows10?"
export sharename="R:\Adam\Videos\TeslaCam\1-TeslaPi"
export shareuser="TeslaPi"
export sharepassword="1234"

marcone, 2019-08-09:
'archiveserver' should be the name or IP address of the machine hosting the network share.

The fact that you're using 'R:' as a drive letter makes me think that maybe you have a NAS and are mounting it as drive 'R' on your Windows 10 machine. If that's the case, then you should point the pi at your NAS, not at your Windows 10 machine.

Basically, take the '\\\computername\sharename\folder' syntax with which you'd access this shared drive from Windows. Then `export archiveserver="computername"`, `export sharename="sharename/folder"`

Once you've added the right values to /root/teslausb_setup_variables.conf, re-run /root/bin/setup-teslausb

afgoody, 2019-08-09:
The R: drive is just another HDD (actually 2 HDDs in RAID 1) I have in my Windows 10 PC (connected directly the the MoBo). So, I’m pointing setup at the PC, then the filepath as if I were using Windows explorer.

I’ve tried pinging the PC from the RPi and it only seems to work with the IP, so I must be using the wrong server name. I’d rather use the server name since I don’t want to set up the PC on a static IP.

But, are you saying I should do this anyways?

archiveserver="R:/"
export sharename="Adam/Videos/TeslaCam/1-TeslaPi"

The ‘1-TeslaPi’ folder is the one I gave the user ‘TeslaPi’ read/write access to and I’m setting the export password to that user’s Windows 10 password. Is that right?

marcone, 2019-08-09:
> But, are you saying I should do this anyways?
>
> archiveserver="R:/" 

No, "R:/" is not a valid hostname, and the "R:" drive letter has no meaning on the network, only on the local machine.  Normally, just entering the computer's name should do the trick, but if that doesn't work for you then just use the IP.

> export sharename="Adam/Videos/TeslaCam/1-TeslaPi" The ‘1-TeslaPi’ folder is the one I gave the user ‘TeslaPi’ read/write access to and I’m setting the export password to that user’s Windows 10 password. Is that right?

It depends on how it's published to the network. I think normally you'd publish a specific folder as a network share, so if you published 1-TeslaPi, then I think you'd just say `export sharename="1-TeslaPi"`. But if you're publishing your 'Videos' folder as a network share, for example, then you'd use: `export sharename="Videos/TeslaCam/1-TeslaPi"`
If you have another Windows machine handy, it might be helpful to try and browse to your "1-TeslaPi" from there, and see what the server and share name/path are.

afgoody, 2019-08-10:
Victory!

I made the changes to /root/teslausb_setup_variables.conf and re-ran /root/bin/setup-teslausb and the RPi LED started flashing at like 3-4 Hz. I was thinking "WTF" for a few seconds, but then I started getting an IFTTT message and the footage started archiving in the shared folder. Brilliant!

Can I do the same configuration changes to allocate space for the music folder or do I need to do this first?

ssh pi@teslausb.
sudo -i
/root/bin/remountfs_rw

Also, now that archiving is complete, the RPi is flashing two lights then a pause each second instead of the steady 1 Hz. Is there a description of the pulse patterns somewhere?

Thanks for your patience and a great program!

marcone, 2019-08-10:
Yes, run remountfs_rw first. I think you also need to delete /backingfiles/cam_disk.bin manually.

Not sure if it's documented somewhere, but 1 single-flash per second means "running but not connected to wifi", a double-flash per second means "running and connected to wifi", and a fast continuous flash means it's archiving.

afgoody, 2019-08-10:
I decided to just restart with a good .conf boot script so I don't end up with more problems. Before I do that, the last "snags" I've noticed are:

1. I still can only reach my Windows 10 PC server via the IP address. Some Googling suggests it may have something to do with my RPi or router's DNS config. I've confirmed the RPi's /etc/resolv.conf file has my Orbi's (router) IP address. Thing I should just configured my PC as a static IP and be done with it or is it worth troubleshooting?
2. The IFTTT notification said 0 files were deleted, but I noticed the 'Saved' folder was empty. Are the archived files being deleted off the RPi SD card each time automatically or do I need to set something differently in the .conf script to make that happen?

marcone, 2019-08-10:
> I decided to just restart with a good .conf boot script so I don't end up with more problems. Before I do that, the last "snags" I've noticed are:
>
> 1. I still can only reach my Windows 10 PC server [...]

Your choice, since that's an issue with your local network.

> 2. The IFTTT notification said 0 files were deleted, but I noticed the 'Saved' folder was empty. Are the archived files being deleted off the RPi SD card each time automatically or do I need to set something differently in the .conf script to make that happen?

They get deleted automatically after archiving.

afgoody, 2019-08-10:
Cool. Does the IFTTT message have an issue then? It moved 15 files, but said 0 deleted.

  1.  The IFTTT notification said 0 files were deleted, but I noticed the 'Saved' folder was empty. Are the archived files being deleted off the RPi SD card each time automatically or do I need to set something differently in the .conf script to make that happen?

marcone, 2019-08-10:
That's the number of files that were deleted instead of archived because they were incomplete.
(also note that "moved" already indicates files were removed from their original location, otherwise it would be a copy)

afgoody, 2019-08-10:
I reformatted and reinstalled with a 90% allocation for the cam. It configured through the 5 flashes and then the LED turned off completely. After a reboot and and unplug/replug, it joined the network successfully and started flashing at an archival rate. Since, I configured it to pull in from a music share that has 9 GB of music, I figured it would take a while. That was over two hours ago. Since it was still flashing rapidly and I never received an IFTTT "complete" notification, I figured it was time to troubleshoot. Here is the log

[teslausb-headless-setup.log](https://github.com/marcone/teslausb/files/3489349/teslausb-headless-setup.log)

Of note, since I sset camsize=90%, shouldn't music be 10%? Instead, there's this line in the log:

Sat 10 Aug 13:32:37 PDT 2019 : create-backingfiles: cam: 90%, music: 100%, mountpoint: /backingfiles

I thought I could just plug the SD card into my Windows 10 PC and navigate the file structure to see if the music was done transferring, but Windows found a total of 3 unrecognizable drives.

[image]

I've also done this:
Try sudo -i and then run /etc/rc.local

Since the log (from the /boot folder) appears to not still be recording activity, how can I see what the RPi is currenttly working on?

afgoody, 2019-08-11:
Nevermind, I gave it another hour and it just finished transferring the music!

Thanks again for the program and your help.

For those other noobs who may be reading this. Here's what I learned:
-Do a complete "Overwrite format" with SD Card Formatter
-When setting up the fire share on Windows 10: Right Click the folder>Properties>Sharing Tab>Advanced Sharing then assign the target (cam archive) and source (music) folder a short name to be used in the .conf file
-Assign server (my Windows 10 PC in my case) a static IP address

closed 2019-08-11 by afgoody