marcone/teslausb #154: PI Zero USB too slow with Tesla AP HW3?

MickyRoth, 2019-08-06:
Hi,

all of my recorded videos on the Zero are not running well - they look corrupted, often have blackouts etc.... Is this a known issue? My sd card should be fast enough (Samsung PRO Endurance microSDXC)

Regards, Micky

marcone, 2019-08-06:
How big are the recording files for HW3?
On my HW2 S they're about 30 megabytes per minute, which for the 3 cameras adds up to 1.5 megabytes per second. USB 2.0 is plenty fast enough for that, and I'd expect H.265 recordings to be smaller.
Have you tried with a regular USB stick (or microsd in an adapter) ?

MickyRoth, 2019-08-06:
> How big are the recording files for HW3?

They are about 35megabytes per one-minute-file, so a little bit bigger.

I am just getting about 3megabytes/s writespeed at my pc (measured by crystaldiskmark)

> Have you tried with a regular USB stick (or microsd in an adapter) ?

Yes, my SanDisk Extreme SSD 500GB (USB3) works fine (but its about 10 times faster).

marcone, 2019-08-06:
> > How big are the recording files for HW3?
> 
> They are about 35megabytes per one-minute-file, so a little bit bigger.
> 
> I am just getting about 3megabytes/s writespeed at my pc (measured by crystaldiskmark) which maybe is a little bit to slow (because measured in optimal environment)
> https://we.tl/t-rn7IiRHGi6

That's for the random access write tests though, and dashcam recording should be fairly sequential (even with 3 cameras). In any case, the car supposedly tells you when the drive is too slow (I've never seen it myself, but it has been mentioned on Tesla forums).

If rpi slowness really is the cause of the dropouts (though judging from the forums, dropouts are somewhat common even with a regular usb stick), there's unfortunately nothing I can do about that (the usb drive emulation is part of linux and not teslausb)

closed 2019-08-12 by marcone

Geesu, 2019-11-06:
Just ran into this issue after a few days using the pi zero. Does buying a faster SDHC card help?