marcone/teslausb #157: Possible to have swap space?

BadPirate, 2019-08-07:
I've built a [script](https://github.com/BadPirate/badcam) that utilizes ffmpeg to combine the 3 videos side-by-side-by-side into a single stream for easy playback, as well as using motion recognition to generate a "highlight" reel of only the interesting bits.  My goal would be to have this functionality run on the raspberry prior to archival, however my first test was thwarted by a failed malloc.  I assume this is because there isn't enough ram on the PI, and swap is disabled.

Am I right in thinking it's not as simple as removing noswap from cmdline.txt and utilizing swap (since it's read only, swap won't work anyway?)

Is it easy / advisable to setup a partition for swap that allows for writes so that I can get myself the extra mem needed (maybe 1G?)

marcone, 2019-08-07:
I suppose you could put a swap file on /backingfiles, but do you really want to do all that processing on the Pi? It'll be very slow...

marcone, 2019-08-07:
(also, I think swapping to SD card is generally considered a bad idea)

BadPirate, 2019-08-07:
Yeah -- it would be slow, however in many cases it will have a long time to
pre-process before I'm back on my home network.  Swapping to the SD card
would cause wear, though less so with faster cards like SDXC.  My vision
was to combine processing with a node page that would allow for viewing /
deleting / sharing videos directly from raspberry pi without having to have
a remote server...  Though another stumbling block towards that vision
<https://medium.com/@badpirate/1-n-badteslacam-hack-kickoff-1f8723b8fd82?source=your_stories_page--------------------------->
seems to be that you can't point the Tesla car browser at a local IP
address, requiring some kind of fancy NAT translation setup which keeps the
solution from being accessible.

closed 2019-08-12 by marcone