marcone/teslausb #158: Archiveloop keeps disconnecting

Nashable, 2019-08-10:
Hi there,

Configured based on the headless setup. The PI is able to reach the network and copy files. What I'm running into is a weird issue where after copying a few files the archive loop decides that it can no longer reach the archive server and kills the loop.

The Pi itself has been tried both connected to USB directly and powered via a battery backup.

Wondering if anyone has had the same experience?

> ==============================================
Thu  3 Nov 10:17:10 PDT 2016: Starting archiveloop...
Thu  3 Nov 10:17:10 PDT 2016: Waiting for time to be set by ntpd
Sat 10 Aug 11:29:12 PDT 2019: Time still not set, attempting to force it
Sat 10 Aug 11:29:12 PDT 2019: Failed to set time
Sat 10 Aug 11:29:12 PDT 2019: Ensuring music backing file is mounted...
Sat 10 Aug 11:29:12 PDT 2019: Mounting /mnt/music...
Sat 10 Aug 11:29:12 PDT 2019: Mounted /mnt/music.
Sat 10 Aug 11:29:12 PDT 2019: Ensured music drive is mounted.
Sat 10 Aug 11:29:12 PDT 2019: Running fsck on /mnt/music...
Sat 10 Aug 11:29:12 PDT 2019: | fsck from util-linux 2.29.2
Sat 10 Aug 11:29:33 PDT 2019: | fsck.fat 4.1 (2017-01-24)
Sat 10 Aug 11:29:33 PDT 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Sat 10 Aug 11:29:33 PDT 2019: |  Automatically removing dirty bit.
Sat 10 Aug 11:29:33 PDT 2019: | Performing changes.
Sat 10 Aug 11:29:33 PDT 2019: | /dev/loop0: 12589 files, 3926898/11560970 clusters
Sat 10 Aug 11:29:33 PDT 2019: Finished fsck on /mnt/music.
Sat 10 Aug 11:29:33 PDT 2019: Unmounting /mnt/music...
Sat 10 Aug 11:29:45 PDT 2019: Unmounted /mnt/music.
Sat 10 Aug 11:29:45 PDT 2019: Archiving...
Sat 10 Aug 11:29:45 PDT 2019: Disconnecting usb from host...
Sat 10 Aug 11:29:45 PDT 2019: Disconnected usb from host.
Sat 10 Aug 11:29:45 PDT 2019: Ensuring cam archive is mounted...
Sat 10 Aug 11:29:45 PDT 2019: /mnt/archive is already mounted.
Sat 10 Aug 11:29:45 PDT 2019: Ensured cam archive is mounted.
Sat 10 Aug 11:29:45 PDT 2019: Ensuring music archive is mounted...
Sat 10 Aug 11:29:45 PDT 2019: /mnt/musicarchive is already mounted.
Sat 10 Aug 11:29:45 PDT 2019: Ensured music archive is mounted.
Sat 10 Aug 11:29:45 PDT 2019: Starting recording archiving...
Sat 10 Aug 11:29:45 PDT 2019
Sending IFTTT message for moved files.
Sat 10 Aug 11:29:52 PDT 2019: Ensuring cam file is mounted...
Sat 10 Aug 11:29:52 PDT 2019: Mounting /mnt/cam...
Sat 10 Aug 11:29:53 PDT 2019: Mounted /mnt/cam.
Sat 10 Aug 11:29:53 PDT 2019: Ensured cam file is mounted.
Sat 10 Aug 11:29:53 PDT 2019: Running fsck on /mnt/cam...
Sat 10 Aug 11:29:53 PDT 2019: | fsck from util-linux 2.29.2
Sat 10 Aug 11:30:02 PDT 2019: | fsck.fat 4.1 (2017-01-24)
Sat 10 Aug 11:30:02 PDT 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Sat 10 Aug 11:30:02 PDT 2019: |  Automatically removing dirty bit.
Sat 10 Aug 11:30:02 PDT 2019: | Performing changes.
Sat 10 Aug 11:30:02 PDT 2019: | /dev/loop0: 3964 files, 3676453/3853634 clusters
Sat 10 Aug 11:30:02 PDT 2019: Finished fsck on /mnt/cam.
Sat 10 Aug 11:30:02 PDT 2019: Moving clips to archive...
Sat 10 Aug 11:30:02 PDT 2019: Creating output directory ''
Sat 10 Aug 11:30:02 PDT 2019: Creating output directory '2019-07-28_12-23-16'
Sat 10 Aug 11:30:03 PDT 2019: Creating output directory '2019-07-28_12-32-29'
Sat 10 Aug 11:30:03 PDT 2019: Creating output directory '2019-07-28_12-37-10'
Sat 10 Aug 11:30:03 PDT 2019: Creating output directory '2019-07-28_12-39-42'
Sat 10 Aug 11:30:03 PDT 2019: Creating output directory '2019-07-28_12-41-14'
Sat 10 Aug 11:30:03 PDT 2019: Creating output directory '2019-07-29_16-26-42'
Sat 10 Aug 11:30:03 PDT 2019: Moving '2019-07-29_16-26-42/2019-07-29_16-23-09-right_repeater.mp4'
Sat 10 Aug 11:32:12 PDT 2019: Moved '2019-07-29_16-26-42/2019-07-29_16-23-09-right_repeater.mp4'
Sat 10 Aug 11:32:12 PDT 2019: Moving '2019-07-29_16-26-42/2019-07-29_16-24-09-front.mp4'
Sat 10 Aug 11:34:35 PDT 2019: Moved '2019-07-29_16-26-42/2019-07-29_16-24-09-front.mp4'
Sat 10 Aug 11:34:35 PDT 2019: Moving '2019-07-29_16-26-42/2019-07-29_16-24-09-left_repeater.mp4'
Sat 10 Aug 11:36:21 PDT 2019: connection dead, killing archive-clips
Sat 10 Aug 11:36:21 PDT 2019: Unmounting /mnt/cam...
umount: /mnt/cam: target is busy
        (In some cases useful info about processes that
         use the device is found by lsof(8) or fuser(1).)
Sat 10 Aug 11:36:25 PDT 2019: Failed to unmount /mnt/cam, trying lazy unmount.
[... 103 lines cut ...]
Sat 10 Aug 11:57:51 PDT 2019: Retrying...
Sat 10 Aug 12:06:38 PDT 2019: Sleeping before retry...
Sat 10 Aug 12:06:39 PDT 2019: Retrying...
Sat 10 Aug 12:08:35 PDT 2019: Sleeping before retry...
Sat 10 Aug 12:08:36 PDT 2019: Retrying...
Sat 10 Aug 12:17:17 PDT 2019: Sleeping before retry...
Sat 10 Aug 12:17:18 PDT 2019: Retrying...
Sat 10 Aug 12:20:36 PDT 2019: Sleeping before retry...
Sat 10 Aug 12:20:37 PDT 2019: Retrying...
Sat 10 Aug 12:28:19 PDT 2019: Sleeping before retry...
Sat 10 Aug 12:28:20 PDT 2019: Retrying...
Sat 10 Aug 12:41:50 PDT 2019: Sleeping before retry...
Sat 10 Aug 12:41:51 PDT 2019: Retrying...
Sat 10 Aug 12:49:41 PDT 2019: Sleeping before retry...
Sat 10 Aug 12:49:43 PDT 2019: Retrying...
Sat 10 Aug 12:51:45 PDT 2019: Sleeping before retry...
Sat 10 Aug 12:51:47 PDT 2019: Retrying...
Sat 10 Aug 13:18:21 PDT 2019: Sleeping before retry...
Sat 10 Aug 13:18:22 PDT 2019: Retrying...
Sat 10 Aug 13:22:15 PDT 2019: Sleeping before retry...
Sat 10 Aug 13:22:16 PDT 2019: Retrying...
Sat 10 Aug 13:34:11 PDT 2019: Sleeping before retry...
Sat 10 Aug 13:34:12 PDT 2019: Retrying...
Sat 10 Aug 13:36:11 PDT 2019: Sleeping before retry...
Sat 10 Aug 13:36:12 PDT 2019: Retrying...

Nashable, 2019-08-10:
As a follow up I moved the device right next to my wifi router and issue went away. I can assume this is down to low signal strength of my wifi in my garage.

Would recommend adding a higher tolerance for the timeout to account for weaker wifi-signal

marcone, 2019-08-11:
Currently it considers the archive unreachable if it can't reach it within 5 seconds twice in a row.
Happy to consider a PR if you can come up with some reasonable values that work for your weak wifi situation.

closed 2019-08-11 by marcone

THESHULK, 2019-09-01:
I'm having the same exact problem but WIFI strength is not the issue since I tried moving the pi right next to the router and got the same results. Any thoughts on what the problem might be?

> Sat 31 Aug 20:38:20 CDT 2019: Archive is reachable.
Sat 31 Aug 20:38:20 CDT 2019: Waiting for time to be set by ntpd...
Sat 31 Aug 20:38:20 CDT 2019: Time now set
Sat 31 Aug 20:38:40 CDT 2019: Archiving...
Sat 31 Aug 20:38:40 CDT 2019: Disconnecting usb from host...
Sat 31 Aug 20:38:40 CDT 2019: Disconnected usb from host.
Sat 31 Aug 20:38:40 CDT 2019: Ensuring cam archive is mounted...
Sat 31 Aug 20:38:40 CDT 2019: Mounting /mnt/archive...
Sat 31 Aug 20:38:53 CDT 2019: Mounted /mnt/archive.
Sat 31 Aug 20:38:53 CDT 2019: Ensured cam archive is mounted.
Sat 31 Aug 20:38:53 CDT 2019: Starting recording archiving...
Sat 31 Aug 20:38:53 CDT 2019: Sending IFTTT message for moved files.
Sat 31 Aug 20:38:54 CDT 2019: Ensuring cam file is mounted...
Sat 31 Aug 20:38:54 CDT 2019: Mounting /mnt/cam...
Sat 31 Aug 20:38:54 CDT 2019: Mounted /mnt/cam.
Sat 31 Aug 20:38:54 CDT 2019: Ensured cam file is mounted.
Sat 31 Aug 20:38:54 CDT 2019: Running fsck on /mnt/cam...
Sat 31 Aug 20:38:54 CDT 2019: | fsck from util-linux 2.29.2
Sat 31 Aug 20:38:57 CDT 2019: | fsck.fat 4.1 (2017-01-24)
Sat 31 Aug 20:38:57 CDT 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Sat 31 Aug 20:38:57 CDT 2019: |  Automatically removing dirty bit.
Sat 31 Aug 20:38:57 CDT 2019: | Performing changes.
Sat 31 Aug 20:38:57 CDT 2019: | /dev/loop0: 866 files, 592313/3785199 clusters
Sat 31 Aug 20:38:57 CDT 2019: Finished fsck on /mnt/cam.
Sat 31 Aug 20:38:57 CDT 2019: wake up car process is 19869
Sat 31 Aug 20:38:57 CDT 2019: Moving clips to archive...
Sat 31 Aug 20:38:57 CDT 2019: Creating output directory ''
Sat 31 Aug 20:38:57 CDT 2019: Creating output directory '2019-08-23_12-41-52'
Sat 31 Aug 20:38:57 CDT 2019: Creating output directory '2019-08-23_19-03-25'
Sat 31 Aug 20:38:58 CDT 2019: Creating output directory '2019-08-23_19-16-51'
Sat 31 Aug 20:38:58 CDT 2019: Creating output directory '2019-08-23_19-24-28'
Sat 31 Aug 20:38:58 CDT 2019: Creating output directory '2019-08-23_19-26-22'
Sat 31 Aug 20:38:58 CDT 2019: Creating output directory '2019-08-23_19-34-12'
Sat 31 Aug 20:38:58 CDT 2019: Creating output directory '2019-08-23_19-38-53'
Sat 31 Aug 20:38:58 CDT 2019: Creating output directory '2019-08-23_19-54-40'
Sat 31 Aug 20:38:58 CDT 2019: Creating output directory '2019-08-23_20-12-30'
Sat 31 Aug 20:38:58 CDT 2019: Moving '2019-08-23_20-12-30/2019-08-23_20-02-46-right_repeater.mp4'
Sat 31 Aug 20:39:13 CDT 2019: connection dead, killing archive-clips
Sat 31 Aug 20:39:13 CDT 2019: killing PID 19869
Sat 31 Aug 20:39:13 CDT 2019: Unmounting /mnt/cam...
umount: /mnt/cam: target is busy
        (In some cases useful info about processes that
         use the device is found by lsof(8) or fuser(1).)
Sat 31 Aug 20:39:14 CDT 2019: Failed to unmount /mnt/cam, trying lazy unmount.
Sat 31 Aug 20:39:14 CDT 2019: lazily unmounted /mnt/cam
Sat 31 Aug 20:39:14 CDT 2019: Finished archiving.
Sat 31 Aug 20:39:14 CDT 2019: Music archive not configured or unreachable
Sat 31 Aug 20:39:14 CDT 2019: unmounting /mnt/archive
Sat 31 Aug 20:39:14 CDT 2019: Connecting usb to host...
Sat 31 Aug 20:39:14 CDT 2019: Connected usb to host.
Sat 31 Aug 20:39:14 CDT 2019: Truncating log...
Sat 31 Aug 20:39:14 CDT 2019: Waiting for archive to be unreachable...
Sat 31 Aug 20:48:20 CDT 2019: Sleeping before retry...
Sat 31 Aug 20:48:21 CDT 2019: Retrying...
Sat 31 Aug 21:38:31 CDT 2019: Sleeping before retry...
Sat 31 Aug 21:38:32 CDT 2019: Retrying...
Sat 31 Aug 21:41:11 CDT 2019: Sleeping before retry...
Sat 31 Aug 21:41:12 CDT 2019: Retrying...
Sat 31 Aug 21:42:12 CDT 2019: Sleeping before retry...
Sat 31 Aug 21:42:13 CDT 2019: Retrying...
>

mohsen-collisure, 2021-02-18:
Having the same problem and the archiveloop.log file only outputs `sleeping before retry` 5 times. Any solutions?