marcone/teslausb #159: teslausb creating files with bad names

ppamidimarri, 2019-08-11:
I have teslausb setup on a Pi Zero W using an SMB share. The SMB share is hosted on a Linux server. The archiving works for the most part, but it is often creating files with the name that I expect will be a directory name. Here is an example of the directory listing of my SMB share directory:

```
pi@fileserver:~ $ ls -lA /samba/fdrive/
total 29088
**-**rwxrw-r-- 1 fdruser sambashare 29783038 Aug 11 03:01 **2019-08-09_16-21-18**
```

Looking at /mutable/archiveloop.log, it may be happening when the Pi loses the connection to the archive while it is transferring files. 

```
Sun 11 Aug 10:34:36 PDT 2019: Creating output directory ''
Sun 11 Aug 10:34:36 PDT 2019: **Creating output directory '2019-08-09_16-21-18'**
Sun 11 Aug 10:34:36 PDT 2019: Moving '2019-08-09_16-21-18/2019-08-09_16-19-55-right_repeater.mp4'
Sun 11 Aug 10:34:55 PDT 2019: Moved '2019-08-09_16-21-18/2019-08-09_16-19-55-right_repeater.mp4'
Sun 11 Aug 10:34:56 PDT 2019: Moving '2019-08-09_16-21-18/2019-08-09_16-20-55-front.mp4'
Sun 11 Aug 10:35:03 PDT 2019: Moved '2019-08-09_16-21-18/2019-08-09_16-20-55-front.mp4'
Sun 11 Aug 10:35:03 PDT 2019: Moving '2019-08-09_16-21-18/2019-08-09_16-20-55-left_repeater.mp4'
Sun 11 Aug 10:35:10 PDT 2019: Moved '2019-08-09_16-21-18/2019-08-09_16-20-55-left_repeater.mp4'
Sun 11 Aug 10:35:10 PDT 2019: Moving '2019-08-09_16-21-18/2019-08-09_16-20-55-right_repeater.mp4'
Sun 11 Aug 10:35:17 PDT 2019: Moved '2019-08-09_16-21-18/2019-08-09_16-20-55-right_repeater.mp4'
Sun 11 Aug 10:35:17 PDT 2019: Moving '2019-08-09_16-21-18/2019-08-09_16-18-54-front.mp4'
Sun 11 Aug 10:35:36 PDT 2019: Moved '2019-08-09_16-21-18/2019-08-09_16-18-54-front.mp4'
Sun 11 Aug 10:35:36 PDT 2019: Moving '2019-08-09_16-21-18/2019-08-09_16-18-54-left_repeater.mp4'
Sun 11 Aug 10:35:56 PDT 2019: Moved '2019-08-09_16-21-18/2019-08-09_16-18-54-left_repeater.mp4'
Sun 11 Aug 10:35:56 PDT 2019: Moving '2019-08-09_16-21-18/2019-08-09_16-18-54-right_repeater.mp4'
Sun 11 Aug 10:36:16 PDT 2019: Moved '2019-08-09_16-21-18/2019-08-09_16-18-54-right_repeater.mp4'
Sun 11 Aug 10:36:16 PDT 2019: Moving '2019-08-09_16-21-18/2019-08-09_16-10-51-front.mp4'
Sun 11 Aug 10:36:34 PDT 2019: Moved '2019-08-09_16-21-18/2019-08-09_16-10-51-front.mp4'
Sun 11 Aug 10:36:34 PDT 2019: Moving '2019-08-09_16-21-18/2019-08-09_16-10-51-left_repeater.mp4'
Sun 11 Aug 10:36:42 PDT 2019: Sending wakeup API command...
Sun 11 Aug 10:36:54 PDT 2019: Moved '2019-08-09_16-21-18/2019-08-09_16-10-51-left_repeater.mp4'
Sun 11 Aug 10:36:54 PDT 2019: Moving '2019-08-09_16-21-18/2019-08-09_16-10-51-right_repeater.mp4'
Sun 11 Aug 10:37:11 PDT 2019: Moved '2019-08-09_16-21-18/2019-08-09_16-10-51-right_repeater.mp4'
Sun 11 Aug 10:37:12 PDT 2019: Moving '2019-08-09_16-21-18/2019-08-09_16-11-52-front.mp4'
Sun 11 Aug 10:37:33 PDT 2019: Moved '2019-08-09_16-21-18/2019-08-09_16-11-52-front.mp4'
Sun 11 Aug 10:37:34 PDT 2019: Moving '2019-08-09_16-21-18/2019-08-09_16-11-52-left_repeater.mp4'
Sun 11 Aug 10:37:53 PDT 2019: Moved '2019-08-09_16-21-18/2019-08-09_16-11-52-left_repeater.mp4'
Sun 11 Aug 10:37:53 PDT 2019: Moving '2019-08-09_16-21-18/2019-08-09_16-11-52-right_repeater.mp4'
Sun 11 Aug 10:38:16 PDT 2019: Moved '2019-08-09_16-21-18/2019-08-09_16-11-52-right_repeater.mp4'
Sun 11 Aug 10:38:16 PDT 2019: Moving '2019-08-09_16-21-18/2019-08-09_16-12-52-front.mp4'
Sun 11 Aug 10:38:38 PDT 2019: Moved '2019-08-09_16-21-18/2019-08-09_16-12-52-front.mp4'
Sun 11 Aug 10:38:38 PDT 2019: Moving '2019-08-09_16-21-18/2019-08-09_16-12-52-left_repeater.mp4'
Sun 11 Aug 10:38:49 PDT 2019: Sending wakeup API command...
Sun 11 Aug 10:38:57 PDT 2019: Moved '2019-08-09_16-21-18/2019-08-09_16-12-52-left_repeater.mp4'
Sun 11 Aug 10:38:57 PDT 2019: Moving '2019-08-09_16-21-18/2019-08-09_16-12-52-right_repeater.mp4'
Sun 11 Aug 10:39:20 PDT 2019: Moved '2019-08-09_16-21-18/2019-08-09_16-12-52-right_repeater.mp4'
Sun 11 Aug 10:39:20 PDT 2019: Moving '2019-08-09_16-21-18/2019-08-09_16-13-52-front.mp4'
Sun 11 Aug 10:39:42 PDT 2019: Moved '2019-08-09_16-21-18/2019-08-09_16-13-52-front.mp4'
Sun 11 Aug 10:39:42 PDT 2019: **Moving '2019-08-09_16-21-18/2019-08-09_16-13-52-left_repeater.mp4'**
Sun 11 Aug 10:39:55 PDT 2019: **connection dead, killing archive-clips**
Sun 11 Aug 10:39:55 PDT 2019: killing PID 15456
Sun 11 Aug 10:39:55 PDT 2019: Unmounting /mnt/cam...
umount: /mnt/cam: target is busy
        (In some cases useful info about processes that
         use the device is found by lsof(8) or fuser(1).)
Sun 11 Aug 10:39:58 PDT 2019: Failed to unmount /mnt/cam, trying lazy unmount.
Sun 11 Aug 10:39:58 PDT 2019: lazily unmounted /mnt/cam
Sun 11 Aug 10:39:58 PDT 2019: Finished archiving.
Sun 11 Aug 10:39:58 PDT 2019: Music archive not configured or unreachable
Sun 11 Aug 10:39:58 PDT 2019: unmounting /mnt/archive
Sun 11 Aug 10:39:58 PDT 2019: Connecting usb to host...
Sun 11 Aug 10:39:58 PDT 2019: Connected usb to host.
Sun 11 Aug 10:39:58 PDT 2019: Waiting for archive to be unreachable...
```

A little bit later when I am back at home:

```
Sun 11 Aug 10:53:12 PDT 2019: Waiting for archive to be reachable...
Sun 11 Aug 11:53:43 PDT 2019: Archive is reachable.
Sun 11 Aug 11:53:43 PDT 2019: Waiting for time to be set by ntpd
Sun 11 Aug 11:53:44 PDT 2019: Time now set
Sun 11 Aug 11:54:04 PDT 2019: Archiving...
Sun 11 Aug 11:54:04 PDT 2019: Disconnecting usb from host...
Sun 11 Aug 11:54:04 PDT 2019: Disconnected usb from host.
Sun 11 Aug 11:54:04 PDT 2019: Ensuring cam archive is mounted...
Sun 11 Aug 11:54:04 PDT 2019: Mounting /mnt/archive...
Sun 11 Aug 11:54:17 PDT 2019: Mounted /mnt/archive.
Sun 11 Aug 11:54:17 PDT 2019: Ensured cam archive is mounted.
Sun 11 Aug 11:54:17 PDT 2019: Starting recording archiving...
Sun 11 Aug 11:54:17 PDT 2019: Ensuring cam file is mounted...
Sun 11 Aug 11:54:17 PDT 2019: Mounting /mnt/cam...
Sun 11 Aug 11:54:17 PDT 2019: Mounted /mnt/cam.
Sun 11 Aug 11:54:17 PDT 2019: Ensured cam file is mounted.
Sun 11 Aug 11:54:17 PDT 2019: Running fsck on /mnt/cam...
Sun 11 Aug 11:54:17 PDT 2019: | fsck from util-linux 2.29.2
Sun 11 Aug 11:54:20 PDT 2019: | fsck.fat 4.1 (2017-01-24)
Sun 11 Aug 11:54:20 PDT 2019: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Sun 11 Aug 11:54:20 PDT 2019: |  Automatically removing dirty bit.
Sun 11 Aug 11:54:20 PDT 2019: | Performing changes.
Sun 11 Aug 11:54:20 PDT 2019: | /dev/loop0: 807 files, 299309/3028154 clusters
Sun 11 Aug 11:54:20 PDT 2019: Finished fsck on /mnt/cam.
Sun 11 Aug 11:54:20 PDT 2019: wake up car process is 31703
Sun 11 Aug 11:54:20 PDT 2019: Moving clips to archive...
Sun 11 Aug 11:54:20 PDT 2019: Creating output directory ''
Sun 11 Aug 11:54:20 PDT 2019: Creating output directory '**2019-08-09_16-21-18**'
Sun 11 Aug 11:54:21 PDT 2019: Failed to create '2019-08-09_16-21-18', check that archive server is writable and has free space
Sun 11 Aug 11:54:21 PDT 2019: Moved 0 file(s), failed to copy 0, deleted 0.
```
Any idea what I can do to prevent this?

marcone, 2019-08-11:
Given that, per the first part of the log you posted, the Pi successfully created a folder named 2019-08-09_16-21-18 and then moved multiple files to that folder, I don't understand how that folder could then apparently disappear and be replaced by a file with the same name.

ppamidimarri, 2019-08-12:
@marcone Thanks for the response. The problem was because of what I was doing with the footage after it was created. I move the footage to an SSD where I merge the three videos using ffmpeg. You can close this issue.

closed 2019-08-12 by marcone