marcone/teslausb #160: No longer syncing

kfulgham, 2019-08-11:
Hello,

Got this up and running a few weeks ago and it's been working great. Then since the 4th it's not transferring files to my networked drive. The TeslaUSB Pi is connected to WiFi and the drive is available for my other devices. 

Plugging the Pi into my computer does have the CAM folder properly appear and all the files from the Tesla Cam are on the drive. It does seem to just be a problem with the process to transfer the files.

Side note, I have the appropriate setup files in the pi folders.

`root@teslausb:/boot# dir
bcm2708-rpi-b.dtb	  config.txt	    overlays
bcm2708-rpi-b-plus.dtb	  COPYING.linux     start_cd.elf
bcm2708-rpi-cm.dtb	  fixup_cd.dat	    start_db.elf
bcm2708-rpi-zero.dtb	  fixup.dat	    start.elf
bcm2708-rpi-zero-w.dtb	  fixup_db.dat	    start_x.elf
bcm2709-rpi-2-b.dtb	  fixup_x.dat	    teslausb-headless-setup.log
bcm2710-rpi-3-b.dtb	  issue.txt	    TESLAUSB_SETUP_FINISHED
bcm2710-rpi-3-b-plus.dtb  kernel7.img	    WIFI_ENABLED
bcm2710-rpi-cm3.dtb	  kernel.img	    wpa_supplicant.conf.bak
bootcode.bin		  LICENCE.broadcom  wpa_supplicant.conf.sample
cmdline.txt		  LICENSE.oracle
root@teslausb:/boot# nano teslausb-headless-setup.log 
Unable to create directory /root/.nano: Read-only file system
It is required for saving/loading search history or cursor positions.

Press Enter to continue

root@teslausb:/boot# cd
root@teslausb:~# /etc/rc.local
My IP address is 192.168.87.42 
/etc/rc.local: line 24: /boot/teslausb-headless-setup.log: Read-only file system
can't write to /boot/teslausb-headless-setup.log
Detecting whether to update wpa_supplicant.conf
root@teslausb:~# already running
/root/bin/remountfs_rw
Remounting / as read/write
Remounting /boot as read/write
Done. Make any changes needed, and reboot the Pi when ready.
Filesystems / and /boot will be mounted as read only after reboot.
root@teslausb:~# /etc/rc.local
My IP address is 192.168.87.42 
Detecting whether to update wpa_supplicant.conf
root@teslausb:~# already running`

I also for some reason am only getting two flashes on the Pi.

kfulgham, 2019-08-11:
Headless Setup Log:

```
Sat 27 Jul 09:17:31 MDT 2019 : configure: Starting download for /root/bin/send-push-message
Sat 27 Jul 09:17:31 MDT 2019 : configure: Done
Sat 27 Jul 09:17:31 MDT 2019 : configure: Checking archive configs:
Sat 27 Jul 09:17:32 MDT 2019 : configure: done
Sat 27 Jul 09:17:32 MDT 2019 : configure: Using archive module: run/cifs_archive
Sat 27 Jul 09:17:32 MDT 2019 : configure: Installing base archive scripts into /root/bin
Sat 27 Jul 09:17:32 MDT 2019 : configure: Starting download for /root/bin/archiveloop
Sat 27 Jul 09:17:32 MDT 2019 : configure: Done
Sat 27 Jul 09:17:32 MDT 2019 : configure: Starting download for /root/bin/remountfs_rw
Sat 27 Jul 09:17:33 MDT 2019 : configure: Done
Sat 27 Jul 09:17:33 MDT 2019 : configure: Starting download for /root/bin/tesla_api.py
Sat 27 Jul 09:17:34 MDT 2019 : configure: Done
Sat 27 Jul 09:17:34 MDT 2019 : configure: Installing archive module scripts
Sat 27 Jul 09:17:34 MDT 2019 : configure: Starting download for /tmp/verify-archive-configuration.sh
Sat 27 Jul 09:17:34 MDT 2019 : configure: Done
Sat 27 Jul 09:17:34 MDT 2019 : configure: Starting download for /tmp/configure-archive.sh
Sat 27 Jul 09:17:35 MDT 2019 : configure: Done
Sat 27 Jul 09:17:35 MDT 2019 : configure: Starting download for /root/bin/archive-clips.sh
Sat 27 Jul 09:17:36 MDT 2019 : configure: Done
Sat 27 Jul 09:17:36 MDT 2019 : configure: Starting download for /root/bin/connect-archive.sh
Sat 27 Jul 09:17:36 MDT 2019 : configure: Done
Sat 27 Jul 09:17:36 MDT 2019 : configure: Starting download for /root/bin/disconnect-archive.sh
Sat 27 Jul 09:17:37 MDT 2019 : configure: Done
Sat 27 Jul 09:17:37 MDT 2019 : configure: Starting download for /root/bin/write-archive-configs-to.sh
Sat 27 Jul 09:17:38 MDT 2019 : configure: Done
Sat 27 Jul 09:17:38 MDT 2019 : configure: Starting download for /root/bin/archive-is-reachable.sh
Sat 27 Jul 09:17:38 MDT 2019 : configure: Done
Sat 27 Jul 09:17:38 MDT 2019 : verify-archive-configuration: Installing/updating required packages if needed
Sat 27 Jul 09:17:43 MDT 2019 : verify-archive-configuration: Done
Sat 27 Jul 09:17:43 MDT 2019 : verify-archive-configuration: Verifying that the archive server 192.168.87.97 is reachable...
Sat 27 Jul 09:17:43 MDT 2019 : verify-archive-configuration: The archive server is reachable.
Sat 27 Jul 09:17:43 MDT 2019 : verify-archive-configuration: Verifying that the archive share is mountable...
Sat 27 Jul 09:17:43 MDT 2019 : verify-archive-configuration: Mount command-line:
Sat 27 Jul 09:17:43 MDT 2019 : verify-archive-configuration: mount -t cifs //192.168.87.97/dashcam /tmp/archivetestmount -o vers=3.0,credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777
Sat 27 Jul 09:17:44 MDT 2019 : verify-archive-configuration: The archive share is mountable.
Sat 27 Jul 09:17:44 MDT 2019 : configure-archive: Configuring the archive...
Sat 27 Jul 09:17:44 MDT 2019 : configure-archive: Configured the archive.
Sat 27 Jul 09:17:44 MDT 2019 : configure: Skipping rc.local installation
Sat 27 Jul 09:17:44 MDT 2019 : make-root-fs-readonly: start
Sat 27 Jul 09:17:44 MDT 2019 : make-root-fs-readonly: Removing unwanted packages...
Sat 27 Jul 09:18:41 MDT 2019 : make-root-fs-readonly: Installing ntp and busybox-syslogd...
Sat 27 Jul 09:19:39 MDT 2019 : make-root-fs-readonly: Configuring system...
Sat 27 Jul 09:19:39 MDT 2019 : make-root-fs-readonly: Mounting the mutable partition...
Sat 27 Jul 09:19:39 MDT 2019 : make-root-fs-readonly: Mounted.
Sat 27 Jul 09:19:39 MDT 2019 : make-root-fs-readonly: Moving fake-hwclock data
Sat 27 Jul 09:19:40 MDT 2019 : make-root-fs-readonly: done
Sat 27 Jul 09:19:40 MDT 2019 : Upgrading installed packages...
Sat 27 Jul 09:23:22 MDT 2019 : All done.
Sun 11 Aug 16:24:30 MDT 2019 : Detecting whether to update wpa_supplicant.conf
```

kfulgham, 2019-08-11:
Pastebin of the diagnosis file

https://pastebin.com/mEzDpRHn

marcone, 2019-08-12:
The diagnostics show a failure every it tries to mount the archive share since August 1st. It worked before that. Maybe the server got a new IP address, or you changed the login credentials?

clean-ev, 2019-08-14:
Somehow I am getting in similar situation but very different errors - all the files found are 595bytes that are not synced

Wed 14 Aug 05:37:42 EDT 2019: Creating output directory ''
Wed 14 Aug 05:37:42 EDT 2019: Creating output directory 'RecentClips'
Wed 14 Aug 05:37:42 EDT 2019: 'RecentClips/2019-08-14_04-44-28-front.mp4' is only 595 bytes
Wed 14 Aug 05:37:42 EDT 2019: 'RecentClips/2019-08-14_04-44-28-left_repeater.mp4' is only 595 bytes
Wed 14 Aug 05:37:42 EDT 2019: 'RecentClips/2019-08-14_04-44-28-right_repeater.mp4' is only 595 bytes

Each loop creates a Recent Clips folder and Saved Clips folder that already exists - if I remember them it gets created again but there are no files.

I came back after 4 days of long distance traveling and seems to have lost everything - there were some fun things that I saved as well as nice weather across the Midwest

ppamidimarri, 2019-08-14:
@clean-ev You can solve this by following the steps here: https://github.com/marcone/teslausb/issues/143#issuecomment-517057307

marcone, 2019-09-03:
Nothing actionable here

closed 2019-09-03 by marcone