marcone/teslausb #161: Upgrade Steps - Working Config

natrlhy, 2019-08-12:
I wanted to be sure that the upgrade steps for a fully working Pi are:

`sudo -i`
`/root/bin/setup-teslausb selfupdate`

marcone, 2019-08-12:
```
sudo -i
/root/bin/remountfs_rw
/root/bin/setup-teslausb selfupdate      # this updates the setup-teslausb script itself
/root/bin/setup-teslausb                 # run the updated setup script to update everything
```

closed 2019-08-12 by marcone

curiousgrge, 2019-08-30:
Will this possibly nuke the Music partition?  It's a hassle to have to sync up a bunch of music again and then you're likely to lose all the favorite songs tagged.

marcone, 2019-08-30:
It only (re)creates the music and cam drives when the cam drive doesn't exist.
So no, just running setup again won't nuke your music. If you need to change the size of the cam and/or music drives though, you'll need to delete the cam drive first, and as a result it will also recreate the music drive.

curiousgrge, 2019-08-31:
It appears something went wrong.  The music appears to be fine after the upgrade but I've had at least twice now where sentry mode would stop recording with the grey camera icon.  There then appears another red icon (not sentry) when pressed would display a message that says, "Dashcam: USB drive is not readable"

How do I got about recreating just the cam partition?  I don't care about the data in that partition.  It still did sync despite the message that appeared and music was playing through the other partition.

samsonusmc, 2020-08-18:
Hadn't had time to look into this yet, but on a fairly recent version (not sure which) getting the following trying to upgrade to latest
root@teslausb:~# /root/bin/remountfs_rw
Remounting / as read/write
Remounting /boot as read/write
Done. Make any changes needed, and reboot the Pi when ready.
Filesystems / and /boot will be mounted as read only after reboot.
root@teslausb:~# /root/bin/setup-teslausb
/root/bin/setup-teslausb is up to date
Updating package index files...

**/root/bin/setup-teslausb: line 248: /boot/cmdline.txt: Read-only file system**

Not sure why it's still showing as RO, but checking if this is a Me issue or if others are seeing the same.