marcone/teslausb #163: Web front-end

michaelrosejr, 2019-08-14:
All,

I've created my own web frontend and TeslaUSB not knowing this repo existed. Do you have any interest in merging projects. [Here's an example of my front-end.](https://imgur.com/a/HXFYODfl). It connects to your home wifi (or whatever is setup in wpa_supplicant) and sync's files at 2AM. If you're away from home, it turns into a hotspot and allows you to connect to it and stream the dashcam videos from your phone.

Let me know if you guys are interested.
Here's the link from above in case it isnt' embedded. 
https://imgur.com/a/HXFYODf

Michael

natrlhy, 2019-08-14:
Looks awesome. If we could get some side by side integration like the desktop viewer from @natemccomb:

https://github.com/NateMccomb/TeslaCamViewer

All playable from a web front-end! :)

michaelrosejr, 2019-08-14:
I actually have code to play all three videos, the problem is the pi zero doesn't have the power. I get the following error when I try to play 3 videos side by side.

syslog.1:Aug 13 11:17:19 TeslaCams kernel: [   21.432079] Under-voltage detected!

When this happens odd things start to happen. The USB drive umounts from the Tesla, wifi pause (it doesn't disassociate, it just stops passing traffic for a bit). 
It works great when I'm testing the code on my mac, but I believe that's why all the side-by-side projects are working from a desktop/laptop or running from the cloud.

ppamidimarri, 2019-08-14:
I ran into the same problem with playing all three side-by-side. 

I tried merging them using ffmpeg, but the Pi Zero W doesn't have the processing power to handle that while maintaining a reliable USB drive for the car. So I came up with a home-based archiving solution that merges all three files, with my project called [TeslaCamMerge](https://github.com/ppamidimarri/TeslaCamMerge).

raushel, 2019-08-14:
Wrapped [Tesla_Dashcam](https://github.com/ehendrix23/tesla_dashcam) for stitching the videos together at home / managing retention and loading to Plex: [tesla_wrappers](https://github.com/raushel/tesla_wrappers)

If a Pi4 could handle it in car, that would be great though.

ppamidimarri, 2019-08-14:
> If a Pi4 could handle it in car, that would be great though.

Can a Pi 4 act as a USB mass storage device? I thought a Pi Zero can do that but not a regular Pi.

mmetzger, 2019-08-14:
A pi 4 should be able to via the USB-C port (I know the Ethernet emulation works).  The bigger issue is the power source - the recommended size is 15w which might be tricky via the USB ports.  It might work for normal booting but when actively encoding it might brownout.

marcone, 2019-09-03:
Closing this because the links in the original comment are dead, and there's already an existing feature request for a web interface.

closed 2019-09-03 by marcone