marcone/teslausb #165: Not Using TeslaCam Folders?

jbienz, 2019-08-15:
Please note that this issue may somehow be related to #164.

After getting everything setup and running last night, I noticed that teslausb isn't organizing videos as I would expect it to on the share. I expected that videos under `SavedClips` in the car would end up under `SavedClips` on the share. But instead I'm seeing the dated folders show up in the root of the share.

[image]

Is this to be expected?

marcone, 2019-08-15:
Yes. If you want the TeslaCam folder, just create it on the share, and add it to the share name

closed 2019-08-15 by marcone

jbienzms, 2019-08-15:
I do have a `TeslaCam` folder and it is part of my share name. The path on the server is shown in the screenshot above. My export is as follows:

`export sharename="Videos/Projects/TeslaCam"`

Are you saying that my export should have instead been:

`export sharename="Videos/Projects"`

?

jbienz, 2019-08-15:
I do have a `TeslaCam` folder and it is part of my share name. The path on the server is shown in the screenshot above. My export is as follows:

`export sharename="Videos/Projects/TeslaCam"`

Are you saying that my export should have instead been:

`export sharename="Videos/Projects"`

?

marcone, 2019-08-16:
Sorry, I meant 'SavedClips': export sharename="Videos/Projects/TeslaCam/SavedClips"

jbienz, 2019-08-16:
Oh I see what you mean. Does TeslaUSB only backup SavedClips then and not RecentClips? This may have been my misunderstanding of what to expect there.