marcone/teslausb #166: If wifi password has ampersand '&' character it breaks it

chiwalfrm, 2019-08-16:
I don't see anyone else reporting this issue, but here it is...

My Wifi password has several ampersand characters in it.  I did use single quotes in the  teslausb_setup_variables.conf file, but it still doesn't work.

For example in my teslausb_setup_variables.conf file, I have this line:
export WIFIPASS='thisisasecret&&&'

Now I used USB networking to boot into the Pi, and then
I looked at /etc/wpa_supplicant/wpa_supplicant.conf and it says:
psk="thisisasecretTEMPPASSTEMPPASSTEMPPASS"

So basically, it replaces each ampersand with the word TEMPPASS which is why the Pi can't get on the Wifi network.

I thought this is a bug so I am reporting this issue.

chiwalfrm, 2019-08-16:
found the problematic code in /root/bin/enable_wifi.sh:

this line:  sed -i'.bak' -e "s/TEMPPASS/${WIFIPASS}/g" /boot/wpa_supplicant.conf

if you set WIFIPASS to 'thisisasecret&&&', you can see what happens:

pi@teslausb:~ $ cat /boot/wpa_supplicant.conf.sample | sed -e "s/TEMPPASS/thisisasecret&&&/g"
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
  ssid="TEMPSSID"
  psk="thisisasecretTEMPPASSTEMPPASSTEMPPASS"
  key_mgmt=WPA-PSK
}
pi@teslausb:~ $

chiwalfrm, 2019-08-16:
i am not a developer so I don't know how to submit pull request, but the easiest fix is by replacing the sed command with perl.

Replace it like this (line #38)

before:
sed -i'.bak' -e "s/TEMPPASS/${WIFIPASS}/g" /boot/wpa_supplicant.conf

after:
cp -p /boot/wpa_supplicant.conf /boot/wpa_supplicant.conf.bak
perl -pe "s/TEMPPASS/${WIFIPASS}/g" /boot/wpa_supplicant.conf.bak > /boot/wpa_supplicant.conf

chiwalfrm, 2019-08-16:
Might also want to change the other sed line that search and replaces the SSID, as it would have the same problem if the SSID has ampersand.

line 37 :
sed -i'.bak' -e "s/TEMPSSID/${SSID}/g" /boot/wpa_supplicant.conf

chiwalfrm, 2019-08-16:
i made the changes but it didn't work, I found another file where it has the same sed code:
/etc/rc.local

So in conclusion, there are TWO files that need modification:
/root/bin/enable_wifi.sh
/etc/rc.local

marcone, 2019-08-17:
This is not limited to ampersand, and even with perl this gets complicated with other character sequences that are special in the context of the replacement string. For example, if your password is `/$1a&b`, you would have to escape that like so: `export WIFIPASS='\/\$1a&b'`

It's unfortunate that there doesn't appear to be a standard unix command that does search-and-replace _without_ regular expressions.

chiwalfrm, 2019-08-18:
How about adding an if/then condition saying that if the file /boot/wpa_supplicant.conf.OVERRIDE exists, then copy that as-is, rather than search and replace the sample file with the environment variables?  In this case, it would ignore the WIFI variables SSID and WIFIPASS in /boot/teslausb_setup_variables.conf.  What do you think?

chiwalfrm, 2019-08-18:
this solution will take care of ANY edge-cases where the wifi SSID or PASSWORD has weird special characters that has meaning to sed or perl.  See screenshot: https://imgur.com/a/KssvFd1

chiwalfrm, 2019-08-18:
```
        if [ -e /boot/wpa_supplicant.conf.OVERRIDE ]
        then
                cp /boot/wpa_supplicant.conf.OVERRIDE \
/etc/wpa_supplicant/wpa_supplicant.conf
        else
#               sed -i'.bak' -e "s/TEMPSSID/${SSID}/g" /boot/wpa_supplicant.conf
                cp -p /boot/wpa_supplicant.conf /boot/wpa_supplicant.conf.bak
                perl -pe "s/TEMPSSID/${SSID}/g" \
/boot/wpa_supplicant.conf.bak > /boot/wpa_supplicant.conf
#               sed -i'.bak' -e "s/TEMPPASS/${WIFIPASS}/g" \
/boot/wpa_supplicant.conf
                cp -p /boot/wpa_supplicant.conf /boot/wpa_supplicant.conf.bak
                perl -pe "s/TEMPPASS/${WIFIPASS}/g" \
/boot/wpa_supplicant.conf.bak > /boot/wpa_supplicant.conf
                cp /boot/wpa_supplicant.conf \
/etc/wpa_supplicant/wpa_supplicant.conf
        fi

```

marcone, 2019-08-18:
You might as well just  replace TEMPSSID and TEMPPASS in /boot/wpa_supplicant.conf

chiwalfrm, 2019-08-18:
Are you saying I can create a file wpa_supplicant.conf on the BOOT FAT32 drive (same place as the conf file) and it will just use it?  I was going through the enable_wifi.sh and rc.local scripts and it seems like it starts off by copying /boot/wpa_supplicant.conf.sample and this will overwrite wpa_supplicant.conf

chiwalfrm, 2019-08-18:
OK I think you meant 'replace TEMPSSID and TEMPPASS in /boot/wpa_supplicant.conf.sample' this will indeed do the job.

chiwalfrm, 2019-08-18:
Ugh I am a dumbass, I just realized that escaping the ampersand works and I had no idea...

this line in the conf file works:
export WIFIPASS='thisisasecret\&\&\&' 

please close this issue.

marcone, 2019-08-18:
I've added the additional escaping requirements to the OneStepSetup doc. Thanks for pointing out the issue.

closed 2019-08-18 by marcone